pschatzmann/ESP32-A2DP #102: Play Position

RustColeone, 2021-10-17:
Is it possible to get the progress of the music currently playing?

pschatzmann, 2021-10-17:
If you get the most recent version you can define the avrc information that are requested: e.g. just call set_avrc_metadata_attribut_mask(ESP_AVRC_MD_ATTR_TITLE | ESP_AVRC_MD_ATTR_ARTIST | ESP_AVRC_MD_ATTR_ALBUM | ESP_AVRC_MD_ATTR_TRACK_NUM | ESP_AVRC_MD_ATTR_NUM_TRACKS | ESP_AVRC_MD_ATTR_GENRE | AVRC_MEDIA_ATTR_ID_PLAYING_TIME);
before you call start();

However I never tried this out myself.

RustColeone, 2021-10-18:
Thanks, although I was not able to get it to work, so far I was only able to get the total time with ESP_AVRC_MD_ATTR_PLAYING_TIME, but AVRC_MEDIA_ATTR_ID_PLAYING_TIME would not compile and complain that it is not defined. No luck with getting the current position. I don't know how but I think the Bluetooth on cars is able to get the progress bar, which is why I wanted to do the same. I tried using a timer, but would instantly break when I fast forward.

PR #550 (2024-05-01): Add play_pos callback with examples and README, also playing_time clarification
After lots of digging around, I realized that rather than mess with ESP_AVRC_RN_PLAY_POS_CHANGED and esp_avrc_ct_send_register_notification_cmd and esp_avrc_rn_param_t.play_pos, we have an example implementation of a callback all ready to go with `avrc_rn_playstatus_callback(event_parameter->playback)` just a few lines above, and indeed a `av_play_pos_changed()` function ready to go, so all that's left is to make a `avrc_rn_play_pos_callback()` and we can easily display song progress with the same API we're already [... cut]
closed 2021-10-21 by RustColeone