pschatzmann/ESP32-A2DP #104: Better volume curve and 32bit DAC range full utilization

elehobica, 2021-10-17:
@pschatzmann Thanks for your great repository.

I'd like to suggest following two items.

1. Better volume curve
 I'd like to suggest more natural curve since current volume curve seems so steep that it doesn't have enough steps in lowest part. 
[image]

2.  Full bits utilization for 24/32bit DAC
 Utilizing full bits of 24/32bit DAC allows to prevent resolution loss even when small volume value is applied.

Above two items have been implemented in my branch of forked repository.
I have also confirmed that sample project works fine.

https://github.com/elehobica/ESP32-A2DP/tree/32bitDAC
sample project: https://github.com/elehobica/ESP32-A2DP/tree/32bitDAC/examples/bt_music_receiver_simple_32b_volume

If you have interest,  I willing to revise my branch to meet with your policy and issue pull request to your repository.

Regards

elehobica, 2021-10-17:
Thanks for your reply.

Yes, I know 24/32bit DAC is supported, however, it seems that i2s_write_expand doesn't utilize the resolution below 16bits. It apparently makes difference in sound quality when small volume value is applied if lower bits are fully used as suggested.

For volume control, I mean improvement from merge of rbruelma, which has changed volume curve from linear to what is indicated in my plot as 'current'. The fatal problem of 'current' curve is that volumeFactor gets 0 when s_volume is 0 ~ 10.

pschatzmann, 2021-10-17:
I am struggling a bit that we end up with 2 different parallel implementations of the audio_data_callback methods which do the same thing just with different data types and I am afraid this will be very difficult to maintain. 
Do you see any chance to merge them into one (maybe parametrized) function. This will avoid the risk that a correction is made only in one place...

On the volume control, I am sold. Just a stupid question: how are you using this. Me personally I am just connecting the DAC to an Amplifier and use the volume control of the Amplifier...

pschatzmann, 2021-10-17:
Just an other thought: I was wondering why so many users were cloning my project and doing changes. I think the 32 bot output method would be the perfect example how to solve this in an Arduino sketch just by subclassing and overwriting the audio_data_callback method.