pschatzmann/ESP32-A2DP #113: A2DP deve name scan

sanbroz, 2021-11-01:
Is there any way to scan the A2DP device around in source mode?

pschatzmann, 2021-11-01:
The following performs a scan the first time it connects and then subsequently just reconnects to the same address
```
a2dp_source.start("RadioPlayer", get_data_channels);  
```
The following always performs a scan in order to determine the device before connecting:
```
a2dp_source. set_auto_reconnect(false);
a2dp_source.start("RadioPlayer", get_data_channels);  
```

sanbroz, 2021-11-01:
Bluetooth speaker name is hard coded in sketch. I need to scan the A2DP sink devices around for a list and choose the name and connect to it dynamically. 

Regards

pschatzmann, 2021-11-01:
Are you aware that you can specify a vector of alternatives ?

pschatzmann, 2021-11-01:
And if this is not good enough just create your own subclass of BluetoothA2DPSource and overwrite the filter_inquiry_scan_result() method...

closed 2021-11-02 by pschatzmann

HankLloydRight, 2022-07-06:
I've been searching the closed issues and I just want to confirm that there is currently no method in the repo to scan for all available A2DP sinks and allow the user to select one to connect to?  And that I have to develop this code myself by changing the  filter_inquiry_scan_result() function? 

Not wanting to re-invent the wheel, has anyone done this yet?  It seems like a very basic function in applications where the A2DP sink device (any bluetooth speaker) is not known in advance. 

Thanks.

basti79, 2022-08-05:
Hi,

I'm also missing the feature of scanning/listing all available speakers.

But I found out you can give an empty String "" to a2dp_source.start an name, this will connect to the first device found.

Greets, Sebastian.