pschatzmann/ESP32-A2DP #116: L and R channels are swapped.

Oppa-YA, 2021-11-04:
Hello, after testing for left and right channels i found out that RIGHT channel is playing from LEFT and wise versa. They are swapped. I double checked PCM 5102 datasheet schematics and my PCM board are definitely are correct with L/R channels.
I double checked my phone with my other bluetooth headphones and they receiving correct L and R channels.
So ESP32 sending I2S signal with swapped L/R channels to PCM 5102. Could this be fixed in library?
You can test this with "Stereo test" app from google play.

pschatzmann, 2021-11-04:
I am confused. My library does not have any concept of left or right.

Oppa-YA, 2021-11-04:
> I am confused. My library does not have any concept of left or right.

Ah i see. So it's probably a bug in ESP-IDF Framework.
BTW could you test this on your hardware?

pschatzmann, 2021-11-04:
Not really. 

I don't have any hardware which would make any hard coded decision if channel 0 or 1 is left or right.

pschatzmann, 2021-11-04:
I could add a method which would swap the channels: 
```
void setup() {
  a2dp_sink.setSwapChannels(true);
  a2dp_sink.start("MyMusic");  
}
```

Oppa-YA, 2021-11-04:
> I could add a method which would swap the channels:
> 
> ```
> void setup() {
>   a2dp_sink.setSwapChannels(true);
>   a2dp_sink.start("MyMusic");  
> }
> ```

That would be great, thanks.

I also found a confirmation of that issue and somebody proposed how to fix it. The only problem that I don't understand how to do it.
https://github.com/espressif/esp-idf/issues/3399

Seems like one guy forked a framework and fixed it, i don't know why official is still not fixed, are you using official one?
Here is how that guy seems to fix it (i cant test it):
https://github.com/redchenjs/esp-idf/commit/3040a4d7341846cb34420393df96c5d8cf9478f7

pschatzmann, 2021-11-04:
If you go to Tools -> Board -> Board Manager and select ESP32
Which version are you using ? The latest is 2.0.0

pschatzmann, 2021-11-04:
I added the method set_swap_lr_channels().
Can you help me to test it ?

Oppa-YA, 2021-11-04:
It does not compile with version 2.0.0. But it compile with 1.0.6. But if i compile it in Arduino IDE it has a bug, devices won't connect to ESP again if I disconnect from ESP. That's why i used VS Code + Platformio now, i don't know how to check board version in it.
[image]

[image]

Oppa-YA, 2021-11-04:
> I added the method set_swap_lr_channels(). Can you help me to test it ?

Yes, how, i need to download code again from github?

pschatzmann, 2021-11-04:
Are you sure that you use the latest library version in Arduino?
I just tested both version in Arduino and I don't get any compile errors.
Dont forget to add   a2dp_sink.set_swap_lr_channels(true); - If you have an old version this does not compile as well

Oppa-YA, 2021-11-04:
> Are you sure that you use the latest library version in Arduino? I just tested both versions in Arduino and I don't get any compile errors. Dont forget to add a2dp_sink.set_swap_lr_channels(true); - If you have an old version this does not compile as well

Don't know why but after second time i deleted ESP32 lib(board) and installed again version 2.0.0 it worked!
I used code like this, L/R channels are playing correct now, thank you very much!
The only problem left is i tried to upload code from Platformio (which i used before)- device don't show in bluetooth search at all, i wonder what can cause it.

Oppa-YA, 2021-11-04:
I also got this project working on Visual Studio + Platformio, much thanks for fixing that bug, you're the best!

closed 2021-11-04 by Oppa-YA