pschatzmann/ESP32-A2DP #117: PlatformIO: cannot compile for 2.0.0 / 2.0.1-RC1

tueddy, 2021-11-04:
Hi,

Your library works fine with Arduino 1.0.6, thank's a lot! 

Now i want to use the AVRC commands starting from ESP32 Release 2.0.0 with PlatformIO. I made a simple empty project with this settings in platform.ini:

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:ttgo-t1]
platform = espressif32
board = ttgo-t1
framework = arduino
lib_deps =
	https://github.com/pschatzmann/ESP32-A2DP.git
platform_packages =
	platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0

Error while compiling:

Compiling .pio\build\ttgo-t1\libe56\ESP32-A2DP\BluetoothA2DPSource.cpp.o
In file included from C:/Users/Dirk/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/hal/esp32/include/hal/i2s_ll.h:30:0,
                 from C:/Users/Dirk/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/hal/include/hal/i2s_hal.h:28,
                 from C:/Users/Dirk/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/driver/include/driver/i2s.h:16,
                 from .pio/libdeps/ttgo-t1/ESP32-A2DP/src/BluetoothA2DPCommon.h:36,
                 from .pio/libdeps/ttgo-t1/ESP32-A2DP/src/BluetoothA2DPCommon.cpp:16:
C:/Users/Dirk/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:70:31: error: expected '}' before '__attribute__'  
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
                               ^
C:/Users/Dirk/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:70:59: error: expected unqualified-id before '=' token
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
                                                           ^
C:/Users/Dirk/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:76:3: error: 'i2s_comm_format_t' does not name a type
 } i2s_comm_format_t;
   ^

Have i missed something, any idea?
Thank's & best
Dirk

pschatzmann, 2021-11-05:
It seems that the release determination which is in config.h is not working properly in PlatformIO. Can you try my adding 
build_flags = -DSSID_NAME= CURRENT_ESP_IDF

If this helps, would you know why config.h fails in PlatformIO ?

tueddy, 2021-11-05:
Hi,

 i've to get compiled with this platform.ini:

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:ttgo-t1]
;platform = espressif32
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream

board = ttgo-t1
framework = arduino
lib_deps =
	https://github.com/pschatzmann/ESP32-A2DP.git
platform_packages =
	platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1-RC1

Thank's!

closed 2021-11-05 by tueddy