pschatzmann/ESP32-A2DP #118: A2DPSink: Crash with Arduino 2.0.1-RC1

tueddy, 2021-11-07:
Starting a BT sink with Arduino 2.0.1-RC1 produces a crash. 
Older version https://github.com/pschatzmann/ESP32-A2DP.git#2800d69 works fine:

```
Guru Meditation Error: Core  1 panic'ed (LoadStoreAlignment). Exception was unhandled.
Core 1 register dump:
PC      : 0x4009907e  PS      : 0x00060733  A0      : 0x800979dc  A1      : 0x3ffd15f0
A2      : 0x3f41f7c5  A3      : 0x0000abab  A4      : 0xb33fffff  A5      : 0x00000001
A6      : 0x00060720  A7      : 0x0000cdcd  A8      : 0x0000abab  A9      : 0x3ffd15f0
A10     : 0x00000003  A11     : 0x00060723  A12     : 0x00060720  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x3ffc2110  SAR     : 0x00000020  EXCCAUSE: 0x00000009  
EXCVADDR: 0x3f41f7c5  LBEG    : 0x40094570  LEND    : 0x4009458c  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x4009907e:0x3ffd15f0 0x400979d9:0x3ffd1620 0x40107edf:0x3ffd1660 0x401076dd:0x3ffd1680 0x4010761c:0x3ffd16b0 0x400d4d1e:0x3ffd16f0 0x400df9ee:0x3ffd1720 0x4010c822:0x3ffd1740 0x40098076:0x3ffd1760
  #0  0x4009907e:0x3ffd15f0 in uxPortCompareSet at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4691
      (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.inc.h:86
      (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.h:98
      (inlined by) vTaskEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4201  #1  0x400979d9:0x3ffd1620 in xQueueGenericSend at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:2520
  #2  0x40107edf:0x3ffd1660 in BluetoothA2DPSink::app_send_msg(app_msg_t*) at .pio/libdeps/ttgo_t8/ESP32-A2DP/src/BluetoothA2DPSink.cpp:332
  #3  0x401076dd:0x3ffd1680 in BluetoothA2DPSink::app_work_dispatch(void (*)(unsigned short, void*), unsigned short, void*, int) at .pio/libdeps/ttgo_t8/ESP32-A2DP/src/BluetoothA2DPSink.cpp:308
  #4  0x4010761c:0x3ffd16b0 in BluetoothA2DPSink::start(char const*, bool) at .pio/libdeps/ttgo_t8/ESP32-A2DP/src/BluetoothA2DPSink.cpp:149
  #5  0x400d4d1e:0x3ffd16f0 in Bluetooth_Init() at src/Bluetooth.cpp:68
  #6  0x400df9ee:0x3ffd1720 in setup() at src/main.cpp:223
  #7  0x4010c822:0x3ffd1740 in loopTask(void*) at C:/users/dirk/.platformio/packages/framework-arduinoespressif32@src-5ca93290a197adbe2ee160f158f219df/cores/esp32/main.cpp:18
  #8  0x40098076:0x3ffd1760 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

Any idea what's going wrong here?
Thank's!

pschatzmann, 2021-11-09:
I just tried with the released Arduino 2.0.1 and did not get any problems...

tueddy, 2021-11-11:
With Arduino 2.0.1 release the crash is gone & everything works as expected! 
Thank's

closed 2021-11-11 by tueddy