pschatzmann/ESP32-A2DP #120: bt_music_sender_write not working

BriscoeTech, 2021-11-08:
Hello pschatzmann,

I am trying to get your bt_music_sender_write example to run and having some issues.

I have an adafruit esp32 and am trying to run the project. I reduced the size of StarWars30_raw so it will fit into my boards ram, and am now able to program my board. 

My Bluetooth speaker makes the successful connection sound, but then no audio plays or can be heard.

I wanted to ask, what version of the esp32 core are you using? Also what esp32 board are your compiling for that has so much ram? I am trying to replicate your exact setup as a starting point to see if everything is working

Thank you for your help :-)

pschatzmann, 2021-11-08:
Hallo, I think the trick is to select the Partition Scheme Huge App.

ps. you also might be interested in the related Arduino Audio Tools project: https://github.com/pschatzmann/arduino-audio-tools where I demo how to use many other audio sources...

pschatzmann, 2021-12-04:
closed due to no activity

closed 2021-12-04 by pschatzmann