pschatzmann/ESP32-A2DP #125: Error getting NVS blob

paulhayes, 2021-11-17:
I'm getting the following errors on an esp32-cam using your ```bt_music_sender_write``` example. I've configured a partition table csv which includes space for the nvs.  However at runtime I get the following error when trying to connect to my device.

```
[E][BluetoothA2DPCommon.cpp:126] get_last_connection(): ERROR GETTING NVS BLOB
[E][BluetoothA2DPCommon.cpp:129] get_last_connection(): NVS NOT FOUND
```
I'm using platformio and the arduino framework. Here's my platformio.ini

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32cam]
platform = espressif32
board = esp32cam
framework = arduino
monitor_speed = 115200
upload_speed = 2000000
board_build.partitions = huge_app.csv
lib_deps = 
    https://github.com/pschatzmann/ESP32-A2DP
```
Is there a step I'm missing?

pschatzmann, 2021-11-17:
I don't think that there is a problem and I think I will change the error level of these messages to Info. If autoconnect is set the system is retrieving the last successful connection. If you connect for the first time there is nothing found in NVS, hence you get this error.

My expectation is that after you have connected successfully the first time - this message will go away....

pschatzmann, 2021-11-18:
The error has been changed into an info message with the last commit

paulhayes, 2021-11-18:
Thanks for getting back to be so quick. Glad it's not a problem. 

Also thanks for making this great library. The Esp-adf looked like a real headache to use.

closed 2021-11-18 by paulhayes