pschatzmann/ESP32-A2DP #126: Compiler Warning

clarkbremer, 2021-11-17:
First, thanks for this awesome library!  Can't wait to dig in further.  Just an FYI, I got some compiler warnings.  If you don't care, that's cool, just close this issue.  CB.

```
C:\Users\clark\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPCommon.cpp:264:1: warning: 'virtual bool BluetoothA2DPCommon::isConnected()' is deprecated [-Wdeprecated-declarations]

 }

 ^

In file included from C:\Users\clark\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPCommon.cpp:16:0:

C:\Users\clark\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPCommon.h:111:22: note: declared here

         virtual bool isConnected(){

                      ^

C:\Users\clark\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPSink.cpp:1096:1: warning: 'virtual bool BluetoothA2DPCommon::isConnected()' is deprecated [-Wdeprecated-declarations]

 }

 ^

In file included from C:\Users\clark\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPSink.h:17:0,

                 from C:\Users\clark\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPSink.cpp:16:

C:\Users\clark\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPCommon.h:111:22: note: declared here

         virtual bool isConnected(){

                      ^

C:\Users\clark\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPSink.cpp:1096:1: warning: 'virtual void BluetoothA2DPSink::set_on_connected2BT(void (*)())' is deprecated [-Wdeprecated-declarations]

 }

 ^

C:\Users\clark\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPSink.cpp:101:6: note: declared here

 void BluetoothA2DPSink::set_on_connected2BT(void (*callBack)()){

      ^

C:\Users\clark\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPSink.cpp:1096:1: warning: 'virtual void BluetoothA2DPSink::set_on_dis_connected2BT(void (*)())' is deprecated [-Wdeprecated-declarations]

 }

 ^

C:\Users\clark\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPSink.cpp:105:6: note: declared here

 void BluetoothA2DPSink::set_on_dis_connected2BT(void (*callBack)()){

      ^

C:\Users\clark\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPSource.cpp:952:1: warning: 'virtual bool BluetoothA2DPCommon::isConnected()' is deprecated [-Wdeprecated-declarations]

 }

 ^

In file included from C:\Users\clark\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPSource.h:22:0,

                 from C:\Users\clark\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPSource.cpp:16:

C:\Users\clark\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPCommon.h:111:22: note: declared here

         virtual bool isConnected(){

                      ^
```

pschatzmann, 2021-11-17:
I will remove these methods in the next release and then the warnings will go away.
I just wanted to alert the users that have used these obsolete methods

closed 2021-11-17 by pschatzmann

clarkbremer, 2021-11-17:
Thanks!  Would not have noticed if I wasn't using vscode.