pschatzmann/ESP32-A2DP #128: OneChannelSoundData data playing at 2x speed.

paulhayes, 2021-11-18:
In your mono example ```bt_music_sender_write```, you StarWars30_raw plays at double speed. Is that meant to happen, or is there a bug where it's sending even samples to left channel and odd samples to right?

pschatzmann, 2021-11-18:
I think I just used the wrong file and never bothered to correct this

closed 2021-11-18 by paulhayes

kzka90, 2021-11-26:
I made an audio header-less file as stated on the **README** with **Audacity**. And converted it to **.c** as follows:

`xxd -i file_example_WAV_1MG.raw file_example_WAV_1MG.c`

And my created file is also playing a **2x** speed when run on ESP32.

What could I be doing wrong?

pschatzmann, 2021-11-27:
Did you double check that you converted it to mono (with 1 channel) and the sampling rate is 44100?

I just did some tests and for me it is working as expected. I took the opportunity to update the audio file in the example with one with the correct sampling rate.

I have also added 8bit support and documented this with an example.