pschatzmann/ESP32-A2DP #129: Any method to send battery %  to phone after connecting.

dev-techshlok, 2021-11-20:
Amazing work.
Just curious to know any method available to send battery % to the phone just like Bluetooth speakers sends.
I am getting battery % in a variable-type float.

Please let me know if it's possible and is there any available method.

Thank you.

pschatzmann, 2021-11-20:
I have never tried but you can find the information here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_avrc.html

I guess it might be something like this
```
esp_avrc_rn_param_t rn_param;
rn_param.batt = ESP_AVRC_BATT_NORMAL;
esp_avrc_tg_send_rn_rsp(ESP_AVRC_RN_BATTERY_STATUS_CHANGE, ESP_AVRC_RN_RSP_CHANGED, &rn_param);
```

dev-techshlok, 2021-11-22:
Can you provide a direction on how I can use this?
What I tried so far was. 

Included is this header file.
`#include "esp_avrc_api.h"`

then wrote these lines in setup.

```
esp_avrc_rn_param_t rn_param;
rn_param.batt = ESP_AVRC_BATT_NORMAL;
esp_avrc_tg_send_rn_rsp(ESP_AVRC_RN_BATTERY_STATUS_CHANGE, ESP_AVRC_RN_RSP_CHANGED, &rn_param);
```
It gave me error ` 'esp_avrc_tg_send_rn_rsp' was not declared in this scope`
.
I also tried to put these by making a function in the original library but got the same error about, declaration.

pschatzmann, 2021-11-22:
You start with one of my examples. 
Then you add the calls from above in the loop and make sure that it is only executed when
- is_connected () returns true
- you execute it only e.g. every 1 minute e.g. if (millis() % 60000 == 0)

```
void loop() {
  if (is_connected () && millis() % 60000 == 0){
    Serial.println("update battery status");
   esp_avrc_rn_param_t rn_param;
   rn_param.batt = ESP_AVRC_BATT_NORMAL;
   esp_avrc_tg_send_rn_rsp(ESP_AVRC_RN_BATTERY_STATUS_CHANGE, ESP_AVRC_RN_RSP_CHANGED, &rn_param);
  }
}

```
You would still need to map your numerical input to the valid status values.

ps. there is no need to include #include "esp_avrc_api.h" again because it is already part of BluetoothA2DPSink.h.

dev-techshlok, 2021-11-23:
Hello @pschatzmann
Thank you for this super-fast response. I tried this method. But it still says the same. 

Here are my steps.

I added these lines recommended by you 

The full Code Screenshot

[image]

The full code

```
#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;

void setup() {
  const i2s_config_t i2s_config = {
    .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN),
    .sample_rate = 44100, // corrected by info from bluetooth
    .bits_per_sample = (i2s_bits_per_sample_t) 16, /* the DAC module will only take the 8bits from MSB */
    .channel_format =  I2S_CHANNEL_FMT_RIGHT_LEFT,
    .communication_format = (i2s_comm_format_t)I2S_COMM_FORMAT_STAND_MSB,
    .intr_alloc_flags = 0, // default interrupt priority
    .dma_buf_count = 8,
    .dma_buf_len = 64,
    .use_apll = false
  };

  a2dp_sink.set_i2s_config(i2s_config);
  a2dp_sink.start("InternalDAC");

}

unsigned long battery_timeout = 0;

void loop() { 
  bool is_connected = a2dp_sink.get_connection_state();
  if (is_connected  && millis() > battery_timeout) {
    Serial.println("update battery status");
    esp_avrc_rn_param_t rn_param;
    rn_param.batt = ESP_AVRC_BATT_NORMAL;
    esp_avrc_tg_send_rn_rsp(ESP_AVRC_RN_BATTERY_STATUS_CHANGE, ESP_AVRC_RN_RSP_CHANGED, &rn_param);
    battery_timeout = millis() + 60000;
  }
}
```

Here is the error,
```

C:\Users\Hitesh\AppData\Local\Temp\arduino_modified_sketch_487165\bt_music_receiver_to_internal_dac.ino: In function 'void loop()':
bt_music_receiver_to_internal_dac:47:4: error: 'esp_avrc_rn_param_t' was not declared in this scope
    esp_avrc_rn_param_t rn_param;
    ^
bt_music_receiver_to_internal_dac:48:4: error: 'rn_param' was not declared in this scope
    rn_param.batt = ESP_AVRC_BATT_NORMAL;
    ^
bt_music_receiver_to_internal_dac:48:20: error: 'ESP_AVRC_BATT_NORMAL' was not declared in this scope
    rn_param.batt = ESP_AVRC_BATT_NORMAL;
                    ^
bt_music_receiver_to_internal_dac:49:63: error: 'ESP_AVRC_RN_RSP_CHANGED' was not declared in this scope
    esp_avrc_tg_send_rn_rsp(ESP_AVRC_RN_BATTERY_STATUS_CHANGE, ESP_AVRC_RN_RSP_CHANGED, &rn_param);
                                                               ^
bt_music_receiver_to_internal_dac:49:97: error: 'esp_avrc_tg_send_rn_rsp' was not declared in this scope
    esp_avrc_tg_send_rn_rsp(ESP_AVRC_RN_BATTERY_STATUS_CHANGE, ESP_AVRC_RN_RSP_CHANGED, &rn_param);
                                                                                                 ^
Using library ESP32-A2DP-main at version 1.6.0 in folder: C:\Users\Hitesh\Documents\Arduino\libraries\ESP32-A2DP-main 
exit status 1
'esp_avrc_rn_param_t' was not declared in this scope
```
The error Screenshot

[image]

Thank you

pschatzmann, 2021-11-23:
I guess you are still on an old version!
https://github.com/pschatzmann/ESP32-A2DP/wiki/Controlling-your-Phone-with-AVRC-Commands

dev-techshlok, 2021-11-23:
Hello @pschatzmann 

Thank You so much. Compiled successfully.

But after a minute also unable to see the battery status, Tested on these phones:
iPhone 6
Realme C21Y
Remi 9A

pschatzmann, 2021-11-23:
Just out of curiosity - where would you see this information ?

dev-techshlok, 2021-11-23:
I was jut trying to make a Bluetooth speaker, as all Bluetooth speaker shows battery % when connected. I was thinking if it can be done.

pschatzmann, 2021-11-23:
I have never seen anything like this when I connect with my IPAD, Android and OS/X.
But I am still using pretty old versions...

clarkbremer, 2021-11-23:
[image]
From my Android phone connected to my bt headphones.

pschatzmann, 2021-11-23:
Thanks - so it's definitely not like this in my Android 7.0
I guess you might have more luck with the official ESP32 IDF support 
https://www.esp32.com/viewforum.php?f=23&sid=b88c799d9c0ab0b506bd5e094342dd44

dev-techshlok, 2021-11-24:
Thank you @pschatzmann.

IF anything comes up I'll share a PR. Or comment here.

closed 2021-11-24 by dev-techshlok