pschatzmann/ESP32-A2DP #130: Is there a way to get the name of the connected source device?

clarkbremer, 2021-11-23:
I looked, and did not see such a method in your library.  I did find this:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_gap_bt.html#_CPPv427esp_bt_gap_read_remote_name13esp_bd_addr_t

That's in the "GAP" api?  Not sure if that makes it out-of-scope for your library.

pschatzmann, 2021-11-23:
Are you willing to extend the library yourself and after it is working submit a pull request ?
It should not be too difficult to implement. Alternatively you could "abuse" the set_address_validator()

clarkbremer, 2021-11-23:
I'll give it a shot!  You prefer a PR from a fork?

pschatzmann, 2021-11-23:
yes

clarkbremer, 2021-11-24:
Sorry to bother you, but I'm stumped why I'm getting this compiler error:
` error: 'ESP_BT_GAP_READ_REMOTE_NAME_EVT' was not declared in this scope`
I see you using constants like `ESP_BT_GAP_KEY_REQ_EVT`, which seems to be of the same ilk.

pschatzmann, 2021-11-24:
What arduino-esp32 version are you using ?
Tools -> Board -> Board Manager and enter ESP32

clarkbremer, 2021-11-24:
Ahhh...  "Heltec ESP32 Series Dev-boards by Heltec Automation(TM) Version 0.0.5"

I used that because it has support for the cute little OLED display built in to the board.  Must have an old version of the Espressif code?

pschatzmann, 2021-11-24:
I am expecting something like esp32 by Espressif Systems version 1.0.6, 2.0.0 or 2.0.1.
You would need at least 2.0.0

pschatzmann, 2021-11-24:
https://github.com/pschatzmann/ESP32-A2DP/wiki/Controlling-your-Phone-with-AVRC-Commands

clarkbremer, 2021-11-24:
Bingo!  Sorry to trouble you with such basic stuff.

clarkbremer, 2021-11-29:
I pushed up a PR, but it's WIP.  It's inconsistent, and usually returns a blanks string.  How do you enable logging?  Does it appear in the serial monitor?

pschatzmann, 2021-11-29:
Tools -> Core Debug Level -> e.g. Verbose
The messages appear in the Serial monitor.

I wait with the merge until you confirm that it is working... 
And dont forget to check that it is also compiling in the old version.

clarkbremer, 2021-11-29:
Please explain what you mean by "also compiling in the old version".

pschatzmann, 2021-11-29:
Espressif Systems version 1.0.6,
I suggest to add the esp_bt_gap_read_remote_name(peer_bd_addr); call to the place where the peer_bd_addr is set. So it will be part of the initialization logic...

clarkbremer, 2021-11-29:
Line 546: That's approximately where I put it.  I placed it at the end of the else clause, after checking that it was valid, etc.

Thanks for the tip on the logs.  I'm seeing messages like:
```[577165][I][BluetoothA2DPSink.cpp:456] app_gap_callback(): ESP_BT_GAP_READ_REMOTE_NAME_EVT remote name:00:00:f0:2d:fd:3f`

clarkbremer, 2021-11-29:
And, no, it does not compile with the old version.  This feature was added in version 4.1-beta1:  https://github.com/espressif/esp-idf/releases/tag/v4.1-beta1

pschatzmann, 2021-11-29:
Then you can just wrap the pieces which give an error with
``` 
#ifdef CURRENT_ESP_IDF
#endif
```

clarkbremer, 2021-11-29:
That works.
I'm still struggling with `app_gap_callback()`.  If I'm understanding correctly, the *param is pointing at a union, so the fields in the struct depend on what the event type was.  How is that you are reading the dba from it, regardless of the event type?  Must be related to why `param->read_rmt_name.rmt_name` returns garbage, even when event type is `ESP_BT_GAP_READ_REMOTE_NAME_EVT`.

clarkbremer, 2021-11-30:
OK.  PR ready now. There's a status field in the response that indicates and error.  Was also not sending the correct board address.

clarkbremer, 2021-11-30:
[image]

pschatzmann, 2021-12-01:
The pull request has been merged. 
Thanks a lot for your contribution!

closed 2021-12-01 by pschatzmann

clarkbremer, 2021-12-01:
Happy to help.  BTW, I still think this line is incorrect: https://github.com/pschatzmann/ESP32-A2DP/blob/main/src/BluetoothA2DPSink.cpp#L421

That would only be valid if event is `ESP_BT_GAP_CFM_REQ_EVT`.  Turns out that the field `bda` *is* defined on a lot of the other members of the union, so it probably works a lot of the time, but not all.  Certainly not for `ESP_BT_GAP_READ_REMOTE_NAME_EVT`.

pschatzmann, 2021-12-01:
I agree

pschatzmann, 2021-12-01:
The correction has been committed.