pschatzmann/ESP32-A2DP #131: Internal DAC noise when used as Expressive IDF component

odisei369, 2021-11-26:
Hi.

First of all thank you for the library.
It has very nice API and is really easy to use.

I've tried to use it both as Arduino IDE library and as Expressive IDF component.
And the thing is it works ok as Arduino IDE library, but has wrong sound when used as Expressive IDF component (it's like making this sound higher so you could not even understand words. Maybe I should record it?).

I've tried to play with this code https://github.com/pschatzmann/ESP32-A2DP/blob/74951f6175510ea0a771eaeaefd6a6f0a99e192a/src/BluetoothA2DPSink.cpp#L906 because my guess is that issue is related to different data format between I2C and internal DAC, but no luck. I don't have osciloscope so it's hard for me to debug it.

I've went through esp-idf issues, but didn't found exact match with my issue. There are some possibly related issues, like this [esp-idf/6470](https://github.com/espressif/esp-idf/issues/6470)

My test environment:

**Board:** ESP32-DevKitC V4
**esp-idf:** tried both v5.0 and v4.4

The code for Arduino IDE:
```C
#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;

void setup() {
  const i2s_config_t i2s_config = {
      .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN),
      .sample_rate = 44100, // corrected by info from bluetooth
      .bits_per_sample = (i2s_bits_per_sample_t) 16, /* the DAC module will only take the 8bits from MSB */
      .channel_format =  I2S_CHANNEL_FMT_RIGHT_LEFT,
      .communication_format = (i2s_comm_format_t)I2S_COMM_FORMAT_STAND_MSB,
      .intr_alloc_flags = 0, // default interrupt priority
      .dma_buf_count = 8,
      .dma_buf_len = 64,
      .use_apll = false
  };

  a2dp_sink.set_i2s_config(i2s_config);  
  a2dp_sink.start("InternalDAC");  

}

void loop() {
}
```

The code for Expressive IDF:
```C
#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;

void setup(){
  const i2s_config_t i2s_config = {
        .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN),
        .sample_rate = 44100, // corrected by info from bluetooth
        .bits_per_sample = (i2s_bits_per_sample_t) 16, /* the DAC module will only take the 8bits from MSB */
        .channel_format =  I2S_CHANNEL_FMT_RIGHT_LEFT,
        .communication_format = (i2s_comm_format_t)I2S_COMM_FORMAT_STAND_MSB,
        .intr_alloc_flags = 0, // default interrupt priority
        .dma_buf_count = 8,
        .dma_buf_len = 64,
        .use_apll = false
    };

    a2dp_sink.set_i2s_config(i2s_config);  
    a2dp_sink.start("InternalDAC");   
}

void loop(){
  //  delay( 500 );
}

extern "C" void app_main(void){
  setup();
  while(true){
    loop();
  }
}
```

I would like to fix this behaviour and create pull request, but need help with direction where should I dig to fix it.

pschatzmann, 2021-11-26:
Since I do not have anything specific to Arduino in my code the issue must be either
- the IDF (Arduino is based on IDF 2.2)
- the sketch:  The loop might not yield properly: therefore you see the delay(). Something like taskYIELD() after the loop() call might also do the work as well.

odisei369, 2021-11-26:
Thank you for your response. It helped a lot. Didn't know about version Arduino is based on.

Then I think that's the issue with IDF version.

Because according to this issue https://github.com/espressif/esp-idf/issues/6470
it was broken somewhere in 3.3.0 version of esp-idf and was not fixed yet.

cfint, 2022-02-03:
It's not the above issue. I've bisect this to a more recent change that broke the internal DAC: https://github.com/espressif/esp-idf/issues/8344

Issue 6470 was a problem with _no audio_. It later got fixed some where in the midst of code refactors since `v4.1`. The current issue with the internal DAC is it produces audio but it's all garbled.

pschatzmann, 2022-04-20:
The Internal DAC has been fixed in the latest IDF release

closed 2022-04-20 by pschatzmann