pschatzmann/ESP32-A2DP #132: distorted sound when using external DAC with non-standard pins

hgreindl, 2021-11-29:
I'm trying to receive audio via Bluetooth and stream it to a amplifier using an uda1334 external dac.

Pin configuration used in my project is:
i2s_pin_config_t my_pin_config = {
      .bck_io_num = 12,             // Bit Clock
      .ws_io_num =  32,             //  wclk, Left/Right Clock
      .data_out_num = 33,           // Data Out
      .data_in_num = 34  // NOT USED
    };

The configuration used in my project is:
static const i2s_config_t i2s_config = {
         .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX),
         .sample_rate = 41000, // also tried with 44100, same effect
         .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
         .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
         .communication_format = (i2s_comm_format_t) (I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_LSB),
         .intr_alloc_flags = 0, // default interrupt priority
         .dma_buf_count = 8, // also tried with more and bigger buffers
         .dma_buf_len = 64,
         .use_apll = false,
         .tx_desc_auto_clear = true // avoiding noise in case of data unavailability
    };

Whe I play a square wave generated in the program, the audio through the external DAC is clear without any noise or distortion, but when I stream from my bluetooth source (doesn't matter if notebook or phone, I tried both of them each with several devices) the audio is very distorted and almost not recognizable.

The power supply is done via USB from either a stabilized source or from a notebook and converted to 3.3V using a DC/DC converter.

Does anyone have an idea what I'm doing wrong?

pschatzmann, 2021-11-29:
Are you sure that you are using the same I2S definitions in your test program and in A2DP ? 
In your test are you generating 2 channels of int16_t data ?

If there are issues usually it helps to switch I2S_COMM_FORMAT_I2S_LSB and I2S_COMM_FORMAT_I2S_MSB, but as far as I know for your DAC I2S_COMM_FORMAT_I2S_LSB should be the right setting

hgreindl, 2021-11-29:
I also used your example bt_music_receiver_simple, changed my pin settings to the one show in the initial question and removed I2S_MODE_DAC_BUILT_IN in the mode settings. Given your example I also tried with I2S_COMM_FORMAT_I2S_LSB and I2S_COMM_FORMAT_I2S_MSB, but the result stays the same, the audio is almost not recognizable. 

Since Serial plotter did not give anything useful, I just checked with serial monitor and values range form 0 to 254, which should be ok.

I'll try to playback some file from internal file system and see what happens.

pschatzmann, 2021-11-29:
if you interpret the array as int16 + int16 you should see some proper sound waves for the 2 channels. The values should be 
between -32,768 to 32,767