pschatzmann/ESP32-A2DP #140: AVRC event notification ids

Xekz, 2021-12-11:
How to get ESP_AVRC_RN_PLAY_POS_CHANGED?

pschatzmann, 2021-12-11:
Have you read the documentation section Support for Metadata?

Xekz, 2021-12-11:
IN AVRC metadata attribute mask
typedef enum {
    ESP_AVRC_MD_ATTR_TITLE = 0x1,                 /*!< title of the playing track */
    ESP_AVRC_MD_ATTR_ARTIST = 0x2,                /*!< track artist */
    ESP_AVRC_MD_ATTR_ALBUM = 0x4,                 /*!< album name */
    ESP_AVRC_MD_ATTR_TRACK_NUM = 0x8,             /*!< track position on the album */
    ESP_AVRC_MD_ATTR_NUM_TRACKS = 0x10,           /*!< number of tracks on the album */
    ESP_AVRC_MD_ATTR_GENRE = 0x20,                /*!< track genre */
    ESP_AVRC_MD_ATTR_PLAYING_TIME = 0x40          /*!< total album playing time in miliseconds */
} esp_avrc_md_attr_mask_t;

needs this 
typedef enum {
    ESP_AVRC_RN_PLAY_STATUS_CHANGE = 0x01,        /*!< track status change, eg. from playing to paused */
    ESP_AVRC_RN_TRACK_CHANGE = 0x02,              /*!< new track is loaded */
    ESP_AVRC_RN_TRACK_REACHED_END = 0x03,         /*!< current track reached end */
    ESP_AVRC_RN_TRACK_REACHED_START = 0x04,       /*!< current track reached start position */
    ESP_AVRC_RN_PLAY_POS_CHANGED = 0x05,          /*!< track playing position changed */
    ESP_AVRC_RN_BATTERY_STATUS_CHANGE = 0x06,     /*!< battery status changed */
    ESP_AVRC_RN_SYSTEM_STATUS_CHANGE = 0x07,      /*!< system status changed */
    ESP_AVRC_RN_APP_SETTING_CHANGE = 0x08,        /*!< application settings changed */
    ESP_AVRC_RN_MAX_EVT
} esp_avrc_rn_event_ids_t;

pschatzmann, 2021-12-11:
I am afraid I don't understand.
I dont have any enums in my code!

Xekz, 2021-12-12:
try get track playing position in bt_music_receiver.
in  a2dp_sink.set_avrc_metadata_callback(avrc_metadata_callback); get only ESP_AVRC_MD_ATTR_TITLE | ESP_AVRC_MD_ATTR_ARTIST | ESP_AVRC_MD_ATTR_ALBUM | ESP_AVRC_MD_ATTR_TRACK_NUM | ESP_AVRC_MD_ATTR_NUM_TRACKS | ESP_AVRC_MD_ATTR_GENRE | ESP_AVRC_MD_ATTR_PLAYING_TIME
but not get  ESP_AVRC_RN_PLAY_POS_CHANGED and ESP_AVRC_RN_TRACK_REACHED_START

pschatzmann, 2021-12-12:
I just double checked: currently my library does not call esp_avrc_ct_send_register_notification_cmd, therefore this is not working. I will not be able to work on this myself for the next couple of weeks...

Will you be able to implement this ?

Xekz, 2021-12-14:
Thanks, well I'll try

hpkoopman, 2022-03-12:
@Xekz dit you have any luck with this? Looks like it is an event update and not a metadata. I am also looking into this. Would be nice if we can see the actual position in the track.

Xekz, 2022-03-12:
@[hpkoopman](https://github.com/hpkoopman) it didn't work out

closed 2022-04-20 by pschatzmann