pschatzmann/ESP32-A2DP #143: ESP_A2D_AUDIO_STATE_REMOTE_SUSPEND vs ESP_A2D_AUDIO_STATE_STOPPED

marcelstoer, 2021-12-18:
Not a bug or a feature request but a question.

How is `ESP_A2D_AUDIO_STATE_REMOTE_SUSPEND` different from `ESP_A2D_AUDIO_STATE_STOPPED` i.e. in which case should I get which state?

I see that when I stop the music on the mobile I get `ESP_A2D_AUDIO_STATE_REMOTE_SUSPEND`. So, how would I detect when the mobile disconnected (intentionally or by loosing connection).

pschatzmann, 2021-12-19:
I guess ESP_A2D_AUDIO_STATE_REMOTE_SUSPEND is used when it is expected that the playing get's resumed at some later time.
If the mobile is connected or not is determined via the connection state related methods:
- isConnected()
- get_connection_state ()
- set_connection_state(callback,ref) callback method

<br class="Apple-interchange-newline">

marcelstoer, 2021-12-20:
Thanks for the explanation

closed 2021-12-20 by marcelstoer