pschatzmann/ESP32-A2DP #146: A2DP + SD -> I2S -> DECODER -> AMP -> SPEAKER

podaen, 2021-12-24:
I am trying to use the A2DP library and the Audiotools library together with an LogarithmicVolumeControl and I get this

> VolumeControl.h: 66:37: error: reference to 'VolumeControl' is ambiguous
>  class DefaultVolumeControl *: public VolumeControl {
> 
> VolumeControl.h: 87:47: error: reference to 'VolumeControl' is ambiguous
>  class SimpleExponentialVolumeControl *: public VolumeControl {
> 
> VolumeControl.h: 106:36: error: reference to 'VolumeControl' is ambiguous
>  class LinearVolumeControl *: public VolumeControl {
> 
> VolumeControl.h: 121:32: error: reference to 'VolumeControl' is ambiguous
>  class NoVolumeControl *: public VolumeControl {
> 
> BluetoothA2DPCommon.h: 127:41: error: reference to 'VolumeControl' is ambiguous
>    virtual void set_volume_control(VolumeControl *ptr){
> 
> BluetoothA2DPCommon.h: 172:9: error: reference to 'VolumeControl' is ambiguous
>    VolumeControl *volume_control_ptr = nullptr
> 
> BluetoothA2DPCommon.h: 198:17: error: reference to 'VolumeControl' is ambiguous
>    virtual VolumeControl* volume_control() {
> 
> BluetoothA2DPSource.h:22: In file included from
> 
> BluetoothA2DPCommon.h: In member function virtual void BluetoothA2DPCommon::set_volume_control(int*)
> 
> BluetoothA2DPCommon.h: 128:13: error: 'volume_control_ptr' was not declared in this scope
>    volume_control_ptr = ptr

I suspect the name VolumeControl is used two times once in A2DP and once in AudioTools. On the last one I have no Idea

pschatzmann, 2021-12-24:
Unfortunalty I am using the same class name in both projects and the A2DP project is not using any namespace.
In my examples the code is compiling if you use the import in the following sequence:

#include "BluetoothA2DPSink.h"
#include "AudioTools.h"

podaen, 2021-12-24:
I diden't try it without the LogarithmicVolumeControl. Maybe this the cause.

pschatzmann, 2021-12-24:
I have tested it the following way:

```
#include "BluetoothA2DPSink.h"
#include "AudioTools.h"

LogarithmicVolumeControl vc;

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

podaen, 2021-12-24:
I am curently not on my place right now. I will try to find out where it came from when I am back.

podaen, 2021-12-24:
Mine looks like this and still giving an error

```
//LibHelix
#include <MP3DecoderHelix.h>
#include <helix_pgm.h>
#include <helix_log.h>
#include <CommonHelix.h>
#include <AACDecoderHelix.h>

//audiotools
#include <AudioTools.h>
#include <AudioConfig.h>

//SdFat
#include <SdFat.h>

//spi
#include <SPI.h>

//A2DP
//#include <VolumeControl.h>
//#include <SoundData.h>
//#include <config.h>
//#include <BluetoothA2DPSource.h>
#include <BluetoothA2DPSink.h>
//#include <BluetoothA2DPCommon.h>

BluetoothA2DPSink a2dp_sink;
int8_t IntStat = 0;
int8_t IntConnected = 0;

//SD
const int SD_CS_PIN = SS;
#define SPI_SPEED SD_SCK_MHZ(4)
// SD_FAT_TYPE = 0 for SdFat/File as defined in SdFatConfig.h,
// 1 for FAT16/FAT32, 2 for exFAT, 3 for FAT16/FAT32 and exFAT.
#define SD_FAT_TYPE 1

#if SD_FAT_TYPE == 0
SdFat sd;
File dir;
File file;
#elif SD_FAT_TYPE == 1
SdFat32 sd;
File32 dir;
File32 file;
File32 DbFile;
//File32 TxtFileWrite;
#elif SD_FAT_TYPE == 2
SdExFat sd;
ExFile dir;
ExFile file;
#elif SD_FAT_TYPE == 3
SdFs sd;
FsFile dir;
FsFile file;
#else  // SD_FAT_TYPE
#error invalid SD_FAT_TYPE
#endif  // SD_FAT_TYPE
// Store error strings in flash to save RAM.
#define error(s) sd.errorHalt(&Serial, F(s))
const uint32_t MAX_FILE_SIZE_MiB = 100;  // 100 MiB file.
const uint32_t MAX_FILE_SIZE = MAX_FILE_SIZE_MiB << 20;
char seperator[2] = ";";
const int MaxTitleLength = 80;// 40;
const int TitleIndex = 1000;
// Geen files in de hoofd directory plaatsen anders wordt playlist niet goed aangemaakt "/" weet niet waarom
#undef _WIN32

#define USE_HELIX 
#define USE_SDFAT

using namespace audio_tools;

const char* startFilePath = "/";
const char* ext = "mp3";
AudioSourceSdFat source(startFilePath, ext, SD_CS_PIN, 16);

I2SStream i2s;
VolumeOutput volume(i2s);
LogarithmicVolumeControl lvc(0.1);
MP3DecoderHelix decoder;
//AudioPlayer player(source, i2s, decoder);
AudioPlayer player(source, volume, decoder);

void setup() {
}
void loop() {
}
```

pschatzmann, 2021-12-24:
As I was writing the sequence of the imports is important!
You have to move the #include <BluetoothA2DPSink.h> to the top...

podaen, 2021-12-24:
Yes, it's working now.

closed 2021-12-24 by podaen