pschatzmann/ESP32-A2DP #149: ESP32-I2S-A2DP SOURCE in 32 bits with masterclock

thanks-for-your-help, 2021-12-31:
Hello,

Thanks again for your greet work.

I'am trying to use BluetoothA2DPSource.h for making an I2S A2DP source from a 2 channels 32 bits I2S stream. 

I know 32 bits won't be the audio better, just my I2S stream work only with 32 bits frame. The ESP32 could be I2S master.

If a masterclock could be generated It could be nice too for ADC or transceiver who need it.

I plain to use this kind of skeleton : 

#include "BluetoothA2DPSource.h"
BluetoothA2DPSource a2dp_source;

// callback 
int32_t get_sound_data(Frame *data, int32_t len) {
    // I NEED HELP HERE !
    return len;
}

void setup() {
  a2dp_source.set_auto_reconnect (false);
  a2dp_source.start("MyMusic", get_sound_data);  
}

void loop() {
}

Could you complete the callback fonction according to my request ? I think It could be nice to add this example in the ESP32-A2DP library. That could help some peoples.

Thanks by advance.

Happy end of year and happy new year.

Best regards

pschatzmann, 2021-12-31:
I suggest that you have a look at my audio tools library: 
https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-i2s-a2dp

thanks-for-your-help, 2021-12-31:
I know this arduino-audio-tools example. 

Just I need the set_auto_reconnect option who seem not implemented in arduino-audio-tools yet. This fonction work fine in BluetoothA2DPSource.h

I let you time, you could answer with code in 2022 !

pschatzmann, 2021-12-31:
You have access to all the functionality. Just look at the documentation!

thanks-for-your-help, 2022-01-18:
I try this code with an ADC PCM1802 as slave, and with a SPDIF receiver WM8804 as slave too : 

/**
 * @file base-i2s-a2dp.ino
 * @author Phil Schatzmann
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/base-i2s-a2dp/README.md
 * 
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#define USE_A2DP
#include "AudioTools.h"
using namespace audio_tools;  
BluetoothA2DPSource a2dp_source;

#define SinkName  "AirPods"

   I2S<int32_t> i2s;

   ChannelConverter<int32_t> converter(&convertFrom32To16);
   ConverterFillLeftAndRight<int32_t> bothChannels;

   const size_t max_buffer_len = 1024;
   int32_t buffer[max_buffer_len][2];

   // callback used by A2DP to provide the sound data
   int32_t get_sound_data(Frame* data, int32_t len) {
   size_t req_len = min(max_buffer_len,(size_t) len);

   // the microphone provides data in int32_t -> we read it into the buffer of int32_t data
   size_t result_len = i2s.read(buffer, req_len);

   // we have data only in 1 channel but we want to fill both
   //bothChannels.convert(buffer, result_len);  // We use both channels
   
   // convert buffer to int16 for A2DP
   converter.convert(buffer, data, result_len);
   return result_len;
}

// Arduino Setup
void setup(void) {
  Serial.begin(115200);
  Serial.println("------------------------ init ----------------------------");
  
  // i2s configuration
  Serial.println("starting I2S...");
  static const i2s_config_t i2s_config = {
    .mode = static_cast<i2s_mode_t>(I2S_MODE_MASTER | I2S_MODE_RX),
    //.mode = static_cast<i2s_mode_t>(I2S_MODE_SLAVE | I2S_MODE_RX),
    .sample_rate = 44100,
    .bits_per_sample = I2S_BITS_PER_SAMPLE_32BIT,
    //.bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
    .communication_format = static_cast<i2s_comm_format_t>(I2S_COMM_FORMAT_I2S|I2S_COMM_FORMAT_I2S_MSB),
    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1, // default interrupt priority
    .dma_buf_count = 8,
    .dma_buf_len = 1000,
  };
  
  // i2s pinout
  static const i2s_pin_config_t pin_config = {
    .bck_io_num = 26,
    .ws_io_num = 25,
    .data_out_num = 22,
    .data_in_num = 17
  };
  
  // now configure i2s with constructed pinout and config
  i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);
  i2s_set_pin(I2S_NUM_0, &pin_config);
    
  // Assign masterclock pin on pin 0
  REG_WRITE(PIN_CTRL, 0xFF0);
  PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);

  // start the bluetooth
  Serial.println("starting A2DP...");
  a2dp_source.set_auto_reconnect(false); /// IMPORTANT
  a2dp_source.start(SinkName, get_sound_data); 
  esp_bredr_tx_power_set(ESP_PWR_LVL_P9,ESP_PWR_LVL_P9);// Enable max TX power
  Serial.println("Connecting to : " SinkName);
  
}

  // Arduino loop - repeated processing 
void loop() {

  while (!a2dp_source.is_connected()){
  Serial.println("disconnected");
  delay(1000);
  }

  while (a2dp_source.is_connected()){
  Serial.println("connected");
  delay(1000);
  }
    
}

In both case, I have a bad sound quality (very noisy). The ADC and SPDIF receiver work fine in slave mode connected in a DSP input. I check all the clocks frequency on scope and it's ok (same when connected to DSP).

Do you see somethink wrong on this code ? Could you check if quality is good with your ADC ? 

Thanks

pschatzmann, 2022-01-19:
Did you already try with I2S_COMM_FORMAT_I2S|I2S_COMM_FORMAT_I2S_LSB ?

thanks-for-your-help, 2022-01-19:
Yes, I tested with the 2 sources. The sound is even worse.

pschatzmann, 2022-01-19:
Another thought: Sometimes logging has a negative impact on the quality. 
Maybe it helps if you deactivate all logging.

thanks-for-your-help, 2022-01-19:
How doing that ?

closed 2022-04-20 by pschatzmann