pschatzmann/ESP32-A2DP #151: Error when compiling using 1.6.0

ignatio, 2022-01-02:
Arduino IDE 1.8.19 throws this error using 1.6.0:

```
C:\Users\xxxx\Documents\Arduino\libraries\ESP32-A2DP-1.6.0\src\BluetoothA2DPCommon.cpp: In member function 'virtual void BluetoothA2DPCommon::set_scan_mode_connectable(bool)':
C:\Users\xxxx\Documents\Arduino\libraries\ESP32-A2DP-1.6.0\src\BluetoothA2DPCommon.cpp:232:58: error: 'discoverability' was not declared in this scope
         if (esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, discoverability)!=ESP_OK) {
                                                          ^~~~~~~~~~~~~~~
exit status 1
Error compiling for board ESP32 Dev Module.
```

This happens even on the bt_music_receiver_simple example. I have an older version I was using and when I upgraded this happens on compile. Not sure why. Is it related to "set_on_connected2BT set_on_dis_connected2BT are obsolete! Please use set_on_audio_state_changed()"?

Thanks

ignatio, 2022-01-02:
I see this is a closed issue from #144 and #110 but the solution is to use the latest zip which as far as I can tell I am (1.6.0). Is there a specific version of esp in board manager I should be using? I'm on 2.0.2.

ignatio, 2022-01-02:
Seems to work on esp board version 1.0.6 but not 2.0.0. Closing issue.

closed 2022-01-02 by ignatio

pschatzmann, 2022-01-03:
Please stop posting issues for old releases and use the latest version of the A2DP library!