pschatzmann/ESP32-A2DP #152: Pairing to a new device using BluetoothA2DPSource

Millimeter94, 2022-01-02:
Hello
When I first tried out the BluetoothA2DPSource library I used a headset (Jabra Evolve 65) I had lying around and it worked at the first try.
Sadly this headset is not the device I want to use in the end. That would be a JBL Go 3. But I can't get it to connect. At first I thought it might be because the Go 3 uses Bluetooth 5.1 but it also doesn't pair with the older Go 1 wit BT 3.1 or in fact any other Bluetooth device i could get my hands on except for the Headset I used for my first tests.
Interestingly this headset still connects immediately no matter what I type into the code as the name of the target device.
Is the target name saved somewhere inside of the bluetooth module? If so how can I change it? If not, how can i convince platformio to use a new name?

pschatzmann, 2022-01-03:
https://github.com/pschatzmann/ESP32-A2DP/wiki/Auto-Reconnect

Millimeter94, 2022-01-12:
Thanks
I still don't completely understand why but it worked ;)

closed 2022-01-12 by Millimeter94