pschatzmann/ESP32-A2DP #153: Can I find an example of using set_avrc_metadata_attribute_mask?

mario44, 2022-01-03:
As in the subject. I would like to use set_avrc_metadata_attribute_mask but I haven't found any example.

pschatzmann, 2022-01-03:
Just look at the readme.
The complete list of values can be found here
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_avrc.html#_CPPv423esp_avrc_md_attr_mask_t

mario44, 2022-01-04:
Thank you for the fast reponse.
There is a lot of information on this page, but I would like an example of use. You wrote in the comments: set_avrc_metadata_attribute_mask method e.g if you just need the title and playing time you can call: set_avrc_metadata_attribute_mask (ESP_AVRC_MD_ATTR_TITLE | ESP_AVRC_MD_ATTR_PLAYING_TIME);
Could I have any example of this method being applied?

ignatio, 2022-01-06:
I do something like this:

```
void avrc_metadata_callback(uint8_t id, const uint8_t *text) {

  //use a switch statement to cycle through metadata tags and fill them to buffers
  switch (id) {
    case 0x1:
      Serial.printf("TITLE DETECTED %s\n", text);
    // Do some other things if you'd like
      break;
    case 0x2:
      Serial.printf("ARTIST DETECTED %s\n", text);
    // Do some other things if you'd like
      break;
    case 0x4:
      Serial.printf("ALBUM DETECTED %s\n", text);
    // Do some other things if you'd like
      break;
    case 0x20:
      Serial.printf("GENRE DETECTED %s\n", text);
    // Do some other things if you'd like
      break;
    case 0x40:
      Serial.printf("TIME DETECTED %s\n", text);
    // Do some other things if you'd like
      break;
 }
}

void setup() {

 //mask metadata so we only get what we need
  a2dp_sink.set_avrc_metadata_attribute_mask(ESP_AVRC_MD_ATTR_TITLE | ESP_AVRC_MD_ATTR_ARTIST | ESP_AVRC_MD_ATTR_ALBUM | ESP_AVRC_MD_ATTR_GENRE | ESP_AVRC_MD_ATTR_PLAYING_TIME);

  //callback method for getting metadata
  a2dp_sink.set_avrc_metadata_callback(avrc_metadata_callback);

  //set sink name and start it
  a2dp_sink.start("My Speaker");

}
```

No doubt I'm doing it wrong but it works for me.

mario44, 2022-01-06:
@ignatio Thank you very much for your example.
I am a beginner in this environment.
Now I understand how it works and I think it will help me a lot in the future in understanding other functions.
Have a nice day and best regards.

closed 2022-01-19 by pschatzmann