pschatzmann/ESP32-A2DP #160: Can't reconnect to ESP32 after reboot: Connection refused

richardklingler, 2022-01-22:
Tried this today with a ESP-A1S module (ES8388) and it works just fine :-)

Only thing:

I can connect to it from my Mac Mini and sue it as default audio output, but after reboot or power up, the connection is lost and after a while this message is displayed:

"Connection refused"

Only soliution is to power-cycle the ESP32 and it is selected as default audio output again...

Looks that the ESP32 keeps the previous connection active and denies further connection attempts...so only a reset helps...

Is there a callback mechanism to detect if audio stopped playing for a while?

cheers
richard

pschatzmann, 2022-01-22:
https://github.com/pschatzmann/ESP32-A2DP/wiki/Auto-Reconnect
It might be that the ESP32 is still trying to reconnect but it fails because the Mc Mini has been restarted. 
If you set the log level to verbose you might be able to follow what is heppening.

richardklingler, 2022-01-22:
Hmm..regardless what I try..I don't get any ESP_LOG output inside Arduino IDE...

And with a lower reconnect time also no difference:

  a2dp_sink.set_auto_reconnect(true, false, 20);

pschatzmann, 2022-01-22:
It's in Arduino Tools -> Core Debug Level: e.g. Verbose

richardklingler, 2022-01-22:
Then it must be not the official Arduino IDE as there is no such menu under "Tools":

[image]

pschatzmann, 2022-01-22:
That's strange: I usually work with the generic ESP32 Dev Module - I suggest to give that a try...

richardklingler, 2022-01-22:
Ah okay..seems not all ESP32 baords enable this and other menu points (o;

So this is the output when I do a restart:

22:07:01.211 -> _D][BluetoothA2DPSink.cpp:1124] ccall_app_a2d_callback(): ccall_app_a2d_callback
22:07:01.211 -> [D][BluetoothA2DPSink.cpp:904] app_a2d_callback(): app_a2d_callback
22:07:01.211 -> [D][BluetoothA2DPSink.cpp:908] app_a2d_callback(): app_a2d_callback ESP_A2D_CONNECTION_STATE_EVT
22:07:01.211 -> [D][BluetoothA2DPSink.cpp:310] app_work_dispatch(): app_work_dispatch event 0x0, param len 16
22:07:01.244 -> [D][BluetoothA2DPSink.cpp:342] app_send_msg(): app_send_msg
22:07:01.244 -> [D][]lueuettht2A2Sininkpcpp65130ppctall_aandrer(t: alp_tcs()h ndlll_asp_ 0_c, 0allba[k]
22:07:01.244 -> BDlu[tluthAotPSiDPS.ck.c368] 9p] aas_rcand_er(lbaakp_taapphan_le_c APbaSIG
22:07:01.244 -> [O][BDIStATth sDPS 1k
22:07:01.244 -> [D]:5lu] apt_r2_ctinkl.bapk(33]app_rcoct_dislback d()_AaRp_Co_C_diECTtcN_STATE_[BTuetoo[BluDtoitk.cDPS1nk.]pcca10] v_pdl_rk_devpa)chcca app_v_rk__a2datct evD][ lue,ootramDPen 12c
22:07:01.277 -> pD][6] etvotdl_a2PSenk():p:3_2]dlpp_d_ent mst 0
22:07:01.277 -> [D]_seuetmsgh
22:07:01.277 -> 2DPSink.cpp:540] av_hdl_a2d_evt(): av_hdl_a2d_evt ESP_A2D_CONNECTION_STATE_EVT
22:07:01.277 -> [D][BluetoothA2DPSink.cpp:641] handle_connection_state(): handle_connection_state evt 0
22:07:01.277 -> [I][BluetoothA2DPSink.cpp:646] handle_connection_state(): partner address: 14:98:77:6d:8c:fc
22:07:01.319 -> [I][BluetoothA2DPSink.cpp:654] handle_connection_state(): A2DP connection state: Disconnected, [14:98:77:6d:8c:fc]
22:07:01.319 -> [I][BluetoothA2DPSink.cpp:657] handle_connection_state(): ESP_A2D_CONNECTION_STATE_DISCONNECTED
22:07:01.319 -> [I][BluetoothA2DPSink.cpp:668] handle_connection_state(): i2s_stop
22:07:01.319 -> [I][BluetoothA2DPSink.cpp:676] handle_connection_state(): Connection try number: 0
22:07:01.319 -> [D][BluetoothA2DPCommon.cpp:177] connect_to_last_device(): connect_to_last_device
22:07:01.343 -> [D][BluetoothA2DPSink.cpp:365] app_tDs]_Bandtoo()A appitnsk_hpn:ler,] cgalx_a 0_0
d[I]llbuct()t Ac2DPSank._p2:d6c]llpb_ckask_h[nBlur(o:oahA_tasknkandpe904APa_p_a_dORKllISPkTCH apg_a1
22:07:01.343 -> [ca[Blbacoo
22:07:01.343 -> hADD][Blu.cpp:3h32Dapp_w.rkp:i0p]tappd(): applwack(): aapchedd
22:07:01.376 -> cDa[BlaekooShA_A2DinC.NNE:TIO9]SccTl__av
22:07:01.376 -> d[D]vBluetoo): 2Dallnkvchpl_avrc_eapp
22:07:01.376 -> [Dr[B_uitoottch(P:inp._wo:k6di aathh eavnc_0vt,(p: av_hdlnavr6
evt ][Bl e
22:07:01.376 -> o[thABlPetonthcpp:S4nk.app_7en] ms_hdl_avrcsent()s A
22:07:01.376 -> RC conn_state evt: state 0, [14:98:77:6d:8c:fc]
22:07:01.376 -> [D][BluetoothA2DPSink.cpp:365] app_task_handler(): app_task_handler, sig 0x1, 0x0
22:07:01.376 -> [I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH si⸮: 1
22:07:01.409 -> [D][BluetoothA2DPSink.cpp:333] app_work_dispatched(): app_work_dispatched
22:07:01.409 -> [D][BluetoothA2DPSink.cpp:1156] ccall_av_hdl_a2d_evt(): ccall_av_hdl_a2d_evt
22:07:01.409 -> [D][BluetoothA2DPSink.cpp:536] av_hdl_a2d_evt(): av_hdl_a2d_evt evt 0
22:07:01.409 -> [D][BluetoothA2DPSink.cpp:540] av_hdl_a2d_evt(): av_hdl_a2d_evt ESP_A2D_CONNECTION_STATE_EVT
22:07:01.409 -> [D][BluetoothA2DPSink.cpp:641] handle_connection_state(): handle_connection_state evt 0
22:07:01.442 -> [I][BluetoothA2DPSink.cpp:646] handle_connection_state(): partner address: 14:98:77:6d:8c:fc
22:07:01.442 -> [I][BluetoothA2DPSink.cpp:654] handle_connection_state(): A2DP connection state: Connecting, [14:98:77:6d:8c:fc]
22:07:01.442 -> [I][BluetoothA2DPSink.cpp:726] handle_connection_state(): ESP_A2D_CONNECTION_STATE_CONNECTING
22:07:06.441 -> [D][BluetoothA2DPSink.cpp:1124] ccall_app_a2d_callback(): ccall_app_a2d_callback
22:07:06.474 -> [D][BluetoothA2DPSink.cpp:904] app_a2d_callback(): app_a2d_callback
22:07:06.474 -> [D][BluetoothA2DPSink.cpp:908] app_a2d_callback(): app_a2d_callback ESP_A2D_CONNECTION_STATE_EVT
22:07:06.474 -> [D][BluetoothA2DPSink.cpp:310] app_work_dispatch(): app_work_dispatch event 0x0, param len 16
22:07:06.474 -> [D][BluetoothA2DPSink.cpp:342] app_send_msg(): app_send_msg
22:07:06.507 -> [D][BluetoothA2DPSink.cpp:365] app_task_handler(): app_task_handler, sig 0x1, 0x0
22:07:06.507 -> [I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
22:07:06.507 -> [D][BluetoothA2DPSink.cpp:333] app_work_dispatched(): app_work_dispatched
22:07:06.507 -> [D][BluetoothA2DPSink.cpp:1156] ccall_av_hdl_a2d_evt(): ccall_av_hdl_a2d_evt
22:07:06.507 -> [D][BluetoothA2DPSink.cpp:536] av_hdl_a2d_evt(): av_hdl_a2d_evt evt 0
22:07:06.540 -> [D][BluetoothA2DPSink.cpp:540] av_hdl_a2d_evt(): av_hdl_a2d_evt ESP_A2D_CONNECTION_STATE_EVT
22:07:06.540 -> [D][BluetoothA2DPSink.cpp:641] handle_connection_state(): handle_connection_state evt 0
22:07:06.540 -> [I][BluetoothA2DPSink.cpp:646] handle_connection_state(): partner address: 14:98:77:6d:8c:fc
22:07:06.540 -> [I][BluetoothA2DPSink.cpp:654] handle_connection_state(): A2DP connection state: Disconnected, [14:98:77:6d:8c:fc]
22:07:06.573 -> [I][BluetoothA2DPSink.cpp:657] handle_connection_state(): ESP_A2D_CONNECTION_STATE_DISCONNECTED
22:07:06.573 -> [I][BluetoothA2DPSink.cpp:668] handle_connection_state(): i2s_stop
22:07:06.573 -> [I][BluetoothA2DPCommon.cpp:255] set_scan_mode_connectable(): set_scan_mode_connectable true

richardklingler, 2022-01-22:
Ah this looks better when doing a restart:

  a2dp_sink.set_auto_reconnect(true, true, 20);

Not sure what the number of tries mean in terms of tries/seconds....

So I assume the number of tries has to be very high so it reconnects when the Mac is shutdown over night...

closed 2022-04-20 by pschatzmann