pschatzmann/ESP32-A2DP #161: a2dp_source triggers watchdog timer after 15-30 seconds

edutzu, 2022-01-24:
I was trying to test your example bt_music_sender and except for a few audible clicks during the streaming, it runs fine by default.
Then I tried to change the frequency to 8000 and after playing fine for 15-30 seconds, it triggers the error:

` ``
E (63081) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (63081) task_wdt:  - IDLE0 (CPU 0)
E (63081) task_wdt: Tasks currently running:
E (63081) task_wdt: CPU 0: BtA2dSourceT
E (63081) task_wdt: CPU 1: loopTask
E (63081) task_wdt: Aborting.
abort() was called at PC 0x40156ac0 on core 0
```
And that will reset the board.

According to the official docs ([https://docs.espressif.com/projects/esp-idf/en/v3.3.1/api-reference/system/wdts.html](url)):
This is a symptom of CPU starvation and is usually caused by a higher priority task looping without yielding to a lower-priority task thus starving the lower priority task from CPU time. 

I had tried to disable the watchgog timer but no luck:
```
#include "soc/rtc_wdt.h"

rtc_wdt_protect_off();
rtc_wdt_disable();
```

I was wondering if you're aware of this issue and maybe you could suggest a fix or some workaround?

pschatzmann, 2022-01-24:
I guess you are using a ESP32 Version > 1.0.6
Have you tried to add a delay in the loop() ? e.g. delay(100).

edutzu, 2022-01-24:
Yeah, pretty close -- my board is v1.1 -- I will definitely try some newer version!
I have tried `vTaskDelay(100)` -- it won't fail this way but that also affects the streaming -- signal gets interrupted as well..

pschatzmann, 2022-01-24:
And don't forget to deactivate the logging....
So just try to decrease the value until it works stable...

edutzu, 2022-01-24:
well I tried that as well but even with a value of 1 I still got a click after each frame...
now, it's even worse as I played around with menuconfig and now I don't get even a connection anymore... very filmsy board.
however, what's still interesting is the fact that trying to play frequencies up to 440 it kept on working without adding any delays (even though had sporadic clicks but never failed). Yet, when frequency was changed to a much higher value, ie. 8000  problems started...

pschatzmann, 2022-04-20:
I suggest to change the sine generate to an implementation which uses tables

closed 2022-04-20 by pschatzmann