pschatzmann/ESP32-A2DP #162: freertos/FreeRTOS.h: No such file or directory on macOS 12.1, Arduino IDE 1.8.19

richardklingler, 2022-01-24:
Morning (o;

Installed Arduino IDE on my macOS 12.1 machine and when trying to compile my example it throws:

/Users/me/Documents/Arduino/libraries/ESP32-A2DP/src/BluetoothA2DPCommon.h:25:10: fatal error: freertos/FreeRTOS.h: No such file or directory
 #include "freertos/FreeRTOS.h" // needed for ESP Arduino < 2.0

This is either with the esp32 1.0.6 or esp32-2.0.2 board package...

Even when removing those freertos includes it complains about the BT include files from platformio....
but the Arduino ESP32 board package doesn't install any platformio packages...

So somehow this is misleading in the Wiki:

"The current code is purely dependent on the ESP-IDF framework (which is also provided by the Arduino ESP32 core). There are no other dependencies and this includes the Arduino API"

pschatzmann, 2022-01-24:
Which board did you select in the Board Manager ? 
You would get this message if it you selected anything other then an ESP32...

richardklingler, 2022-01-24:
Ah bugger.....compiles now...though seems Arduino hasn't done any updates for M1 silicon and still uses python 2 .

closed 2022-01-24 by richardklingler