pschatzmann/ESP32-A2DP #163: Connection problem with iOS 15 and no playback

richardklingler, 2022-01-24:
When I disconnect my mac mini and want to add a connection from my iPhone 8 instead, a connection isn't possible at all...

I have to turn off Bluetooth on my mac mini or shut it down...
Then I am able to make a connection from my iPhone but playing back is not possible as the device isn't listed under the airdrop menu.

Under bluetooth settings it offers me to select what device type it is...like heaphone, speaker and so on.....
can this be set in the firmware?

Others not able to play back from an iOS device?

This is with the ESP32 -2.0.2 version as I use volume control from the source.

pschatzmann, 2022-01-24:
https://github.com/pschatzmann/ESP32-A2DP/wiki/Auto-Reconnect

If you plan to use it from different devices I would recommend to switch the automatic reconnect off or set the retry to a very low value

pschatzmann, 2022-01-24:
Did you switch the auto reconnect off ?

richardklingler, 2022-01-24:
Very confusing this auto reconnect....

So for macOS I need to turn it on and for iOS off.....

pschatzmann, 2022-01-24:
On the ESP32 you would deactivate it with  a2dp_sink.set_auto_reconnect(false, false, 0);

richardklingler, 2022-01-24:
Yes this works for iOS...but then for macOS it won't reconnect automatically on startup, which is the preferred method for a PC when the esp32 was selected as audio output...

Can I detect what device type is connected to the esp32?

RossiDanieleFW, 2022-03-11:
I have found the same issue with Android (autoconnect=false): if I try to turn off Bluetooth on smartphone without disconnect or unpair the device, turn on Bluetooth and retry to connect, Android warn "connection failed".
This happens also in other condition, randomly.
After disconnection by remote, I have also try to call "end" method followed by "start" method in order to clean memory or variables, but the device still not connect (until I do a hard reset).
It is like as the ESP32 memory is not perfectly clean and Bluetooth core refuse or ignore connection.
I am testing connection from various devices (Android, iPhone, Windows Laptop).

tomtomhoward, 2022-09-07:
I am finding the same thing - I am trying to build a device that will be used by multiple users, in a music practice room.
I am using ESP32 so I can create devices that are nameable for each room.

I have auto reconnect off, I am using an example sketch but I have changed the pins (for physical soldering the DAC to the pins directly) and added a callback to turn on/off the LED for connection status.

If I connect with iPhone, disconnect, then reconnect I either get silence, or the ESP32 hangs and the light stays on after the phone has disconnected.
With a laptop, the ESP32 regularly hangs on disconnection, especially if I turn the bluetooth off directly rather than disconnecting properly.

Is there a watchdog that could be checked to restart the ESP32 in case of error? Manually resetting the ESP32 fixes it each time.

dhdhdhdhdhdhhd, 2023-10-05:
c:\Users\LENOVO\Documents\Arduino\libraries\ESP32-A2DP\src\BluetoothA2DPSource.cpp:538:42: error: missing binary operator before token "("
 #if ESP_IDF_VERSION > ESP_IDF_VERSION_VAL(4, 4, 4)
                                          ^

exit status 1

Compilation error: exit status 1

How to solve this error?

mrhempman69, 2023-10-18:
See also: https://github.com/pschatzmann/ESP32-A2DP/discussions/495

pschatzmann, 2026-04-15:
See https://github.com/pschatzmann/ESP32-A2DP/issues/533

closed 2026-04-15 by pschatzmann