pschatzmann/ESP32-A2DP #164: Exception was unhandled

Findus76, 2022-01-31:
Hello

Thank you for your work with these drivers.
This is the first project with ESP32, BT. So the chances of me making a mistake are high.
I get the error message "Exeption was unhalded". See log.
Run code on "ESP32 WROOM-32" with I2S, has re-mapped the pins from the default setting (WS and BCK have changed places) 
Has reportioned Flash to fit the code.

SW versions
PACKAGES:
  - framework-arduinoespressif32 3.10006.210326 (1.0.6)
  - tool-esptoolpy 1.30100.210531 (3.1.0)
  - tool-openocd-esp32 2.1000.20210721 (10.0)
  - toolchain-xtensa32 2.50200.97 (5.2.0)

Espressif 32 3.5.0

PlatformIO
Core 5.2.2
Home 3.4.0

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
▒UIzxfc980;3>"#%
Running system on ESP32
CPU revision: 1
No of core: 2
Chip has Bluetooth LE
Chip has Bluetooth Classic
Mount filesystem spiffs
Function file up to date
Ok CompresFunctionFile
Checking function string
Function write file up to date
Default parameters loaded
Scan OneWire for devices...
Unit 1, address: {0x28, 0xFF, 0x86, 0x5C, 0x52, 0x16, 0x03, 0x45} is of type DS18xx family
1 devices found
Init DS18B20 device
SDA pin: 15
SCL pin: 23
Scanning...
I2C device found at address 0x20
Found: 1 unit on bus
[D][BluetoothA2DPSink.cpp:123] start(): start
[I][BluetoothA2DPCommon.cpp:207] log_free_heap(): Available Heap: 201980
[I][BluetoothA2DPSink.cpp:136] start(): Device name will be set to 'MyMusic'
[D][BluetoothA2DPSink.cpp:1006] init_nvs(): init_nvs
[D][BluetoothA2DPCommon.cpp:113] get_last_connection(): get_last_connection
[I][BluetoothA2DPCommon.cpp:127] get_last_connection(): nvs_blob does not exist
[D][BluetoothA2DPCommon.cpp:137] get_last_connection(): => 00:00:00:00:00:00
[I][BluetoothA2DPSink.cpp:184] init_i2s(): init_i2s
[I][BluetoothA2DPSink.cpp:186] init_i2s(): init_i2s is_i2s_output
[D][BluetoothA2DPSink.cpp:266] init_bluetooth(): init_bluetooth
[I][BluetoothA2DPSink.cpp:271] init_bluetooth(): controller initialized
[I][BluetoothA2DPSink.cpp:280] init_bluetooth(): bluedroid initialized
[I][BluetoothA2DPSink.cpp:289] init_bluetooth(): bluedroid enabled
[D][BluetoothA2DPSink.cpp:387] app_task_start_up(): app_task_start_up
[D][BluetoothA2DPSink.cpp:310] app_work_dispatch(): app_work_dispatch event 0x0, param len 0
[D][BluetoothA2DPSink.cpp:342] app_send_msg(): app_send_msg
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40092416  PS      : 0x00060b33  A0      : 0x80090d60  A1      : 0x3ffce030
A2      : 0x0000ff4c  A3      : 0x0000abab  A4      : 0xb33fffff  A5      : 0x00000001
A6      : 0x00060b20  A7      : 0x0000cdcd  A8      : 0x0000abab  A9      : 0x3ffce030
A10     : 0x00000003  A11     : 0x00060b23  A12     : 0x00060b20  A13     : 0x00000001
A14     : 0x00060d20  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001d
EXCVADDR: 0x0000ff4c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd

ELF file SHA256: 0000000000000000

Backtrace: 0x40092416:0x3ffce030 0x40090d5d:0x3ffce060 0x400d7d07:0x3ffce0a0 0x400d6fea:0x3ffce0d0 0x400d6f25:0x3ffce110 0x400dc95d:0x3ffce150 0x400e155d:0x3ffce290 0x400e488e:0x3ffce300 0x400913fa:0x3ffce320

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
UIzxfc980;3>"#%
Running system on ESP32
CPU revision: 1
No of core: 2
Chip has Bluetooth LE
Chip has Bluetooth Classic
Reset cause: software reset due to exception/panic, entring fallback mode
Init system fail

Brg
Findus

pschatzmann, 2022-02-01:
I was just testing the bt_music_receiver_simple sketch with a Wrover Module myself and could not reproduce your issue. 
I used the following platform.ino

```
[platformio]
description = A2DP Example
default_envs = esp32dev

[env:esp32dev]
platform = espressif32
board = esp-wrover-kit
framework = arduino
lib_deps =  https://github.com/pschatzmann/ESP32-A2DP.git
build_flags = -Wno-deprecated  -DCORE_DEBUG_LEVEL=5 
monitor_speed = 115200
```

I would really need a stacktrace from you to know where you are crashing...

Findus76, 2022-02-01:
Not sure how to get stack trace.
Have only set the log that comes in case of crash, which was attached.
Core 1 resiter dump and Backtrace.

I will try to run the example as it is, without other applications running. 

Brg
Findus

Findus76, 2022-02-02:
If I run the example "bt_music_receiver_simple.ino" and change pin for I2S, everything seems to work. I can connect to the module, I can stream data.
`"#include <Arduino.h>

/ *
  Streaming Music from Bluetooth
  
  Copyright (C) 2020 Phil Schatzmann
  This program is free software: you can redistribute it and / or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.
  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
  GNU General Public License for more details.
  You should have received a copy of the GNU General Public License
  along with this program. If not, see <http://www.gnu.org/licenses/>.
* /

// ==> Example A2DP Receiver which uses I2S to an external DAC

#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;

void setup () {
 
  i2s_pin_config_t my_pin_config = {
.bck_io_num = 25,
.ws_io_num = 26,
.data_out_num = 22,
.data_in_num = I2S_PIN_NO_CHANGE
};
a2dp_sink.set_pin_config (my_pin_config);
a2dp_sink.start ("MyMusic");

}

void loop () {
  delay (1000); // do nothing
} "`

But if I run the code in my application
void setup ()
{

const char FailSafeString [] = "UIzxfc980; 3> \" #% ";
char RXbuff [16];

Serial.begin (115200); // initialize serial

Serial.flush ();
Serial.write (FailSafeString);
delay (50);
if (Serial.available ()> = 15)
{
Serial.readBytes (RXbuff, 15);
if (memcmp (FailSafeString, RXbuff, 15) == 0)
HaltBoot = true;
}
Serial.println ("");

SystemBusy = false;
Fallback = false;

BluetoothA2DPSink a2dp_sink;
i2s_pin_config_t my_pin_config = {
.bck_io_num = 25,
.ws_io_num = 26,
.data_out_num = 22,
.data_in_num = I2S_PIN_NO_CHANGE
};
a2dp_sink.set_pin_config (my_pin_config);
a2dp_sink.start ("MyMusic");

if (System.Init (& MQTTclient) &&! HaltBoot)
{
if (Function.Init_BeforeWifi (& MQTTclient))
{
Serial.println ("All functions initialized at low level");
}
else
{// else Put the unit in fallback and send alert!
Fallback = true;
HaltBoot = true;
}
}
else
{
HaltBoot = true;
Serial.println ("Init system fail");
}
"
Then I get the result
Rebooting ...
ets Jun 8 2016 00:22:57

rst: 0xc (SW_CPU_RESET), boot: 0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP: 0xee
clk_drv: 0x00, q_drv: 0x00, d_drv: 0x00, cs0_drv: 0x00, hd_drv: 0x00, wp_drv: 0x00
mode: DIO, clock div: 2
load: 0x3fff0018, len: 4
load: 0x3fff001c, len: 1044
load: 0x40078000, len: 10124
load: 0x40080400, len: 5828
entry 0x400806a8
UIzxfc980; 3> "#%
[D] [BluetoothA2DPSink.cpp: 123] start (): start
[I] [BluetoothA2DPCommon.cpp: 207] log_free_heap (): Available Heap: 206688
[I] [BluetoothA2DPSink.cpp: 136] start (): Device name will be set to 'MyMusic'
[D] [BluetoothA2DPSink.cpp: 1006] init_nvs (): init_nvs
[D] [BluetoothA2DPCommon.cpp: 113] get_last_connection (): get_last_connection
[D] [BluetoothA2DPCommon.cpp: 137] get_last_connection (): => 04: b4: 29: 56: c3: 86
[I] [BluetoothA2DPSink.cpp: 184] init_i2s (): init_i2s
[I] [BluetoothA2DPSink.cpp: 186] init_i2s (): init_i2s is_i2s_output
[D] [BluetoothA2DPSink.cpp: 266] init_bluetooth (): init_bluetooth
[I] [BluetoothA2DPSink.cpp: 271] init_bluetooth (): controller initialized
[I] [BluetoothA2DPSink.cpp: 280] init_bluetooth (): bluedroid initialized
[I] [BluetoothA2DPSink.cpp: 289] init_bluetooth (): bluedroid enabled
[D] [BluetoothA2DPSink.cpp: 387] app_task_start_up (): app_task_start_up
[D] [BluetoothA2DPSink.cpp: 310] app_work_dispatch (): app_work_dispatch event 0x0, param len 0
[D] [BluetoothA2DPSink.cpp: 342] app_send_msg (): app_send_msg
Guru Meditation Error: Core 1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC: 0x40092416 PS: 0x00060933 A0: 0x80090d60 A1: 0x3ffce0f0
A2: 0xa5a5a5f1 A3: 0x0000abab A4: 0xb33fffff A5: 0x00000001
A6: 0x00060920 A7: 0x0000cdcd A8: 0x0000abab A9: 0x3ffce0f0
A10: 0x00000003 A11: 0x00060923 A12: 0x00060920 A13: 0x00000001
A14: 0x00060b20 A15: 0x00000000 SAR: 0x00000004 EXCCAUSE: 0x0000001d
EXCVADDR: 0xa5a5a5f1 LBEG: 0x400014fd LEND: 0x4000150d LCOUNT: 0xfffffffd

ELF file SHA256: 0000000000000000000

Backtrace: 0x40092416: 0x3ffce0f0 0x40090d5d: 0x3ffce120 0x400d7d07: 0x3ffce160 0x400d6fea: 0x3ffce190 0x400d6f25: 0x3ffce1d0 0x400e1371: 0x3ffce990xxxxx3

Rebooting ...
ets Jun 8 2016 00:22:57

What I can see in the log is the same up to "[D] [BluetoothA2DPSink.cpp: 387] app_task_start_up (): app_task_start_up"
Then it is not the same.
Guess it fails to start application "task_start_up"?

Brg
Findus

pschatzmann, 2022-02-03:
You really need to decode the stacktrace to analyze the issue. Unfortunately I rarely use PlatformIO, so I can't help you there. In Arduino we have the following tool: https://github.com/me-no-dev/EspExceptionDecoder

Anyhow: what you try to do will not work and I am closing your issue as duplicate to [issue 19](https://github.com/pschatzmann/ESP32-A2DP/issues/19). 

Please feel free to reopen the issue if the stacktrace points to some improper error handling in my code.

closed 2022-02-03 by pschatzmann

joba-1, 2022-02-03:
@Findus76 I recently had the need to decode a stacktrace with platformio and found out it is pretty simple:
 
Add "monitor_filters = esp32_exception_decoder" to your platformio.ini environment like this:

```
[env:esp32]
platform = espressif32
...
monitor_filters = esp32_exception_decoder
...

```
Then, if you use the platformio device monitor (e.g. in vscode) you get Backtrace lines like this on exceptions:

```
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d2711  PS      : 0x00060030  A0      : 0x800d5367  A1      : 0x3ffc8fa0  
A2      : 0x00000002  A3      : 0x00000000  A4      : 0x00000020  A5      : 0x80000020  
A6      : 0x00000003  A7      : 0x00060a23  A8      : 0x00000001  A9      : 0xffffffff  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffc3cd4  
A14     : 0x00000003  A15     : 0x00000000  SAR     : 0x0000001e  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d2711:0x3ffc8fa0 0x400d5364:0x3ffc8fc0 0x400901ae:0x3ffc8fe0
  #0  0x400d2711:0x3ffc8fa0 in loop() at .pio/libdeps/esp32-aat/audio-tools/src/AudioTools/AudioStreams.h:62
  #1  0x400d5364:0x3ffc8fc0 in loopTask(void*) at /home/joachim/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:23
  #2  0x400901ae:0x3ffc8fe0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...

```
Pretty neat :)

Findus76, 2022-02-03:
Thanks!
Case no. 19 was not fun, that it is not possible or very difficult to get WiFi and BT to work at the same time. Debris...
May think of a few different ways around the problem.
Many thanks for the PlatformIO tip!
Not the first time I get an exception ...

Brg Findus