pschatzmann/ESP32-A2DP #168: Soft clicks generated by a2dp to spdif chain

joba-1, 2022-02-07:
Hi Phil,

now that I got this great new feature of spdif output I did some experiments and unfortunately found clicks where there should be none.

Setting is: basic-a2dp-spdif.ino with different buffer sizes and cpu speeds. These make no difference.

- The clicks are not very prominent, but clearly audible. 
- I tested different songs and rechecked that the originals do not have these clicks.
- I tested them with bt earphones to be sure it is not the phone's a2dp stack: no clicks
- I rewrote the data buffer in read_data_stream() with zeros: total silence
  Rewrote the buffer with square waves (100% and 50% amplitude): no clicks  -> seems it is not the new spdif code
- Recorded the spdif out via optical in of a soundcard -> click is visible

It is not a big jump in the wave, like I would expect from buffer overflows or if buffers are not appended seamlessly.
It is more a burst of high frequency waves - looks to me more like a decoding error.

Here is a short sample:

[image]

[tr-01-click.wav.gz](https://github.com/pschatzmann/ESP32-A2DP/files/8014008/tr-01-click.wav.gz)

These clicks do not happen all the time and not in regular intervals. In general maybe each minute or so.
I hope you have an idea about how to look into this further.

joba-1, 2022-02-07:
ok, I hoped I'm not the first one with this problem and there already is a "quick fix". Then I'll go on testing.

yes, I have checked different buffer sizes and count. Makes no difference.

I also made more tests and have discovered, that the first song is always flawless, the second (in a playlist or the same one in auto-repeat) starts having these hickups. 

I will try the delays. loop() is already empty, except for a delay(100) though.

pschatzmann, 2022-02-07:
If I remember right you run it with a decreased processor speed ?  Could this be it ?

joba-1, 2022-02-07:
I am back at 240MHz already. Delay(5) in the callback also does not help. Going up to 10ms

joba-1, 2022-02-07:
The stream reader callback is called often enough with a big enough buffer to provide continuous data.

It would have been strange if this callbacks timing would be the culprit, since then I would expect a clean cut of the sine wave. But the glitch always consists of ~30 samples and those change values in a continuous sine like fashion - just a bit irregular and with higher frequency.
 
Need to dig deeper in the a2dp chain.

joba-1, 2022-02-08:
maybe I really should reconsider :)

In the meantime I tried to use a separate spdif task and what I saw hints at something different: 
I get queue full errors with small queues or out of memory errors when trying to allocate space for the sample data.

Bluetooth sends data at some fixed rate we cannot influence.
i2s also sends at a fixed rate we set up and it should be **exactly** the same as the bt rate. 
If the bluetooth stream is a bit faster, it will overflow the buffers at some point.
If it is slower, the i2s stream will run out of data.

I doubt the clocks of my phone and the ESP are synchronized exactly, so how does this work?
Are the buffers just big enough that it will not be a problem until a song ends? What about very long songs? Or 10h long youtube videos?

Not sure how to solve this in a clean fashion, i.e without dropping or doubling samples now and then.

pschatzmann, 2022-02-08:
The counter argument is that you have stated that changing the i2s buffer size does not have any impact.

If I remember right I am using the apll per default. I don't have any big hopes but you could also try
config.use_apll = false;
or
config.use_apll = true;
config.fixed_mclk = sample_rate * 256

joba-1, 2022-02-08:
maybe "not any impact" was to bold a statement. I always heard the clicks, but maybe less often - I didn't count them.

Need to read docs about apll and mclk - will surely try...

joba-1, 2022-02-08:
would you mind looking at my task code? 
It works just as good or bad as before, but maybe you see an immediate problem - multitasking is always a bit challenging :)

You can find it here in branch spdif-task: https://github.com/joba-1/arduino-audio-tools/commit/a8da200cee81d9be983b014fbcf1edd90f9f22c4

Maybe you want to even use (parts of) it as currently the app core does nothing while the sys core may be close to overload.

What it logs when playing a 1 minute sine wave mp3 (I started playback within ~1-5s after reboot, out of mem starts after ~15-20s and repeats every 4-5s):

```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I] basic-a2dp-spdif.ino : 101 - setup core: 1
[I][BluetoothA2DPCommon.cpp:207] log_free_heap(): Available Heap: 197212
[I][BluetoothA2DPSink.cpp:136] start(): Device name will be set to 'test-spdif'
[I][BluetoothA2DPSink.cpp:184] init_i2s(): init_i2s
[I][BluetoothA2DPSink.cpp:271] init_bluetooth(): controller initialized
[I][BluetoothA2DPSink.cpp:280] init_bluetooth(): bluedroid initialized
[I][BluetoothA2DPSink.cpp:289] init_bluetooth(): bluedroid enabled
[I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[I][BluetoothA2DPCommon.cpp:255] set_scan_mode_connectable(): set_scan_mode_connectable true
[I][BluetoothA2DPCommon.cpp:207] log_free_heap(): Available Heap: 82624
[I] I2SConfig.h : 66 - rx/tx mode: TX
[I] I2SConfig.h : 67 - port_no: 0
[I] I2SConfig.h : 68 - is_master: Master
[I] I2SConfig.h : 69 - sample rate: 88200
[I] I2SConfig.h : 70 - bits per sample: 32
[I] I2SConfig.h[I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[I][BluetoothA2DPSink.cpp:646] handle_connection_state(): partner address: a0:cb:fd:33:11:19
[I][BluetoothA2DPSink.cpp:654] handle_connection_state(): A2DP connection state: Connecting, [a0:cb:fd:33:11:19]
[I][BluetoothA2DPSink.cpp:726] handle_connection_state(): ESP_A2D_CONNECTION_STATE_CONNECTING
channels: 2
[I] I2SConfig.h : 72 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 75 - use_apll: true
[I] I2SConfig.h : 76 - fixed_mclk: 22579200
[I] I2SConfig.h : 80 - pin_bck: -1
[I] I2SConfig.h : 81 - pin_ws: -1
[I] I2SConfig.h : 82 - pin_data: 27
[I] I2SESP32.h : 107 - I2S_BUFFER_COUNT=16
[I] I2SESP32.h : 108 - I2S_BUFFER_SIZE=512
[I] I2SESP32.h : 109 - I2S_AUTO_CLEAR=1
[I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[I][BluetoothA2DPSink.cpp:646] handle_connection_state(): partner address: a0:cb:fd:33:11:19
[I][BluetoothA2DPSink.cpp:654] handle_connection_state(): A2DP connection state: Disconnected, [a0:cb:fd:33:11:19]
[I][BluetoothA2DPSink.cpp:657] handle_connection_state(): ESP_A2D_CONNECTION_STATE_DISCONNECTED
[I][BluetoothA2DPCommon.cpp:255] set_scan_mode_connectable(): set_scan_mode_connectable true
[I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[I][BluetoothA2DPSink.cpp:646] handle_connection_state(): partner address: a0:cb:fd:33:11:19
[I][BluetoothA2DPSink.cpp:654] handle_connection_state(): A2DP connection state: Connecting, [a0:cb:fd:33:11:19]
[I][BluetoothA2DPSink.cpp:726] handle_connection_state(): ESP_A2D_CONNECTION_STATE_CONNECTING
[I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[I][BluetoothA2DPSink.cpp:578] handle_audio_cfg(): a2dp audio_cfg_cb , codec type 0
[I][BluetoothA2DPSink.cpp:590] handle_audio_cfg(): a2dp audio_cfg_cb , sample_rate 44100
[I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[I][BluetoothA2DPSink.cpp:646] handle_connection_state(): partner address: a0:cb:fd:33:11:19
[I][BluetoothA2DPSink.cpp:654] handle_connection_state(): A2DP connection state: Connected, [a0:cb:fd:33:11:19]
[I][BluetoothA2DPSink.cpp:692] handle_connection_state(): ESP_A2D_CONNECTION_STATE_CONNECTED
[I][BluetoothA2DPCommon.cpp:255] set_scan_mode_connectable(): set_scan_mode_connectable false
[I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[I][BluetoothA2DPSink.cpp:768] av_hdl_avrc_evt(): AVRC conn_state evt: state 1, [a0:cb:fd:33:11:19]
[I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[... 6 lines cut ...]
2DP ID 2: 'joba-1'
[I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[I][BluetoothA2DPSink.cpp:792] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x4, sines
[I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[I][BluetoothA2DPSink.cpp:792] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x8, 1
[I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[I][BluetoothA2DPSink.cpp:792] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x10, 0

[I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[I][BluetoothA2DPSink.cpp:792] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x20, Test

[I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[... cut ...]
[E] basic-a2dp-spdif.ino : 45 - 21.939: No memory
[E] basic-a2dp-spdif.ino : 45 - 25.636: No memory
[E] basic-a2dp-spdif.ino : 45 - 30.619: No memory
[E] basic-a2dp-spdif.ino : 45 - 34.683: No memory
[E] basic-a2dp-spdif.ino : 45 - 38.804: No memory
[E] basic-a2dp-spdif.ino : 45 - 43.566: No memory
[E] basic-a2dp-spdif.ino : 45 - 47.176: No memory
[E] basic-a2dp-spdif.ino : 45 - 52.041: No memory
[E] basic-a2dp-spdif.ino : 45 - 56.178: No memory
[E] basic-a2dp-spdif.ino : 45 - 59.864: No memory

```

joba-1, 2022-02-08:
with apll = false, the out of memory errors are gone.
Unfortunately the glitches have increased substantially:

Every peak outside the blue band is a glitch:
[image]

But looking closer, they impact less samples. 
[image]

joba-1, 2022-02-08:
config.use_apll = true;
config.fixed_mclk = sample_rate * 256

the glitches now coincide with the no memory messages.
With the separate task they look different though, more like a clean cut - probably one missing buffer due to no mem)

[image]

pschatzmann, 2022-02-08:
Wouln't you be able to address the out of memory errors by decreasing the I2S buffers ?

joba-1, 2022-02-08:
not sure why, but will try...

joba-1, 2022-02-08:
Now the other side fails. Interesting...
```
[I][BluetoothA2DPSink.cpp:616] handle_audio_state(): A2DP audio state: Started
[E] basic-a2dp-spdif.ino : 40 - 30.321: Queue full
[E] basic-a2dp-spdif.ino : 40 - 34.938: Queue full
[E] basic-a2dp-spdif.ino : 40 - 39.116: Queue full
[E] basic-a2dp-spdif.ino : 40 - 42.482: Queue full

```

joba-1, 2022-02-08:
need to stop for today. 
And need to think about how this sync between phone and spdif clock should work. 
Any hints welcome :)

pschatzmann, 2022-02-08:
I dont have any details, but my understanding is that bluetooth determines the speed how the data is delivered and processed and adjusts the speed acordinglly. In the worst case the sample_rate is adjusted...

joba-1, 2022-02-08:
I changed xQueueSend to block instead of fail so if spdif output is not ready, the bt callback has to wait.
No more out of memory or full queue errors. 

Result: nearly the same as before: ~1 min 20s clean sound, then a glitch every 4-5 seconds
But the sample wave glitches look more like the first sine graphic. Not a cut but more like mending over several samples.

I could reduce CPU speed from 240 to 160MHz, reduce the queue size from 3 to 2 and I2S buffers from 16*512 to 4*128 and DEFAULT_BUFFER_SIZE from 3584 to 1024 (not tried smaller yet) with still the same result. These seem not to be relevant for this pattern.

I added counters for the bytes sent to the queue by the callback and the bytes retrieved by the spdif task and log them with milliseconds. Even after 40 min the log messages are completely in sync. Receive follows ~30ms after the send
```
                                 secs.ms   bytes
[W] basic-a2dp-spdif.ino :  50 - 1311.101: 229120000  <- send 
[W] basic-a2dp-spdif.ino : 101 - 1311.131: 229120000  <- recv

```
drives me mad :)

pschatzmann, 2022-02-09:
I think it would be interesting to see if we can reproduce this issue by switching to I2S and try with a sample rate of 44100 and 88200. If we can reproduce it at 88200 we could confirm that the issue is in the OS.

How are you getting the data into Audacity ? Is it with a logic analyzer ?
Did you already try with the latest Arduino ESP32 version 2.0.2 ?

joba-1, 2022-02-09:
> Did you already try with the latest Arduino ESP32 version 2.0.2 ?

The idf version the arduino core reports with "LOGI("IDF version: %s", ESP.getSdkVersion());" makes me dizzy

Compiled with Arduino IDE (using standard esp32 core):
[I] streams-generator-spdif.ino : 27 - IDF version: v3.3.5-1-g85c43024c

Compiled with platformio:
[I] basic-a2dp-spdif.ino : 123 - IDF version: v3.3.5-1-g85c43024c

And this is what the build says (and compared with their esp32 platform github repo, this is the latest):
PLATFORM: Espressif 32 (3.5.0) > MH ET LIVE ESP32MiniKit

AFAIK the esp32 arduino core uses ESP IDF 4.x, not 3.x???
Tasmota devices I recently updated report this with the same call to ESP.getSdkVersion(): v4.4-3-g6afb23d90a

Anyways, I got rid of Arduino and retested amendes version with latest idf V5 again and it has exactly the same symptom: ~1min 10s all good, then glitches start every 4-5s.

There I got this message (I believe from the ESP core libs?) for each glitch:

W (108004) BT_APPL: Pkt dropped

> How are you getting the data into Audacity ? Is it with a logic analyzer ?
Plugged toslink into soundcard input and directly recorded it with Audacity (see settings in top left of the window).
Logic analyzer would be difficult, because I would only get numbers from the protocol decoder - hard to see glitches there.

Will try i2s without spdif later. 1/2 bit frequency might really change things...

I'm currently quite sure that this is an ESP32 or ESP-IDF internal thing - not arduino-audio-tools or esp32-a2dp, not even arduino core.

pschatzmann, 2022-02-09:
In Arduino -> Tools -> Board -> Board Manager
Enter ESP32 then you get esp32 by Espressif Systems version 2.0.2 installed.
If 1.0.6 is shown instead you are still on an old version

joba-1, 2022-02-09:
I'm on the old version 1.0.6. But it is the latest shown to me in the version list

Board URL I used is https://dl.espressif.com/dl/package_esp32_index.json. Can't remember where I got this from a few days ago (other than google), but espressif.com as source looked convincing :(

Now trying to find out what to use for platformio.
```
platform = espressif32
;platform = espressif32@3.5.0
;platform = https://github.com/platformio/platform-espressif32/archive/refs/tags/v3.5.0.zip
framework = arduino
```
these also look right, but seem to give the same old version as arduinos 1.0.6...
I already purged ~/Arduino and ~/.platformio folders to dump any old versions I might have had.

joba-1, 2022-02-09:
Much better with your board json :)
  
[I] streams-generator-spdif.ino : 27 - IDF version: v4.4-beta1-189-ga79dc75f0a

Not sure why it says beta1 for version 4.4 when there is an idf 5.x out already, but what do I know!

Now on to fixing it for platformio...

pschatzmann, 2022-02-09:
I was wondering , is your "better" referring to the success to have a new version or the audio issue resolved ?"""

joba-1, 2022-02-09:
unfortunately only related to having the new version.

I tried hard to get it to work with platformio. Even reached out to the tasmota community because I knew they use arduino core 2.0.2 based on idf 4.4. With their help I could get it to compile, but it is not stable: serial out gets stuck after several lines. BT connect and play works anyways, but spdif out has the same 4/5s glitches after ~1min. Hard to tell why without serial out.
```
[I] basic-a2dp-spdif.ino : 128 - IDF version: v4.4-2.0.2-267-g9f781b4633
[I] I2SConfig.h : 66 - rx/tx mode: TX
[I] I2SConfig.h : 67 - port_no: 0
[I] I2SConfig.h : 68 - is_master: Master
[I] I2SConfig.h : 69 - sample rate: 88200
[I] I2SConfig.h : 70 - bits per sample: 32
[I] I2SConfig.h : 71 - number of channels: 2
[I] I2SConfig.h : 72 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 75 - use_apll: true
[I] I2SConfig.h : 76 - fixed_mclk: 22579200
[I] I2SConfig.h : 80 - pin_bck: -1
[I] I2SConfig.h : 81 - pin_ws: -1
[I] I2SConfig.h : 82 - pin_data: 27
[I] I2SESP32.h : 107 - I2S_BUFFE  <-- stuck here
```
So I tried to use Arduino IDE with my modified example: also stuck serial and same glitches.

Since it also shows up with amendes spdif version and IDF V5.x I think it is best to bother espressif with this.

But if you want to try different i2s configs, I will do that. Maybe mono? Not sure if that is possible with spdif.

pschatzmann, 2022-02-09:
I think it would be interesting to see to reproduce the issue with I2S and a DAC with a sample rate of 88200.
I just double checked if I can use my AI thinker audiokit - but this does not support such high rates, so I would really need to take the time to plug a solution together

joba-1, 2022-02-09:
Any additional info is good info :)

I just opened an issue here: https://github.com/espressif/esp-idf/issues/8368

In my test I will try to get rid of I2S/DAC entirely and see if the A2DP drops happen even if i send the data to /dev/nul

joba-1, 2022-02-09:
Sorry, no surprise for you. 

No i2s_write(), no packet drops.

Even if I send 100 bytes less than the callback delivered, all is well.

If I send only 10 bytes less, package drops are back.

joba-1, 2022-02-10:
Tried to rule out that spdif encoding takes too long when done serially between i2s.write() calls.

* Replaced the i2s.write() with copying the prepared spdif_buffer and putting the address on a queue. 
* Added task that reads a buffer address from a queue and writes the buffer to I2S as soon as available.

https://github.com/joba-1/arduino-audio-tools/commit/428fc2d27c44a8f0783092b29eb8a74baaa38181

Unfortunately the same result: Glitches after ~1min.
I used app core (1). For completeness I'll also try core 0 later...

joba-1, 2022-02-11:
Using two tasks now, but still no joy.

One task takes the data from the a2dp callback and converts to spdif and one task just does the i2s.writes():
https://github.com/joba-1/arduino-audio-tools/commit/97d3ba4995d430d9ef332266016971073a2840b5

Tested with tasks on same or different cores with smaller or bigger buffers and full CPU speed. No change.

I am sure that the data they receive from the a2dp callback gets processed in time and properly. The glitches are already there from BT packet drops.
The hope was that the 2 tasks allow the esp internal bt stack task to not drop packages, because there should be enough idle time while these tasks wait in the queues for more data. Didn't work.
Looks like it breaks an esp internal limit to operate a 44100Hz a2dp and a 88200Hz I2S stream continuously :(

Not sure how I could test the theory that less I2S sample frequency could work. And what would that help in the end...
Do you have a favorite DAC from the list on your site that you would recommend over the others?

pschatzmann, 2022-02-11:
The first on the list was the most simple one. 
I was struggling today with the topic today because I couldn't figure out a way to test this with my local hardware.
Now I just realize that I was thinking too complicated and that I don't need to record anything: A simple hearing test should be good enough!
I will try tomorrow....

pschatzmann, 2022-02-11:
I was just running the test with [my sine generator](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-generator-i2s) first at 44100 and then at 88200 samples per second.

I could not hear any difference and did not notice any of your issues....

joba-1, 2022-02-11:
Not so strange I think. Local generated sine has always worked for me, too.

pschatzmann, 2022-02-11:
What do you mean with local generated ? How should I try to reproduce the issue ?

joba-1, 2022-02-11:
I generated a sine mp3, put it on my phone and send it via a2dp to the esp. Always works fine for 1-1,5min then problems start.

pschatzmann, 2022-02-11:
OK...I would not trust mp3....I will try a test tone with A2DP tomorrow..

joba-1, 2022-02-12:
A DAC is on its way to me. Should be here next week...

pschatzmann, 2022-02-12:
Cool....on my side I realized that the testing with 88200 is not so easy because my framework does not provide any upsampling functionality yet.

pschatzmann, 2022-02-12:
My resampling logic is finally working: Here is the [test](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/sandbox/basic-a2dp-i2s) that used.

I can hear some clicks every 3-4 seconds with a sampling rate of 88200. So it is really the combination of A2DP and I2S speed which leads to the issue.

joba-1, 2022-02-13:
will try ASAP and let you know. 

I also contacted amendes earlier. He responded and doubled the ringbuffer between bt callback and i2s writes and throttles writing to it. Some options to test...

joba-1, 2022-02-13:
just for laughs: 

tested with only I2S buffers changed to 30/384 and recorded with audacity:

[image]

just at the time where usually the problems began my phone decided to send me reminder pings :D

pschatzmann, 2022-02-13:
Hmm, strange...

joba-1, 2022-02-14:
I tried iPhone12 mini now, instead of old S7 and Tried your online sine generator (never ocurred to me I could use a BT dongle on the linux pc).

No change with amedes version. I used it because checking for the "Pkt dropped" messages is much easier than recording the audio with audacity and then search for wave form irregularities. 
Do you know if/how I can enable these IDF BT messages with your code as well?

Anyways, will also try your latest code now...

pschatzmann, 2022-02-14:
Can you past some of these messages ?
In Platformio you can set the ESP Log level with 
build_flags = -DCORE_DEBUG_LEVEL=5

joba-1, 2022-02-14:
same with your code and iphone or web site.

these are the messages. They are not there for the first minute while audio is good, but then below is the general pattern

```
...
W (566496) BT_APPL: Pkt dropped

I (567526) BT_AV: Audio packet count 5300
I (569856) BT_AV: Audio packet count 5400
I (572176) BT_AV: Audio packet count 5500
W (573766) BT_APPL: Pkt dropped

I (574516) BT_AV: Audio packet count 5600
I (576846) BT_AV: Audio packet count 5700
W (578516) BT_APPL: Pkt dropped

I (579186) BT_AV: Audio packet count 5800
I (581506) BT_AV: Audio packet count 5900
I (583836) BT_AV: Audio packet count 6000
W (585416) BT_APPL: Pkt dropped

I (586176) BT_AV: Audio packet count 6100
I (588506) BT_AV: Audio packet count 6200
I (590836) BT_AV: Audio packet count 6300
I (593156) BT_AV: Audio packet count 6400
I (595496) BT_AV: Audio packet count 6500
W (595536) BT_APPL: Pkt dropped

I (597826) BT_AV: Audio packet count 6600
I (600156) BT_AV: Audio packet count 6700
W (601416) BT_APPL: Pkt dropped

```

pschatzmann, 2022-02-14:
Hmm, I would think that this is happening if the buffers are full and A2DP can not post the data in any reasonable time. 
Did you try to play with the sampe_rate settings by adjusting the values a bit ?

joba-1, 2022-02-14:
I use debug level 5 now (and removed the logs from audio_data_callback()). Those messages dont appear. I guess I would need to enable them while compiling the arduino core with custom settings. 

Did not adjust sampling rate yet.
You mean like changing
  cfg.sample_rate = a2dp_sink.sample_rate();
to
  cfg.sample_rate = (a2dp_sink.sample_rate() * 1000) / 1005;
or so?

joba-1, 2022-02-14:
cfg.sample_rate = (a2dp_sink.sample_rate() * 1000) / 1005; // no change
cfg.sample_rate = (a2dp_sink.sample_rate() * 1000) / 995; // no change

joba-1, 2022-03-04:
FYI:
I continued my tests with the amedes code because I see the bt drop pkg message there.
In the original code there is a circular buffer that drops a single frame per callback when it gets too full. 
I need to drop up to 4% of the frames in my code to avoid the bt packet drops.
Maybe my ESP has a slow oscillator so that it thinks it does 44.1kHz but it is 4% less

joba-1, 2022-03-06:
for reference, this is my current rate limiting code.

Works as follows:
Bluetooth task callbacks put samples in a ringbuffer. 
The spdif/i2s task fetches the samples from the buffer, converts them to spdif and sends them to i2s in the required speed.
If the ringbuffer is full because the spdif task fetches the data slower, this blocks the bluetooth task and leads to dropped packages. 
It is not possible to slow down the bluetooth device and it is not possible to increase the speed of i2s to **exactly** match. 
The codec chips used in DACs usually solve this by resampling the buffer to generate more or less samples. 
The existing code did that by duplicating or throwing away one sample per buffer. 
That was not enough. I implemented code that adapts the number of samples dropped to the percentage of used buffer space (and evenly distributed). I end up throwing away ~10 samples per 2.5k buffer.

```
#define RINGBUF_SIZE (48 * 1024)
#define AUDIO_SAMPLE_SIZE (16 * 2 / 8) // 16bit, 2ch, 8bit/byte

size_t write_ringbuf(const uint8_t *data, size_t size)
{
    // advanced rate control:
    // * if buffer is 40% to 60% full: no action
    // * if buffer is less than 40% full, add samples up to 10% proportionally to decreasing buffer level
    // * if buffer is more than 60% full, drop samples up to 10% proportionally to increasing buffer level
    // ideas for future: quad interpolation over prev/curr/next sample or resample over whole buffer?

    UBaseType_t items;
    vRingbufferGetInfo(s_ringbuf_i2s, NULL, NULL, NULL, NULL, &items);
    items *= 100;
    items /= RINGBUF_SIZE;  // items in percent
    if( items < 40 ) {
        // duplicate first sample
        // TODO use slices to add more samples
        xRingbufferSend(s_ringbuf_i2s, (void *)data, AUDIO_SAMPLE_SIZE, (portTickType)portMAX_DELAY);
    }
    else if( items > 60 ) {
        UBaseType_t samples = size / AUDIO_SAMPLE_SIZE;          // number of samples in data 
        UBaseType_t drop_rate = (items - 60) * 1000 / 40;        // drop ratio (0-10% * 100): 0-1000
        UBaseType_t drop_samples = samples * drop_rate / 10000;  // number of samples to drop (or slices)
        UBaseType_t drop_step = samples / drop_samples;          // drop every drop_step sample from data
        UBaseType_t drop_size = drop_step*AUDIO_SAMPLE_SIZE;     // chunk size between drops in bytes

        for( size_t step = 1; step < drop_samples; step++ ) {
            const uint8_t *slice_start = &data[step*drop_size + step*AUDIO_SAMPLE_SIZE];            
            uint8_t *slice_target = (uint8_t *)&data[step*drop_size];
            if( slice_start + drop_size >= data + size ) {
                // cut down last chunk
                drop_size = data + size - slice_start - AUDIO_SAMPLE_SIZE;
            }

            // ESP_LOGW(BT_APP_CORE_TAG, "%s, items=%u, drops=%u, size=%u, chunk=%u, src=%u, dst=%u", 
            //     __func__, items, drop_samples, size, drop_size, slice_start-data, slice_target-data);

            memmove(slice_target, slice_start, drop_size);
        }
        size -= drop_samples*AUDIO_SAMPLE_SIZE;
        ESP_LOGW(BT_APP_CORE_TAG, "%s, items=%u, drops=%u, size=%u", __func__, items, drop_samples, size);
    }

    BaseType_t done = xRingbufferSend(s_ringbuf_i2s, (void *)data, size, (portTickType)portMAX_DELAY);
    if(done){
        return size;
    } else {
        return 0;
    }
}

```
It basically works, but leads to frequent esp restarts and so far I can't find a reason for that. 
Backtraces look not very helpful to me:

```
W (72675) BT_APP_CORE: write_ringbuf, items=62, drops=3, size=2548
W (72675) BT_APP_CORE: write_ringbuf, items=67, drops=11, size=2516
W (72685) BT_APP_CORE: write_ringbuf, items=71, drops=17, size=2492
W (72685) BT_APP_CORE: write_ringbuf, items=76, drops=25, size=2460
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4010ea78  PS      : 0x00060a30  A0      : 0x8010f103  A1      : 0x3ffcd780  
0x4010ea78: r_rw_schedule at ??:?

A2      : 0x00000000  A3      : 0x3ffc6e98  A4      : 0x00000001  A5      : 0x3ff71030  
A6      : 0x400d7eb0  A7      : 0x3ffc6e90  A8      : 0x80083856  A9      : 0x3ffcd740  
0x400d7eb0: task_create_wrapper at /home/joachim/esp/esp-idf/components/bt/controller/esp32/bt.c:1004

A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffb1a64  A13     : 0x00000001  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000168  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x4010ea75:0x3ffcd7800x4010f100:0x3ffcd7a0 0x40093ca1:0x3ffcd7d0 
0x4010ea75: r_rw_schedule at ??:?

0x4010f100: btdm_controller_task at ??:?

0x40093ca1: vPortTaskWrapper at /home/joachim/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:141

ELF file SHA256: 0d3499e755d5c0b3

Rebooting...
```

Ideas?

pschatzmann, 2022-03-06:
I have a new class which implements resampling but it has not been desinged to handle these small differences. Maybe it’s worth if I extende it to hadle this case as well.
It’s strange that an incrlease of the sampling rate does not help here…

joba-1, 2022-03-06:
> I have a new class which implements resampling but it has not been desinged to handle these small differences. Maybe it’s worth if I extende it to hadle this case as well.

If you tell me it might work, I'd give it a try

> It’s strange that an incrlease of the sampling rate does not help here…

Maybe it does. I didn't try hard because I thought It would be difficult to sync it to +/- 1 sample per 2.5k

> if you past the stacktrace into the ESP Exception Decoder you should see the offending line which is causing the issue.

The above is already decoded - there is just not much info. And it always looks different - sometimes the decoder  even notices the stack is corrupted, like here:
```
W (69115) BT_APP_CORE: write_ringbuf, items=78, drops=28, size=2448
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008a5c8  PS      : 0x00060034  A0      : 0x80083518  A1      : 0x3ffc0c88  
0x4008a5c8: r_rwbtdm_isr_wrapper at intc.c:?

A2      : 0x3ffc6e94  A3      : 0x3ffc0d18  A4      : 0x8012b92e  A5      : 0x3ffbc890  
A6      : 0x3ffc2658  A7      : 0x3ffbc8b0  A8      : 0x00000b00  A9      : 0x00060024  
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x80093f55  A13     : 0x00000003  
A14     : 0x00000003  A15     : 0x3ffc0d48  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000b68  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x4008a5c5:0x3ffc0c88 |<-CORRUPTED
0x4008a5c5: r_rwbtdm_isr_wrapper at intc.c:?
```
Maybe the task stack sizes are too small or I sometimes give wrong addresses to memmove. Or the increased buffer size causes memory pressure, ... I'll check.

pschatzmann, 2022-03-07:
That's an example sketch using the ResampleStream which is dropping  every 250th frame
```
#define USE_A2DP
#include "AudioConfigLocal.h"
#include "AudioTools.h"

BluetoothA2DPSink a2dp_sink;
SPDIFStream spdif;
ResampleStream<int16_t> resample(spdif);

// Write data to SPDIF in callback
void read_data_stream(const uint8_t *data, uint32_t length) {
    resample.write(data, length);
}

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);

  // register callback
  a2dp_sink.set_stream_reader(read_data_stream, false);

  // Start Bluetooth Audio Receiver
  a2dp_sink.set_auto_reconnect(false);
  a2dp_sink.start("a2dp-spdif");

  // setup output
  auto cfg = spdif.defaultConfig();
  cfg.pin_data = 23;
  cfg.sample_rate = a2dp_sink.sample_rate();
  cfg.channels = 2;
  cfg.bits_per_sample = 16;
  spdif.begin(cfg);

  // configure resample
  auto rcfg = resample.defaultConfig();
  rcfg.channels = 2;
  rcfg.skip_every_nth = 250; 
  resample.begin(rcfg);
 
}

void loop() {
  delay(100);
}
```

joba-1, 2022-03-14:
Had a little time to test last weekend:
Using your example as is overlays low frequency clicks on the left channel over the usual sound.
With usual sound I mean: perfect audio with added soft clicks.
Low frequency is about 47Hz
The new clicks are even there if I pause bt audio.

joba-1, 2022-03-18:
I finally got spdif out working. No need to increase buffer size or cpu frequency.
The espresssif forum helped by hinting on the need for rate limiting in my code to sync bt in and i2s out frequency.
For now I had to use the [amedes code](https://github.com/joba-1/esp_a2dp_sink_spdif/blob/c7d33f29a52e7da59f4baee206bcc0fad7681f7e/main/bt_app_core.c#L192) because I needed the average buffer usage to decide how many samples to drop per callback (ended up going back and forth between 2 and 3 samples) and I couldn't easily figure out how to do this with the streaming interfaces here.
If I could, I would be interested to use your resampling instead of simple dropping.

pschatzmann, 2022-04-20:
I was wondering if always dropping 3 samples would work as well. In this case the resampling should do the trick as well.

joba-1, 2022-04-20:
Not for me unfortunately. I implemented a hysteresis of 10% buffer level to make the number of frames dropped as stable as possible.
It ran a long time (maybe a minute or two) on 3 frames, then buffer was too empty and it went up to only 2 dropped frames for maybe 30s before it went up to three again. 

Overall the ESP32 seems to be a nightmare for realtime applications. I had drops due to full buffer at average buffer level of 65% and buffer running empty on average level 35% (pseudo floating averages over 100 callbacks) so I adjusted the 3/2 frames at 45% and 55% to be reasonably safe - but still not perfect, still sometimes the buffer is running empty (phone and ESP32 no more than 0.5m apart). I have put the project on hold for now...

pschatzmann, 2022-04-20:
I think the conclusion is that it is easier to use a 5 USD I2S DAC...

joba-1, 2022-04-20:
I'm sure you are right - I already ordered twice and  it got cancelled by the seller both times - bad luck

joba-1, 2023-01-22:
I finally got two pcm5102 and had a little time to experiment.

I used the example basic-a2dp-i2s on a mh et live esp32 minikit and it worked great!

Then, being nosy, I also used basic-a2dp-spdif. First I got reboots due to failed memory allocation. After reducing to 24 256-buffers it also worked fine! No soft clicks or other artifacts even after several songs. 

In direct comparison the spdif sound is a bit more crisp and has more dry base. I am very happy with it. Finally my good old RX-V1600 has bluetooth input :) 

Thanks for your support!

closed 2023-01-22 by joba-1