pschatzmann/ESP32-A2DP #170: audio_state_changed callback not triggering for "suspended"

clarkbremer, 2022-02-07:
In both my code, and your example code (https://github.com/pschatzmann/ESP32-A2DP/blob/2ac43830d17d3fdd73daecab5414d6882d06c7cb/examples/bt_music_receiver_status_callback/bt_music_receiver_status_callback.ino), The callback triggers for `Started`, but does not trigger for `Suspended`.  I see this log statement firing:  https://github.com/pschatzmann/ESP32-A2DP/blob/183ae4d9208f0bac370afbe57370aefdffe49eb0/src/BluetoothA2DPSink.cpp#L616 for both cases, but the callback does not trigger for the `Suspended` case.  Looking at the code, it makes no sense to me.  The only thing that looks different in the output is the i2s stuff:

```
[ 19952][I][BluetoothA2DPSink.cpp:366] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 19956][I][BluetoothA2DPSink.cpp:610] handle_audio_state(): A2DP audio state: Started
ESP32:  Audio State Change.  State: 2
ESP32:  Audio Started
[ 19974][I][BluetoothA2DPSink.cpp:621] handle_audio_state(): i2s_start
[ 29125][I][BluetoothA2DPSink.cpp:366] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 29125][I][BluetoothA2DPSink.cpp:610] handle_audio_state(): A2DP audio state: Suspended
[ 29131][W][BluetoothA2DPSink.cpp:626] handle_audio_state(): i2s_stop
[ 32264][I][BluetoothA2DPSink.cpp:477] app_gap_callback(): event: 13
[ 40254][I][BluetoothA2DPSink.cpp:477] app_gap_callback(): event: 13
[ 40254][I][BluetoothA2DPSink.cpp:366] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 40258][I][BluetoothA2DPSink.cpp:610] handle_audio_state(): A2DP audio state: Started
ESP32:  Audio State Change.  State: 2
ESP32:  Audio Started
[ 40276][I][BluetoothA2DPSink.cpp:621] handle_audio_state(): i2s_start
[ 51781][I][BluetoothA2DPSink.cpp:366] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 51782][I][BluetoothA2DPSink.cpp:610] handle_audio_state(): A2DP audio state: Suspended
```

pschatzmann, 2022-02-07:
The line numbers are not matching with the latest code. 
I suggest that you update the a2dp library to the latest version...

clarkbremer, 2022-02-07:
Yeah, I was a few commits behind.  That fixed it.  Thanks.

closed 2022-02-07 by clarkbremer