pschatzmann/ESP32-A2DP #171: bt_music_sender example - Timed tones on demand

pskerrett, 2022-02-07:
Hi Phil,

Thank you so much for putting together an amazing project. I've been working with your library for a few days now. 

I have everything working in the music_sender_sample where I can have the ESP32 generate a tone and play through bluetooth speakers. This is *almost* what I need for a project I'm working on, but I'm struggling to finish it off, and if anyone can help I'd appreciate it.

I would like to modify the example so that it would just play a short tone of x-milliseconds when a particular condition is met. Right now I cannot figure out how to port the functionality out of setup() where it plays a constant tone.

Thank you so much in advance!

pschatzmann, 2022-02-08:
The easiest would be that normally you would just send an array of 0 - (by clearing the data) and when your event happens you make sure that you return the audible audio data (at the begiinning fo the buffer in the length that you need), just to return empty data agein...

pskerrett, 2022-02-08:
Thank you Phil. I think I understand, we would essentially keep the BT
module sending data, just interrupting the data from 0’s to a tone at
specific times.

I will give this a try & report back what I find. Thank you again.

closed 2022-04-08 by pschatzmann