pschatzmann/ESP32-A2DP #174: Driving a single speaker from stereo source

bance, 2022-02-27:
was trying to drive a single speaker from internal dac (gpio25).

tried I2S_CHANNEL_FMT_ALL_RIGHT and I2S_CHANNEL_FMT_ONLY_RIGHT in i2s config, but the result is same as I2S_CHANNEL_FMT_RIGHT_LEFT - only right channel sound is heard.

any possible way to config the bt speaker as a downmixed mono output from internal dac?

thanks!

pschatzmann, 2022-02-27:
I have 2 methods that actually do the same
- a2dp_sink .set_channels(MONO)
- a2dp_sink .set_mono_downmix(true)
Both calls make sure that the data of the left and right channels are combined. Then you can then use either gpio25 or gpio26 since both have the same signal...

bance, 2022-02-27:
Thanks Phil for the quick reply.

Indeed a2dp_sink.set_channels(I2S_CHANNEL_MONO) and a2dp_sink .set_mono_downmix(true) both work as expected.

Your library is a wonderful piece and very easy to use. Thank you!

Bance

closed 2022-02-27 by bance