pschatzmann/ESP32-A2DP #179: A2DP works perfectly with "QCY-qs2" earplugs, but does not connect to "boAt Rockerz 510" headset.

DevPhone, 2022-03-03:
I'm testing with the example player-sd-a2dp.ino on a ESP32 connected to SD card module as mentioned in the example readme.
Trying to get the music playing on bluetooth headset.
It worked with the earplugs, but for some reason it does not pair with the headset - would love to get it working with both. 

Here are the things I tried without any success in pairing with the boAt Rockerz 510....

1) I tried changing the pin_code to 0000 in BluetoothA2DPSource.cpp from the default 1234 in line 117 - no success.
2) I tried changing to 16 digit code and 4 digit code in line 476 as well - but no success.

I wonder if the spaces in the device name is the problem ? Or am I overlooking something silly?

Works well : Manual : https://images-eu.ssl-images-amazon.com/images/I/91b8Vl8xeIS.pdf

Does not work : Manual : https://manual-hub.com/manuals/boat-rockerz-510-01-pdf-manual/

Profile : 
1.Bluetooth Device : boAt Rockerz 510
2.MAC Address : 00:XX:XX:00:E9:D6
3.Type : Classic - BR/EDR
4.Profiles Supported:
HSP
HFP
A2DP - SNK
AVRCP
5.UUID List :
00001XXX-0000-1000-8000-00805f9XX4fb
00001XXX-0000-1000-8000-00805f9XX4fb

Any advice to get this working ? Thanks in advance : )

pschatzmann, 2022-03-04:
I dont really have any experiance with devices that need an password. Maybe if you change the loglevel to Debug you might be able to figure out what's going wrong...

DevPhone, 2022-03-10:
Just sharing my findings with you....
Here are the Debug logs - One bluetooth device works, the other just does not. :) 
Something mysterious happens at AudioA2DP.h : 147 that I do not understand at the moment. 
Will look into this later when I have more time to spare.
-------------------------------QCY-qs2_R - connects and works ---------------------------------

10:59:21.501 -> [I] AudioA2DP.h : 134 - Connecting to QCY-qs2_R
10:59:21.501 -> [I] AudioA2DP.h : 141 - Starting a2dp_source...
10:59:21.998 -> [D] AudioA2DP.h : 147 - waiting for connection
10:59:21.998 -> [D] AudioA2DP.h : 292 - static void audio_tools::A2DPStream::a2dpStateCallback(esp_a2d_connection_state_t, void*)
10:59:22.031 -> [W] AudioA2DP.h : 297 - ==> state: Connecting
10:59:22.991 -> [D] AudioA2DP.h : 147 - waiting for connection
10:59:23.985 -> [D] AudioA2DP.h : 147 - waiting for connection
10:59:24.979 -> [D] AudioA2DP.h : 147 - waiting for connection
10:59:26.005 -> [D] AudioA2DP.h : 147 - waiting for connection
10:59:26.701 -> [D] AudioA2DP.h : 292 - static void audio_tools::A2DPStream::a2dpStateCallback(esp_a2d_connection_state_t, void*)
10:59:26.734 -> [W] AudioA2DP.h : 297 - ==> state: Connected
10:59:26.999 -> [I] AudioA2DP.h : 150 - a2dp_source is connected...
10:59:26.999 -> [I] AudioPlayer.h : 772 - setVolume(0.900000)
10:59:26.999 -> [I] AudioStreams.h : 718 - setVolume: 0.900000
10:59:26.999 -> [D] AudioPlayer.h : 619 - virtual bool audio_tools::AudioPlayer::begin(int, bool)
10:59:26.999 -> [D] AudioOutput.h : 324 - void audio_tools::EncodedAudioStream::begin()
10:59:27.032 -> [D] CodecMP3Helix.h : 74 - virtual void audio_tools::MP3DecoderHelix::begin()

-------------------------------boAt Rockerz 510 - does not connect---------------------------------

11:00:31.013 -> [D] AudioA2DP.h : 292 - static void audio_tools::A2DPStream::a2dpStateCallback(esp_a2d_connection_state_t, void*)
11:00:31.013 -> [W] AudioA2DP.h : 297 - ==> state: Disconnected
11:00:32.007 -> [D] AudioA2DP.h : 292 - static void audio_tools::A2DPStream::a2dpStateCallback(esp_a2d_connection_state_t, void*)
11:00:32.007 -> [W] AudioA2DP.h : 297 - ==> state: Connecting
11:00:37.106 -> [D] AudioA2DP.h : 292 - static void audio_tools::A2DPStream::a2dpStateCallback(esp_a2d_connection_state_t, void*)
11:00:37.139 -> [W] AudioA2DP.h : 297 - ==> state: Disconnected
11:01:18.845 -> [I] AudioA2DP.h : 134 - Connecting to boAt Rockerz 510
11:01:18.845 -> [I] AudioA2DP.h : 141 - Starting a2dp_source...
11:01:19.341 -> [D] AudioA2DP.h : 147 - waiting for connection
11:01:19.375 -> [D] AudioA2DP.h : 292 - static void audio_tools::A2DPStream::a2dpStateCallback(esp_a2d_connection_state_t, void*)
11:01:19.375 -> [W] AudioA2DP.h : 297 - ==> state: Connecting
11:01:20.335 -> [D] AudioA2DP.h : 147 - waiting for connection
11:01:21.361 -> [D] AudioA2DP.h : 147 - waiting for connection
11:01:22.355 -> [D] AudioA2DP.h : 147 - waiting for connection
11:01:23.348 -> [D] AudioA2DP.h : 147 - waiting for connection
11:01:24.341 -> [D] AudioA2DP.h : 147 - waiting for connection
11:01:24.507 -> [D] AudioA2DP.h : 292 - static void audio_tools::A2DPStream::a2dpStateCallback(esp_a2d_connection_state_t, void*)
11:01:24.507 -> [W] AudioA2DP.h : 297 - ==> state: Disconnected
11:01:25.335 -> [D] AudioA2DP.h : 147 - waiting for connection
11:01:26.362 -> [D] AudioA2DP.h : 147 - waiting for connection
11:01:27.355 -> [D] AudioA2DP.h : 147 - waiting for connection
11:01:28.349 -> [D] AudioA2DP.h : 147 - waiting for connection
11:01:29.342 -> [D] AudioA2DP.h : 147 - waiting for connection
11:01:29.342 -> [D] AudioA2DP.h : 292 - static void audio_tools::A2DPStream::a2dpStateCallback(esp_a2d_connection_state_t, void*)
11:01:29.376 -> [W] AudioA2DP.h : 297 - ==> state: Connecting
11:01:30.336 -> [D] AudioA2DP.h : 147 - waiting for connection
11:01:31.363 -> [D] AudioA2DP.h : 147 - waiting for connection
11:01:32.356 -> [D] AudioA2DP.h : 147 - waiting for connection
11:01:33.350 -> [D] AudioA2DP.h : 147 - waiting for connection
11:01:34.343 -> [D] AudioA2DP.h : 147 - waiting for connection
11:01:34.476 -> [D] AudioA2DP.h : 292 - static void audio_tools::A2DPStream::a2dpStateCallback(esp_a2d_connection_state_t, void*)
11:01:34.476 -> [W] AudioA2DP.h : 297 - ==> state: Disconnected

-------------------------------Comparing Debug Logs - End---------------------------------

For now, I guess Ill go back to the esp-adf / idf code that works for both headsets.
No worries :)

Another related query :
Do SD Card and A2DP work simultaneously for Audiokit AC101? Or do they use the same CS 13 ?

I ask this because SD stopped working when I enable A2DP, but it works ok when using earphones wired connector, in this library. 

Again for this case, esp-adf examples works ok with both A2DP and SD simultaneously.
 
Thank you for your time.

pschatzmann, 2022-03-10:
Are you referring to [basic-a2dp-audiokit/](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/basic-a2dp-audiokit/basic-a2dp-audiokit.ino) ?
A2DP itself does not use any pins - so there can't be any conflict from there ...

pschatzmann, 2022-03-29:
I just merged a correction which might help

pschatzmann, 2023-07-28:
I guess you just forgot to set your device into pairing mode.
After it has paired, auto reconnect might help.
The password request needs to come from the speaker and this is obviously missing here!

pschatzmann, 2023-07-28:
My library was a copy from the a2dp_source from 2 years ago, so it does not contain all the corrections to the actual code. I noticed that the current start up logic looks differently now and there might be some new events that are not handled

AlbertoLopSie, 2023-07-28:
Well, what I need is a way to tunnel an I2S stream to Bluetooth through an a2dp_source.

The esp-idf sample has the bluetooth part working, but I couldn't link your I2SStream library into that code.

Would you suggest me any way to do it?

Thanks!!

pschatzmann, 2023-08-02:
I think this should be trivial, just read the data into the array. Since the project is C you might need to create a c function in the C++ code.

However I would have hoped that you help me to find the differences in the code, so that others could profit from the correction as well. It should not take more then 1 or 2h to compare the methods.

pschatzmann, 2023-08-03:
I kept the method names more or less the same and just moved it to a  C++ class so that we can easily create a subclass and overwrite the implementation. 

So you could start with the example provided by Espressif and compare the methods my just searching the corresponding class method in BluetoothA2DPSource.cpp

If it's just the login which is not working you can set the priority on the methods which handle the password requests.

pschatzmann, 2023-08-03:
Did you try to activate Secure Simple Pairing ?

pschatzmann, 2023-08-03:
If you check in https://github.com/espressif/esp-idf/blob/53ff7d43dbff642d831a937b066ea0735a6aca24/examples/bluetooth/bluedroid/classic_bt/a2dp_source/main/main.c you can find #if (CONFIG_BT_SSP_ENABLED == true)

AlbertoLopSie, 2023-08-06:
Hi Phil,

I managed to re-build my old versión and installed it on my board (the ONLY one I have is installed over a ceiling hatch, that's why I had to wait to weekend :-))

First thing, I couldn't compile with your new branch, it throws a lo of warnings and a couple of compile errors. 
``` sh
.pio/libdeps/esp32dev/ESP32-A2DP/src/BluetoothA2DPSource.cpp:546:8: error: 'ESP_BT_GAP_ACL_CONN_CMPL_STAT_EVT' was not declared in this scope
   case ESP_BT_GAP_ACL_CONN_CMPL_STAT_EVT:
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32dev/ESP32-A2DP/src/BluetoothA2DPSource.cpp:546:8: note: suggested alternative: 'ESP_BT_GAP_QOS_CMPL_EVT'
   case ESP_BT_GAP_ACL_CONN_CMPL_STAT_EVT:
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        ESP_BT_GAP_QOS_CMPL_EVT
In file included from .pio/libdeps/esp32dev/ESP32-A2DP/src/BluetoothA2DPCommon.h:56,
                 from .pio/libdeps/esp32dev/ESP32-A2DP/src/BluetoothA2DPSource.h:22,
                 from .pio/libdeps/esp32dev/ESP32-A2DP/src/BluetoothA2DPSource.cpp:16:
.pio/libdeps/esp32dev/ESP32-A2DP/src/BluetoothA2DPSource.cpp: In member function 'virtual void BluetoothA2DPSource::bt_av_hdl_avrc_ct_evt(uint16_t, void*)':
.pio/libdeps/esp32dev/ESP32-A2DP/src/BluetoothA2DPSource.cpp:976:69: error: 'struct esp_avrc_ct_cb_param_t::avrc_ct_psth_rsp_param' has no member named 'rsp_code'; did you mean 'key_code'?
         rc->psth_rsp.key_code, rc->psth_rsp.key_state, rc->psth_rsp.rsp_code);
                                                                     ^~~~~~~~
/Users/Alberto/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-log.h:127:72: note: in definition of macro 'log_i'
 #define log_i(format, ...) log_printf(ARDUHAL_LOG_FORMAT(I, format), ##__VA_ARGS__)
                                                                        ^~~~~~~~~~~
.pio/libdeps/esp32dev/ESP32-A2DP/src/BluetoothA2DPSource.cpp:973:5: note: in expansion of macro 'ESP_LOGI'
     ESP_LOGI(
     ^~~~~~~~
In file included from .pio/libdeps/esp32dev/audio-tools/src/AudioLibs/AudioA2DP.h:14,
                 from /Users/Alberto/dev/esp32/BlueRepeater/BlueReptr-Source/src/bt_i2s_2_source_api.ino:9:
.pio/libdeps/esp32dev/ESP32-A2DP/src/BluetoothA2DPSource.h:50:1: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
 };
```

But, on the original versión I got some interesting findings. To my surprise, on the first attempt, the ESP32 connected to the BT Speaker without a glitch... WHAT ???

Looking to the logs a little closer I noted there was no trace of the PIN request event... How is this possible???  Well... By the hard way I learnt that Bluetooth caches the paired devices (and its keys) on NVS. Given this module was previously running the ESP-ADF code that connected fine, its credentials were saved in NVS and thus it was never  to provide a new PIN to the speaker.

``` sh
[     5][D][BluetoothA2DPSource.cpp:111] BluetoothA2DPSource(): [BT_API] BluetoothA2DPSource, 
[    33][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
Starting  Blue Repeater/Source...
Starting I2S...
[I] I2SConfig.h : 118 - rx/tx mode: RX_MODE
[I] I2SConfig.h : 119 - port_no: 0
[I] I2SConfig.h : 120 - is_master: Slave
[I] I2SConfig.h : 121 - sample rate: 44100
[I] I2SConfig.h : 122 - bits per sample: 16
[I] I2SConfig.h : 123 - number of channels: 2
[I] I2SConfig.h : 125 - signal_type: Digital
[I] I2SConfig.h : 127 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 129 - auto_clear: false
[I] I2SConfig.h : 131 - use_apll: true
[I] I2SConfig.h : 136 - buffer_count:6
[I] I2SConfig.h : 137 - buffer_size:1024
[I] I2SConfig.h : 147 - pin_bck: 14
[I] I2SConfig.h : 149 - pin_ws: 15
[I] I2SConfig.h : 151 - pin_data: 32
Starting A2DP...
[   103][D][BluetoothA2DPSource.cpp:141] set_pin_code(): [BT_API] set_pin_code, 
[   104][I][BluetoothA2DPCommon.h:129] set_volume(): [BT_AV] set_volume: 255
[   111][D][BluetoothA2DPSource.cpp:153] start(): [BT_API] start, 
[   116][D][BluetoothA2DPSource.cpp:171] start_raw(): [BT_API] start_raw, 
[   123][D][BluetoothA2DPCommon.cpp:149] get_last_connection(): [BT_AV] get_last_connection
[   132][D][BluetoothA2DPCommon.cpp:172] get_last_connection(): [BT_AV] => 15:f7:b0:e4:ad:e8
[   665][D][BluetoothA2DPSource.cpp:324] bt_app_task_start_up(): [BT_AV] bt_app_task_start_up
[   666][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x0, param len 0
[   673][D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): [BT_API] bt_app_task_handler, sig 0x1, 0x0
[   682][D][BluetoothA2DPSource.cpp:536] bt_av_hdl_stack_evt(): [BT_AV] bt_av_hdl_stack_evt evt 0
[   691][I][BluetoothA2DPCommon.cpp:277] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable true
[   701][W][BluetoothA2DPSource.cpp:562] bt_av_hdl_stack_evt(): [BT_AV] Reconnecting to 15:f7:b0:e4:ad:e8
[   704][D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): [BT_API] bt_app_work_dispatch event 0x4, param len 16
[... 61 lines cut ...]
[  3020][D][BluetoothA2DPSource.cpp:459] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 5
[  3020][I][BluetoothA2DPSource.cpp:495] bt_app_gap_callback(): [BT_AV] ESP_BT_GAP_PIN_REQ_EVT min_16_digit:0
[  3027][I][BluetoothA2DPSource.cpp:501] bt_app_gap_callback(): [BT_AV] Input pin code: 1234
[  3133][D][BluetoothA2DPSource.cpp:459] bt_app_gap_callback(): [BT_AV] bt_app_gap_callback evt 4
[... cut ...]
And the speaker connected fine again but this time after sending to it the requested PIN code (your log message always says it's sending '1234' but that is not true)

## Conclusion

Your code appear to work ok, even in the current version. Apparently the problem was related to cached credentials and perhaps some bad initialized NVS during my first development stages.

I'm sorry for the wasted time. As I told you, there's a lot in the ESP32 Bluetooth stack internals I'm not aware of and evidently this problem was mainly related to that ignorance.

Thanks a lot for your help!

Alberto.

pschatzmann, 2023-08-06:
If you get compile error you are using an obsolete ESP32 version!
Please update using the Boards manager: The actual version is 2.0.11

closed 2023-08-12 by pschatzmann