pschatzmann/ESP32-A2DP #185: Help implementing a2dp_sink with ADAU1701

curtis-j-campbell, 2022-03-12:
Hi all, 

I'm attempting to use this library to stream audio via i2s from an ESP32 DEV kit into an ADAU1701 DSP board however I'm only getting noise on the DSP's outputs. 

I have linked the up the ESP32's default i2s pins and MCLK pin to the DSP's i2s input and used the following code setup:

``` 
#include <BluetoothA2DPSink.h>

BluetoothA2DPSink a2dp_sink;

void setup() {
  static i2s_config_t i2s_config = {
    .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX),
    .sample_rate = 44100, // updated automatically by A2DP   
    .bits_per_sample = (i2s_bits_per_sample_t)16,
    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
    .communication_format = (i2s_comm_format_t)I2S_COMM_FORMAT_STAND_MSB,
    .intr_alloc_flags = 0, // default interrupt priority
    .dma_buf_count = 8,
    .dma_buf_len = 64,
    .use_apll = true,
    .tx_desc_auto_clear = true // avoiding noise in case of data unavailability
  };
  a2dp_sink.set_i2s_config(i2s_config);
  a2dp_sink.i2s_mclk_pin_select(1);
  a2dp_sink.start("bluetooth speaker");
}

void loop() {
}
```

The DSP's datasheet mentions that "Data is processed in twos complement, MSB-first format" but I'm struggling to identify which `communication_format` to use.  Additionally, can anyone clarify the correct value I should be using for `.bits_per_sample` ?

Let me know if I need to add anymore information,

Thanks in advance.

pschatzmann, 2022-03-13:
As far as I can understand you need to make sure that the settings are matching with what you have defined in the Serial Input Control Register of your ADAU1701. 
I would suggest to set M[2:0]  to I2S and on the esp32 use I2S_COMM_FORMAT_STAND_I2S. 
For the bits_per_sample I would expect that I2S_BITS_PER_SAMPLE_16BIT and I2S_BITS_PER_SAMPLE_24BIT should work.

curtis-j-campbell, 2022-03-13:
Thanks, I've realised that my hardware setup was incorrect and causing issues

closed 2022-03-13 by curtis-j-campbell