pschatzmann/ESP32-A2DP #186: help diagnosing sink example

belm0, 2022-03-13:
I'm trying to run a [sink example for M5Stack Core](https://github.com/m5stack/M5Unified/blob/master/examples/Advanced/Bluetooth_with_ESP32A2DP/Bluetooth_with_ESP32A2DP.ino) that uses this library.  (I'm not sure the example is too sophisticated.)

I've tried Android, macOS, and iOS devices-- they all time out when paring.  I've enabled verbose debugging, but it doesn't indicate much.  Just before the timeout, there is this message:

```
[D][BluetoothA2DPSink.cpp:1136] ccall_app_gap_callback(): ccall_app_gap_callback
```

I instrumented the callback, and found that it's event 4 (ESP_BT_GAP_AUTH_CMPL_EVT).

I'd appreciate any guidance-- thank you.

esp32 version is 1.0.6

log
```
[D][BluetoothA2DPSink.cpp:123] start(): start
[I][BluetoothA2DPCommon.cpp:207] log_free_heap(): Available Heap: 221212
[I][BluetoothA2DPSink.cpp:136] start(): Device name will be set to 'ESP32'
[D][BluetoothA2DPSink.cpp:1006] init_nvs(): init_nvs
[I][BluetoothA2DPSink.cpp:184] init_i2s(): init_i2s
[D][BluetoothA2DPSink.cpp:266] init_bluetooth(): init_bluetooth
[I][BluetoothA2DPSink.cpp:271] init_bluetooth(): controller initialized
[I][BluetoothA2DPSink.cpp:280] init_bluetooth(): bluedroid initialized
[I][BluetoothA2DPSink.cpp:289] init_bluetooth(): bluedroid enabled
[D][BluetoothA2DPSink.cpp:387] app_task_start_up(): app_task_start_up
[D][BluetoothA2DPSink.cpp:1112] ccall_app_task_handler(): ccall_app_task_handler
[D][BluetoothA2DPSink.cpp:358] app_task_handler(): app_task_handler
[D][BluetoothA2DPSink.cpp:310] app_work_dispatch(): app_work_dispatch event 0x0, param len 0
[D][BluetoothA2DPSink.cpp:342] app_send_msg(): app_send_msg
[D][BluetoothA2DPSink.cpp:365] app_task_handler(): app_task_handler, sig 0x1, 0x0
[I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[D][BluetoothA2DPSink.cpp:333] app_work_dispatched(): app_work_dispatched
[D][BluetoothA2DPSink.cpp:1142] ccall_av_hdl_stack_evt(): ccall_av_hdl_stack_evt
[D][BluetoothA2DPSink.cpp:834] av_hdl_stack_evt(): av_hdl_stack_evt evt 0
[D][BluetoothA2DPSink.cpp:839] av_hdl_stack_evt(): av_hdl_stack_evt av_hdl_stack_evt BT_APP_EVT_STACK_UP
[D][BluetoothA2DPSink.cpp:848] av_hdl_stack_evt(): AVRCP controller initialized!
[D][BluetoothA2DPSink.cpp:888] av_hdl_stack_evt(): set_scan_mode_connectable(true)
[I][BluetoothA2DPCommon.cpp:255] set_scan_mode_connectable(): set_scan_mode_connectable true
[I][BluetoothA2DPCommon.cpp:207] log_free_heap(): Available Heap: 107024
[D][BluetoothA2DPSink.cpp:1136] ccall_app_gap_callback(): ccall_app_gap_callback
app_gap_callback: 4
```

pschatzmann, 2022-03-17:
I think you would start with a simple example and I would expect that this would work...

belm0, 2022-03-18:
I tried just `a2dp_sink.start("MyMusic")`.

I was able to partially pair, with the library logging "authentication success: Pixel 4a (5G)".  But from the phone's view, the pairing seems to hang and time out, and the device is left unconnected (and not able to reconnect).

Is it dud hardware, a poor antenna, ...?

belm0, 2022-03-18:
I tried a 3rd Android device, and got it working...

closed 2022-03-18 by belm0