pschatzmann/ESP32-A2DP #193: Bluetooth Source Not Pairing

degnarraer, 2022-03-16:
What does this code do:

/* search for device with MAJOR service class as "rendering" in COD */
    if (!esp_bt_gap_is_valid_cod(cod) ||
            !(esp_bt_gap_get_cod_srvc(cod) & ESP_BT_COD_SRVC_RENDERING)) {
        return;
    }

I am not able to get the ESP32 to connect to most anything other than another ESP32... However, when I comment out that code, suddenly it works.

also, the bluetooth soundbar is responding with a COD of 0x414

[I][BluetoothA2DPSource.cpp:455] bt_app_gap_callback(): Discovery started.
[I][BluetoothA2DPSource.cpp:381] filter_inquiry_scan_result(): Scanned device: cc:6e:a4:30:c6:c8
[I][BluetoothA2DPSource.cpp:387] filter_inquiry_scan_result(): --Class of Device: 0x414
[I][BluetoothA2DPSource.cpp:391] filter_inquiry_scan_result(): --RSSI: -86
[I][BluetoothA2DPSource.cpp:411] filter_inquiry_scan_result(): Device discovery found: [AV] Samsung Soundbar MM55 M-Series
[I][BluetoothA2DPSource.cpp:416] filter_inquiry_scan_result(): Checking name: [AV] Samsung Soundbar MM55 M-Series
[I][BluetoothA2DPSource.cpp:424] filter_inquiry_scan_result(): Found a target device, address cc:6e:a4:30:c6:c8, name [AV] Samsung Soundbar MM55 M-Series
[D][BluetoothA2DPCommon.cpp:142] set_last_connection(): set_last_connection: cc:6e:a4:30:c6:c8
[I][BluetoothA2DPSource.cpp:428] filter_inquiry_scan_result(): Cancel device discovery ...
[I][BluetoothA2DPSource.cpp:446] bt_app_gap_callback(): Device discovery stopped.
[I][BluetoothA2DPSource.cpp:447] bt_app_gap_callback(): a2dp connecting to peer: [AV] Samsung Soundbar MM55 M-Series
[D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): bt_app_work_dispatch event 0x0, param len 16
[D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x0
[I][BluetoothA2DPSource.cpp:611] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 4, evt 0x0
[D][BluetoothA2DPSource.cpp:579] process_user_state_callbacks(): process_user_state_callbacks
[D][BluetoothA2DPSource.cpp:587] process_user_state_callbacks(): process_user_state_callbacks ESP_A2D_CONNECTION_STATE_EVT: Connecting
[D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): bt_app_work_dispatch event 0x0, param len 16
[D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x0
[I][BluetoothA2DPSource.cpp:611] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 4, evt 0x0
[D][BluetoothA2DPSource.cpp:579] process_user_state_callbacks(): process_user_state_callbacks
[D][BluetoothA2DPSource.cpp:587] process_user_state_callbacks(): process_user_state_callbacks ESP_A2D_CONNECTION_STATE_EVT: Disconnected
[D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): bt_app_work_dispatch event 0xff00, param len 0
[D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0xff00
[I][BluetoothA2DPSource.cpp:611] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 3, evt 0xff00
[D][BluetoothA2DPSource.cpp:579] process_user_state_callbacks(): process_user_state_callbacks
[I][BluetoothA2DPSource.cpp:652] bt_app_av_state_unconnected(): a2dp connecting to peer: cc:6e:a4:30:c6:c8
[D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): bt_app_work_dispatch event 0x0, param len 16
[D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x0
[I][BluetoothA2DPSource.cpp:611] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 4, evt 0x0
[D][BluetoothA2DPSource.cpp:579] process_user_state_callbacks(): process_user_state_callbacks
[D][BluetoothA2DPSource.cpp:587] process_user_state_callbacks(): process_user_state_callbacks ESP_A2D_CONNECTION_STATE_EVT: Connecting
[D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): bt_app_work_dispatch event 0x0, param len 12
[D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x0
[D][BluetoothA2DPSource.cpp:860] bt_av_hdl_avrc_ct_evt(): bt_av_hdl_avrc_ct_evt evt 0
[I][BluetoothA2DPSource.cpp:866] bt_av_hdl_avrc_ct_evt(): AVRC conn_state evt: state 1, [cc:6e:a4:30:c6:c8]
[D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): bt_app_work_dispatch event 0x0, param len 16
[D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x0
[I][BluetoothA2DPSource.cpp:611] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 4, evt 0x0
[D][BluetoothA2DPSource.cpp:579] process_user_state_callbacks(): process_user_state_callbacks
[D][BluetoothA2DPSource.cpp:587] process_user_state_callbacks(): process_user_state_callbacks ESP_A2D_CONNECTION_STATE_EVT: Connected
[I][BluetoothA2DPSource.cpp:671] bt_app_av_state_connecting(): a2dp connected
[I][BluetoothA2DPCommon.cpp:255] set_scan_mode_connectable(): set_scan_mode_connectable false
[D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): bt_app_work_dispatch event 0x5, param len 12
[D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x5
[D][BluetoothA2DPSource.cpp:860] bt_av_hdl_avrc_ct_evt(): bt_av_hdl_avrc_ct_evt evt 5
[I][BluetoothA2DPSource.cpp:897] bt_av_hdl_avrc_ct_evt(): AVRC remote features
[D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): bt_app_work_dispatch event 0xff00, param len 0
[D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0xff00
[I][BluetoothA2DPSource.cpp:611] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0xff00
[D][BluetoothA2DPSource.cpp:579] process_user_state_callbacks(): process_user_state_callbacks
[I][BluetoothA2DPSource.cpp:707] bt_app_av_media_proc(): a2dp media ready checking ...
[D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): bt_app_work_dispatch event 0x3, param len 16
[D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x3
[I][BluetoothA2DPSource.cpp:611] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0x3
[D][BluetoothA2DPSource.cpp:579] process_user_state_callbacks(): process_user_state_callbacks
[I][BluetoothA2DPSource.cpp:713] bt_app_av_media_proc(): a2dp media ready, starting ...
[D][BluetoothA2DPSource.cpp:923] get_data_default(): => len: 512 / result_len: 0
[D][BluetoothA2DPSource.cpp:932] get_data_default(): get_data_default - end of data: stopping
[D][BluetoothA2DPSource.cpp:923] get_data_default(): => len: 512 / result_len: 0
[D][BluetoothA2DPSource.cpp:932] get_data_default(): get_data_default - end of data: stopping
[D][BluetoothA2DPSource.cpp:923] get_data_default(): => len: 512 / result_len: 0
[D][BluetoothA2DPSource.cpp:932] get_data_default(): get_data_default - end of data: stopping
[D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): bt_app_work_dispatch event 0x3, param len 16
[D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x3
[I][BluetoothA2DPSource.cpp:611] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0x3
[D][BluetoothA2DPSource.cpp:579] process_user_state_callbacks(): process_user_state_callbacks
[I][BluetoothA2DPSource.cpp:725] bt_app_av_media_proc(): a2dp media start successfully.
[D][BluetoothA2DPSource.cpp:250] bt_app_work_dispatch(): bt_app_work_dispatch event 0x1, param len 16
[D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x1
[I][BluetoothA2DPSource.cpp:611] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0x1
[D][BluetoothA2DPSource.cpp:579] process_user_state_callbacks(): process_user_state_callbacks
[D][BluetoothA2DPSource.cpp:598] process_user_state_callbacks(): process_user_state_callbacks ESP_A2D_AUDIO_STATE_EVT: Started
[D][BluetoothA2DPSource.cpp:923] get_data_default(): => len: 512 / result_len: 0
[D][BluetoothA2DPSource.cpp:932] get_data_default(): get_data_default - end of data: stopping
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d8627  PS      : 0x00060030  A0      : 0x800d7bc0  A1      : 0x3ffeb5c0  
A2      : 0x3ffc3d94  A3      : 0x00000200  A4      : 0x00000200  A5      : 0x00080000  
A6      : 0x00000000  A7      : 0x3ffeada0  A8      : 0x800d8615  A9      : 0x3ffeb5a0  
A10     : 0x3ffdf3e0  A11     : 0x00000000  A12     : 0x00000200  A13     : 0x3ffea728  
A14     : 0x00000020  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00080000  LBEG    : 0x40091f4a  LEND    : 0x40091f55  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d8627:0x3ffeb5c0 0x400d7bbd:0x3ffeb5e0 0x400d86f2:0x3ffeb600 0x401321f5:0x3ffeb620 0x40132395:0x3ffeb650 0x401324ee:0x3ffeb670 0x401324ff:0x3ffeb690 0x40132527:0x3ffeb6b0 0x40132562:0x3ffeb6d0 0x40094386:0x3ffeb700

Rebooting...
⸮

pschatzmann, 2022-03-17:
In practice, most Bluetooth clients scan their surroundings in two successive steps: they first look for all bluetooth devices around them and find out their "class". You can do this on Linux with the hcitool scan command. Then, they use SDP in order to check if a device in a given class offers the type of service that they want.

As indicated in the readme I converted the solution from the Espressif example. I suggest that you have a look at the following https://github.com/espressif/esp-idf/issues/6548

Do you see any chance to apply the proposed correction and submit a pull request ?

clarkbremer, 2022-03-17:
You need to fork the project to your own account, then push your branch to that fork.  From there, you can open a PR.

pschatzmann, 2022-03-24:
Did you manage to test your correction. Are you still planning to provide a pull request ?

degnarraer, 2022-03-29:
#199

degnarraer, 2022-03-29:
sorry for the delay. I am slow at doing coding. I do have a functional project now though where I use 2 ESP32's to provide both a Bluetooth sink and source, and an audio processor that analyzes the data and controls LED visualizations. It is pretty sweet. Been working on it since Nov 2021.

pschatzmann, 2022-03-29:
Your project sounds interesting. Will you add it to https://github.com/pschatzmann/ESP32-A2DP/discussions/categories/show-and-tell ?

pschatzmann, 2022-04-08:
Code has been merged

closed 2022-04-08 by pschatzmann