pschatzmann/ESP32-A2DP #203: possible memory leak when disconnecting and reconnecting to an A2DP sink

yliniemi, 2022-04-04:
Hi! Thank you for a really nice and stable library :)

I've made a really fast and accurate spectrum analyzer and almost ran our of memory with it. There are is less than 16 kB left.
Here is a link to the show and tell.
https://github.com/pschatzmann/ESP32-A2DP/discussions/202

My problem is that ram is really precious to me. I have noticed that if I disconnect and reconnect my Debian PC from the Bluetooth speaker spectrum analyzer I get about 1 kB less free ram every time. I don't know if the problem is in the library or somewhere else. The weird thing is that when I disconnect and reconnect my Android phone there is no memory leak. Maybe the phone disconnects properly and my PC doesn't.

Here is the amount of free ram reported after each reconnect.

heap_caps_get_largest_free_block(MALLOC_CAP_8BIT) = 10076
heap_caps_get_largest_free_block(MALLOC_CAP_8BIT) = 7328
heap_caps_get_largest_free_block(MALLOC_CAP_8BIT) = 5976
heap_caps_get_largest_free_block(MALLOC_CAP_8BIT) = 4620
heap_caps_get_largest_free_block(MALLOC_CAP_8BIT) = 3232
heap_caps_get_largest_free_block(MALLOC_CAP_8BIT) = 2556
heap_caps_get_largest_free_block(MALLOC_CAP_8BIT) = 984

So currently I can reconnect 6 times before my esp32 crashes. This is not much especially since I'm planning to double the width of the led panel and that will eat at least 3 kB more ram. I'm planning on doing a few more things and they require 512 kB of ram each. I'm running quite tight. I never imagined that I would be counting free bytes in 2022 but here we are.

pschatzmann, 2022-07-14:
I did some tests with OSX and  and I can reproduce a memory loss of 36 bytes only after each disconnect - connect.
I tried to do some code review, but I did not find anything in my code that could cause this.

yliniemi, 2022-07-16:
Thanks for the suggestion to change log level to debug. I didn't know such a thing existed. I'll use that in my programs later on.

I did some testing. I had hard time replicating the issue. I've changed my source code in the mean time and FastLED library and esp-idf were also updated. Problem was a lot less pronounced. I didn't get a massive 1 kB memory leak per reconnect. It was more like a random amount of ram was free after a reconnect. Sometimes less, sometimes more.
https://pastebin.com/y6dkVUU7

I'm using two libraries. https://github.com/yash-sanghvi/ESP32/tree/master/FFT_on_ESP32_Arduino and FastLED. I'm pretty sure the FFT library only allocates ram when I first initialize it. FastLED might be doing some dynamic ram allocation. Perhaps your library allocates some ram in some location after a reconnect. Then FastLED allocates some ram in the remaining area. Then your library releases part of it's ram when it's disconnected but when reconnecting gets it's ram from a location after FastLED. Do this a few times and you run out of ram.
So my guess is that it was an issue with how esp-idf libraries allocate ram. Perhaps it was fixed with an update to esp-idf. I'm only guessing here. I haven't looked too deep into the source code. I'm also planning to switch from FastLED to https://github.com/hpwit/I2SClocklessLedDriver. Maybe I can save some ram by switching to it.

closed 2022-07-27 by pschatzmann