pschatzmann/ESP32-A2DP #205: Receiving AVRC Events as a Source

lemonjesus, 2022-04-07:
Hello!

I'm having trouble working with AVRC events with the ESP32 acting as the source. It's probably because I don't understand much wrt low-level Bluetooth stacks, so I'm a little lost as to how this is configured here. I'm using `android-audio-tools`, but I think the change would need to go in this library. I would like my ESP32 to receive remote commands (sent from a Bluetooth speaker with buttons) such as play/pause, volume up, volume down, etc. so that I can act on them appropriately. 

Correct me if I'm wrong, but it appears that AVRC commands end up here:

https://github.com/pschatzmann/ESP32-A2DP/blob/6895ec2bf74f7c25fb7b38dccda0224f9ded0ab8/src/BluetoothA2DPSource.cpp#L829-L844

which immediately calls `bt_app_work_dispatch`. However, when I add a logging statement to this function and I press a button on my speaker, I see no different output.

Perhaps I'm confused as to what this library does when it comes to AVRC. They're definitely initialized, but after that I'm completely lost from there. Are there additional steps that must be taken to tell the ESP32 that we want these events?

Again, I'm trying to learn about this low-level Bluetooth and FreeRTOS stuff. Hopefully I can end up making a contribution to this awesome library to support this feature. Thanks for any guidance you can provide.

pschatzmann, 2022-04-07:
I have never looked into this on the Source side, but there is some logic for handling of the volume.
Which Arduino ESP32 Version are you using ? 
see https://github.com/pschatzmann/ESP32-A2DP/wiki/Controlling-your-Phone-with-AVRC-Commands

lemonjesus, 2022-04-19:
Progress! <sub>(sorry for the following novel)</sub>

I think I misunderstood how the volume control works. Even though the ESP32 is the A2DP source in this case, it's acting as the AVRCP Controller when it comes to volume. My interpretation of the code is that if you call `set_volume()`, you have to notify the speaker (which, in this case, acts as the Target?) to change its volume (if it's capable of doing so). Please correct me if I'm wrong. 

This makes sense, but play/pause/prev/next can't work like this. It has to originate from the speaker which acts as the AVRCP Controller. The ESP32 _must_ be the AVRCP Target for these commands.

The repeated crashes I was getting from my previous comment was that `esp_avrc_tg_set_rn_evt_cap` was returning `ESP_ERR_NOT_SUPPORTED` when I tried to add support for `ESP_AVRC_RN_PLAY_STATUS_CHANGE` and `ESP_AVRC_RN_TRACK_CHANGE`. Infuriating, but according to page 16 of [ESP32 Bluetooth Architecture](https://s3-sa-east-1.amazonaws.com/robocore-lojavirtual/1205/ESP32_Bluetooth.Architecture.pdf) (published in 2018):

> Two roles defined in AVRCP are Controller (CT) and Target (TG). CT is a device that initiates a transaction by sending a command frame to a target. [...] For the time being, A2DP (SRC) and AVRCP (CT) are supported and the device can work as a loudspeaker which can also send remote control messages to the audio source.

Either I'm misunderstanding what the AVRCP target is supposed to do, or ESP-IDF is missing this functionality right now. In the case of my speaker - or just in general - it's certainly the former, because according to macOS's PacketLogger the commands are sent as AVRCP Passthrough commands. Perhaps there is no functionality missing at all and this was always how this was supposed to work. 🤔

I'm able* to receive the button presses as passthrough AVRCP commands with the following steps:

1. After initializing the Controller side of things on line 542 of `BluetoothA2DPSource.cpp`, I initialize the Target side as well:

        assert(esp_avrc_tg_init() == ESP_OK);
        esp_avrc_tg_register_callback(ccall_bt_app_rc_tg_cb); // along with the necessary functions to support this

1. I'm enabling all passthrough buttons just to be safe:

        esp_avrc_psth_bit_mask_t cmd_set = {0};
        esp_avrc_tg_get_psth_cmd_filter(ESP_AVRC_PSTH_FILTER_ALLOWED_CMD, &cmd_set);
        esp_avrc_tg_set_psth_cmd_filter(ESP_AVRC_PSTH_FILTER_SUPPORTED_CMD, &cmd_set);

1. In my `bt_app_rc_tg_cb()` it logs when it gets called with `event == ESP_AVRC_TG_PASSTHROUGH_CMD_EVT`. The params bear the button that was pressed (as well as if it's pressed or if it's released).

I now do not believe that play status/track change were ever their own AVRCP commands intended to be sent from the Controller to the Target. Instead, they're probably intended for the Target to send to the Controller. After all, they're just notifications, not commands. It's all making sense now.

\*The caveat is that the callback stops getting called after a short bit. Notably, I think it stops getting called after these log messages appear:

```
# first, an example of my logging the button press successfully
[ 26088][D][BluetoothA2DPSource.cpp:865] bt_app_rc_tg_cb(): bt_app_rc_tg_cb TG event 2
[ 26088][I][BluetoothA2DPSource.cpp:885] bt_app_rc_tg_cb(): AVRC passthrough cmd: key_code 0x44, key_state 1
# but then this gets logged
[ 30661][D][BluetoothA2DPSource.cpp:254] bt_app_work_dispatch(): bt_app_work_dispatch event 0xff00, param len 0
[ 30662][D][BluetoothA2DPSource.cpp:305] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0xff00
[ 30669][I][BluetoothA2DPSource.cpp:639] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0xff00
[ 30678][D][BluetoothA2DPSource.cpp:607] process_user_state_callbacks(): process_user_state_callbacks
# these four lines get logged over and over again, but no further button presses are logged
```

Once I solve this issue, I think everything will work the way I wanted. **So my new question is this:** what is happening during these four log messages that might goof up the state of the AVRCP target? I'll keep crawling the ESP-IDF documentation in the mean time and try to find what this event/state/whatever means.

Sorry this was so long, I just wanted to write down what I learned and share my new - possibly correct? - understanding. Hopefully I can solve this issue and drum up a PR because I think this would be a great addition to the A2DP Source capabilities of the ESP32.

pschatzmann, 2022-04-20:
I am sorry, this is outside of my expertise. I suggest to post this question on an ESP32 IDF forum...

lemonjesus, 2022-05-19:
ok so I have something that works. it's sorta weird and somewhat unstable so I'm not opening a PR yet, but you can look at [the code](https://github.com/pschatzmann/ESP32-A2DP/compare/main...lemonjesus:add-passthrough-commands-to-target). I need this and some other features for my project so I'll refine it, test it, and some day come back and make a PR here for it.

thank you for the help!

closed 2022-05-19 by lemonjesus