pschatzmann/ESP32-A2DP #206: Is there a way to implement sample mixing?

zenbooster, 2022-04-07:
Is there a way to implement sample mixing? Those. for example, I have several buttons, each of which has some sample attached to it. And if I press a certain button without waiting for the current sample to finish playing, the new sample should be mixed into what is currently playing.

pschatzmann, 2022-04-07:
Not yet..

zenbooster, 2022-04-08:
((
This would be a very useful feature.

pschatzmann, 2022-04-09:
Extended Audio Functionality is part of the [arduino-audio-tools](https://github.com/pschatzmann/arduino-audio-tools). Here is an [example](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-generator_mixer-audiokit/streams-generator_mixer-audiokit.ino)
And here is the [link to the documentation](https://pschatzmann.github.io/arduino-audio-tools/html/classaudio__tools_1_1_mixer.html).

closed 2022-04-20 by pschatzmann