pschatzmann/ESP32-A2DP #207: use this lib no voice output

BruceAKABear, 2022-04-10:
when I upload this lib to my esp32 dev module,not voice output, I use pcm5102a to dec i2s ,this chip work very well . how can I use this lib in right way?

here is my code:
`
#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;

void setup()
{

  i2s_pin_config_t i2SPinConfig = { .bck_io_num = GPIO_NUM_27,
                                    .ws_io_num = GPIO_NUM_26,
                                    .data_out_num = GPIO_NUM_25,
                                    .data_in_num = I2S_PIN_NO_CHANGE
                                  };
  a2dp_sink.set_volume(30);
  a2dp_sink.set_pin_config(i2SPinConfig);
  a2dp_sink.start("MyMusic");
  Serial.begin(115200);
  Serial.println(a2dp_sink.get_volume());
}

void loop()
{
  delay(1000); // do nothing
}

`

BruceAKABear, 2022-04-10:
> Did you read the Wiki - External-DAC ?

yes, and I use my DAC module in another project and it work good . and I redefine the i2s pin ,I think it well work but no output,so I dont konw what should I do net step

pschatzmann, 2022-04-10:
Which I2S Module exactly do you have ?
Are you sure that your other project is not defining additional non I2S pins to feed the module ?

BruceAKABear, 2022-04-10:
> Well, if you have defined the pins correctly it should work. Which I2S Module exactly do you have ? Are you sure that your other project is not defining additional non I2S pins to feed the module (e.g. for the XMT pin)? How did you test ? Maybe you just need to increase the volume on your phone ? Did you activate the logging and look at the log messages ?

I bulid a DAC module by myself ,it based on pcm5102a chip. and in order to use that module I hava handle other pin such as xmt in right way .
here is pin config and i2s config in my wifi audio project:

`
// i2s配置
i2s_config_t i2sConfig = { .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX),
                           .sample_rate = 44100,
                           .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
                           .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,
                           .communication_format =
                             i2s_comm_format_t(I2S_COMM_FORMAT_I2S),
                           .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
                           .dma_buf_count = 4,
                           .dma_buf_len = 512,
                           .use_apll = true,
                           .tx_desc_auto_clear = true,
                           .fixed_mclk = I2S_PIN_NO_CHANGE
                         };
// i2s pin配置
i2s_pin_config_t i2SPinConfig = { .bck_io_num = GPIO_NUM_27,
                                  .ws_io_num = GPIO_NUM_26,
                                  .data_out_num = GPIO_NUM_25,
                                  .data_in_num = I2S_PIN_NO_CHANGE
                                };
`
I upload my wifi audio firamewate to my esp32 to test the module work or not, the volume  in my cellphone is 100%

here is my log:

`
ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8
[    32][I][BluetoothA2DPSink.cpp:1052] set_volume(): set_volume 30
[    33][I][BluetoothA2DPSink.cpp:1202] volume_set_by_local_host(): Volume is set locally to: 23
[    36][I][BluetoothA2DPCommon.cpp:207] log_free_heap(): Available Heap: 200492
[    43][I][BluetoothA2DPSink.cpp:136] start(): Device name will be set to 'MyMusic'
[    51][I][BluetoothA2DPSink.cpp:184] init_i2s(): init_i2s
[    56][I][BluetoothA2DPSink.cpp:186] init_i2s(): init_i2s is_i2s_output
[   151][I][BluetoothA2DPSink.cpp:263] init_bluetooth(): controller initialized
[   478][I][BluetoothA2DPSink.cpp:272] init_bluetooth(): bluedroid initialized
[   548][I][BluetoothA2DPSink.cpp:281] init_bluetooth(): bluedroid enabled
[   548][I][BluetoothA2DPSink.cpp:360] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[   553][I][BluetoothA2DPCommon.cpp:241] set_scan_mode_connectable(): set_scan_mode_connectable true
[   561][I][BluetoothA[  Sin1][pp:360] app_task_handler(): app_taskBlantoor, APP_SIG_WORK_DIAPATSins.g: :475]   5_1apI]alluetco():2DPent:.10
p:5[   a74]dl_[Bd_etoo[   586][I][BluetoothA2DPSink.cpp:475] app_gap_callback(): event: 10
30
[   589][I][BluetoothA2DPSink.cpp:360] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[   591][I][BluetoothA2DPSink.cpp:638] handle_connection_state(): partner address: 14:c6:97:aa:64:0a
[   600][I][BluetoothA2DPSink.cpp:646] handle_connection_state(): A2DP connection state: Connecting, [14:c6:97:aa:64:0a]
[   611][I][BluetoothA2DPSink.cpp:718] handle_connection_state(): ESP_A2D_CONNECTION_STATE_CONNECTING
[  2128][I][BluetoothA2DPSink.cpp:360] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  2128][I][BluetoothA2DPSink.cpp:570] handle_audio_cfg(): a2dp audio_cfg_cb , codec type 0
[  2134][I][BluetoothA2DPSink.cpp:582] handle_audio_cfg(): a2dp audio_cfg_cb , sample_rate 44100
[  2143][I][BluetoothA2DPSink.cpp:593] handle_audio_cfg(): configure audio player 21-15-2-35

[  2151][I][BluetoothA2DPSink.cpp:599] handle_audio_cfg(): audio player configured, samplerate=44100
[  2178][I][BluetoothA2DPSink.cpp:360] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[... 83 lines cut ...]
[ 89319][I][BluetoothA2DPSink.cpp:784] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x1, Run Free (feat. IVIE)
[ 89328][I][BluetoothA2DPSink.cpp:360] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 89337][I][BluetoothA2DPSink.cpp:784] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x2, Deep Chills/IVIE
[ 89347][I][BluetoothA2DPSink.cpp:360] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 89356][I][BluetoothA2DPSink.cpp:784] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x4, Run Free (feat. IVIE)
[ 89366][I][BluetoothA2DPSink.cpp:360] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 89375][I][BluetoothA2DPSink.cpp:784] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x8, 0
[ 89384][I][BluetoothA2DPSink.cpp:360] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 89393][I][BluetoothA2DPSink.cpp:784] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x10, 0
[ 89402][I][BluetoothA2DPSink.cpp:360] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 89411][I][BluetoothA2DPSink.cpp:784] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x20, 
[... cut ...]
[ 89487][I][BluetoothA2DPSink.cpp:784] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x8, 0
[ 89496][I][BluetoothA2DPSink.cpp:360] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 89505][I][BluetoothA2DPSink.cpp:784] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x10, 0
[ 89513][I][BluetoothA2DPSink.cpp:360] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 89523][I][BluetoothA2DPSink.cpp:784] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x20, 

`

pschatzmann, 2022-04-10:
When the phones sends the audio you should see the following log lines
12:33:24.545 -> [125087][D][BluetoothA2DPSink.cpp:936] audio_data_callback(): stream_reader
12:33:24.545 -> [125108][D][BluetoothA2DPSink.cpp:929] audio_data_callback(): audio_data_callback
12:33:24.545 -> [125110][D][BluetoothA2DPSink.cpp:936] audio_data_callback(): stream_reader
12:33:24.580 -> [125131][D][BluetoothA2DPSink.cpp:929] audio_data_callback(): audio_data_callback
12:33:24.580 -> [125132][D][BluetoothA2DPSink.cpp:936] audio_data_callback(): stream_reader
12:33:24.580 -> [125154][D][BluetoothA2DPSink.cpp:929] audio_data_callback(): audio_data_callback

I suggest to check and visualize the received audio data with the bt_music_receiver_datacallback.ino sketch in the Serial Plotter...

BruceAKABear, 2022-04-10:
> When the phones sends the audio you should see the following log lines (with Log Level Verbose) 12:33:24.545 -> [125087][D][BluetoothA2DPSink.cpp:936] audio_data_callback(): stream_reader 12:33:24.545 -> [125108][D][BluetoothA2DPSink.cpp:929] audio_data_callback(): audio_data_callback 12:33:24.545 -> [125110][D][BluetoothA2DPSink.cpp:936] audio_data_callback(): stream_reader 12:33:24.580 -> [125131][D][BluetoothA2DPSink.cpp:929] audio_data_callback(): audio_data_callback 12:33:24.580 -> [125132][D][BluetoothA2DPSink.cpp:936] audio_data_callback(): stream_reader 12:33:24.580 -> [125154][D][BluetoothA2DPSink.cpp:929] audio_data_callback(): audio_data_callback
> 
> I suggest to check and visualize the received audio data with the bt_music_receiver_datacallback.ino sketch in the Serial Plotter...
> 
> This will help to identify if the issue in with A2DP or I2S.
> 
> ps. what is the reason why you used channel_format = I2S_CHANNEL_FMT_ONLY_LEFT in the other project ?

ok, my bad. I define a pin to control chip mute.the lib worked now

closed 2022-04-10 by BruceAKABear