pschatzmann/ESP32-A2DP #210: how set UDA1334A Breakout Board with LSB-justified 24 bits input

0neday, 2022-04-16:
I have check UDA1334A  pcb scheme [here ](https://cdn-learn.adafruit.com/assets/assets/000/048/419/original/adafruit_products_schem.png?1511325937), and found LSB-justified 24 bits input when sfor0 and sfor1 is HIGH. 
[image]

set 
```
.bits_per_sample =  .bits_per_sample = (i2s_bits_per_sample_t) 24,
 .communication_format = (i2s_comm_format_t) (I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_LSB), 
```
it will be work fine?

pschatzmann, 2022-04-16:
What exactly are you trying to do ? Is the [described settings](https://github.com/pschatzmann/ESP32-A2DP/wiki/External-DAC) not working ?

0neday, 2022-04-16:
thank you, just to test

pschatzmann, 2022-04-16:
I don't think that you need to mess with any SFOR settings. 
Why don't you just set the i2s_config.bits_per_sample to 24 or 32 ?

0neday, 2022-04-17:
I set back to 16bit sample

closed 2022-04-17 by 0neday