pschatzmann/ESP32-A2DP #211: Question about auto reconnect

franzSusemichel, 2022-04-17:
Hi, 

I'm using the ESP32-A2DP library and i'm not sure about the auto reconnect feature. 

I try to build a speaker with a power save mode and therefore  i want to stop Bluetooth sink when entering power saving. It works for me when using a2dp_sink.end(false) and restart it with a2dp_sink.start(true, true) which sould use the auto reconnect feature (but it doesn't work). 

Isn't it working as aspected or do i use the functions not as intended? 

Kind regards
Franz

pschatzmann, 2022-04-17:
By calling end() you basically indicate that you do not want to reconnect. 
What happens if you don't call end ?

franzSusemichel, 2022-04-17:
If I don't call end the device will stay connected. Any other options to disable bluetooth and activate it later again?

pschatzmann, 2022-04-17:
You can try disconnect() ?

franzSusemichel, 2022-04-17:
I'm getting core panic exceptions as soon as i call end method (without any own code). 

Is there an option to make device not visible?

pschatzmann, 2022-04-17:
Depends what you mean with this and your IDF release.
Maybe you look for set_discoverability()

franzSusemichel, 2022-04-17:
Thx. When i manage visibility, scan_mode and last connection on my own everything works as aspect. Shall i close the issue or do you want to look after twice end call? 

Really appreciate your help :)

pschatzmann, 2022-04-17:
I suggest that you close.

closed 2022-04-17 by franzSusemichel