pschatzmann/ESP32-A2DP #212: Incompatible with ESP8266Audio

0neday, 2022-04-17:
when I merge ESP32-A2DP into [MrDiy-Audio-Notifier](https://github.com/schmurtzm/MrDiy-Audio-Notifier) which use MQTT to subcribe message to play media,  cannot play it. 

when I commit code related ESP32-A2DP, it work.
I guess it is incompatible with ` out = new AudioOutputI2S(); ` 

```
[MQTT] media/IPAddress  192.168.1.225
[MQTT] media/status     idle
State changed from: 3 to 4

Requested [media/play] http://192.168.1.2:8124/2100d0b3e2307e1ae2aea8fb366dba020c57048e_zh-cn_-_google_translate.mp3
[MQTT] media/status     playing
...#Interrupted!
```

pschatzmann, 2022-04-17:
Did you activate the logging. I expect that you just run out of memory!

0neday, 2022-04-18:
how to enable a2dp debug info?
without out of memory.
the callback just for music sender, not music reciever, that not work for me

pschatzmann, 2022-04-18:
If you use Arduino -> Tools -> Core Debug Lebel 
Please read the section Accessing the Sink Data Stream with Callbacks
There is even an [example](https://github.com/pschatzmann/ESP32-A2DP/blob/main/examples/bt_music_receiver_datacallback/bt_music_receiver_datacallback.ino)

0neday, 2022-04-18:
using platform IO ,  still not work 
```
-1192,-656
-1402,-985
-1549,-1131
-1667,-1210
-1715,-1328
-1719,-1414
-1734,-1410

Requested [media/play] http://192.168.1.2:8124/f0a68ea54d3012aa1b46e37445097bd98362cf95_zh-cn_-_google_translate.mp3
[MQTT] media/status     playing
...#Interrupted!
```

0neday, 2022-04-18:
I test, commit all code except 
```
  a2dp_sink.start("NodeMCU-32S-BLE");

```
bluetooth speeker work fine,  ESP8266Audio not

pschatzmann, 2022-04-18:
If you use PlatformIO read https://github.com/pschatzmann/ESP32-A2DP/wiki/PlatformIO
The log level is defined by CORE_DEBUG_LEVEL

0neday, 2022-04-18:
log is here

```

[I][BluetoothA2DPSink.cpp:360] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[D][BluetoothA2DPSink.cpp:325] app_work_dispatched(): app_work_dispatched
[D][BluetoothA2DPSink.cpp:1154] ccall_av_hdl_a2d_evt(): ccall_av_hdl_a2d_evt
[D][BluetoothA2DPSink.cpp:528] av_hdl_a2d_evt(): av_hdl_a2d_evt evt 1
[D][BluetoothA2DPSink.cpp:537] av_hdl_a2d_evt(): av_hdl_a2d_evt ESP_A2D_AUDIO_STATE_EVT
[D][BluetoothA2DPSink.cpp:606] handle_audio_state(): handle_audio_state evt 1
[I][BluetoothA2DPSink.cpp:608] handle_audio_state(): A2DP audio state: Suspended
[W][BluetoothA2DPSink.cpp:624] handle_audio_state(): i2s_stop

Requested [media/play] http://192.168.1.2:8124/484795998ec386b0b05ead13c83fd84b20702be2_zh-cn_-_google_translate.mp3
[V][HTTPClient.cpp:245] beginInternal(): url: http://192.168.1.2:8124/484795998ec386b0b05ead13c83fd84b20702be2_zh-cn_-_google_translate.mp3
[D][HTTPClient.cpp:293] beginInternal(): protocol: http, host: 192.168.1.2 port: 8124 url: /484795998ec386b0b05ead13c83fd84b20702be2_zh-cn_-_google_translate.mp3
[D][HTTPClient.cpp:579] sendRequest(): request type: 'GET' redirCount: 0

[D][HTTPClient.cpp:1125] connect():  connected to 192.168.1.2:8124
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'HTTP/1.0 200 OK'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'Server: SimpleHTTP/0.6 Python/3.9.7'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'Date: Mon, 18 Apr 2022 06:11:21 GMT'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'Content-type: audio/mpeg'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'Content-Length: 40237'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'Last-Modified: Mon, 18 Apr 2022 06:11:21 GMT'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: ''
[D][HTTPClient.cpp:1257] handleHeaderResponse(): code: 200
[D][HTTPClient.cpp:1260] handleHeaderResponse(): size: 40237
[D][HTTPClient.cpp:603] sendRequest(): sendRequest code=200

[MQTT] media/status     playing
...#Interrupted!

[D][HTTPClient.cpp:378] disconnect(): still data in buffer (5556), clean up.

[D][HTTPClient.cpp:387] disconnect(): tcp stop
[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[MQTT] media/status     idle

```

0neday, 2022-04-18:
after commit 
```
  // a2dp_sink.start("NodeMCU-32S-BLE");

```
log is here 
```
==================================================================
  MrDIY Notifier - Modified By Schmurtz 
==================================================================

Connected to Wifi       [192.168.1.225]
Connected to MQTT
[MQTT] media/LWT        online
[MQTT] media/SSID       NodeMCU-32S-AP
[MQTT] media/IPAddress  192.168.1.225
[MQTT] media/status     idle
State changed from: 3 to 4

Requested [media/play] http://192.168.1.2:8124/367f479200d9e0f1f1be0301d3422d84bf147d9b_zh-cn_-_google_translate.mp3
[V][HTTPClient.cpp:245] beginInternal(): url: http://192.168.1.2:8124/367f479200d9e0f1f1be0301d3422d84bf147d9b_zh-cn_-_google_translate.mp3
[D][HTTPClient.cpp:293] beginInternal(): protocol: http, host: 192.168.1.2 port: 8124 url: /367f479200d9e0f1f1be0301d3422d84bf147d9b_zh-cn_-_google_translate.mp3
[D][HTTPClient.cpp:579] sendRequest(): request type: 'GET' redirCount: 0

[D][HTTPClient.cpp:1125] connect():  connected to 192.168.1.2:8124
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'HTTP/1.0 200 OK'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'Server: SimpleHTTP/0.6 Python/3.9.7'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'Date: Mon, 18 Apr 2022 06:15:58 GMT'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'Content-type: audio/mpeg'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'Content-Length: 40237'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'Last-Modified: Mon, 18 Apr 2022 06:15:57 GMT'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: ''
[D][HTTPClient.cpp:1257] handleHeaderResponse(): code: 200
[D][HTTPClient.cpp:1260] handleHeaderResponse(): size: 40237
[D][HTTPClient.cpp:603] sendRequest(): sendRequest code=200

[MQTT] media/status     playing
Free: 139972  (46980 lost)  -----------
Free: 140828  (47836 lost)  -----------
Free: 141032  (48040 lost)  -----------
Free: 141032  (48040 lost)  -----------
Free: 141032  (48040 lost)  -----------
Free: 142548  (49556 lost)  -----------
Free: 141004  (48036 lost)  -----------
Free: 141028  (48036 lost)  -----------
Free: 142544  (49552 lost)  ----------- 
Free: 146412  (53420 lost)  ----------- 
[D][WiFiClient.cpp:514] connected(): Disconnected: RES: 0, ERR: 128
[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

...#Interrupted!

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[MQTT] media/status     idle
```

pschatzmann, 2022-04-18:
As stated in my last comment this can't work. I am closing this issue as duplicate of https://github.com/pschatzmann/ESP32-A2DP/issues/19

closed 2022-04-18 by pschatzmann

0neday, 2022-04-18:
Thank your big work!

0neday, 2022-04-18:
I search 
```
[D][HTTPClient.cpp:378] disconnect(): still data in buffer (5556), clean up.
```
and found 
https://github.com/espressif/arduino-esp32/issues/3715#issuecomment-582606180

so, esp32 bluetooth and wifi share same radio channel ?