pschatzmann/ESP32-A2DP #214: Finding an example for i2s > a2dp > windows

street-grease-coder, 2022-04-21:
trying to record from a INP443 over bluetooth on my windows machine (to register as a bluetooth mic). Coulnd't find the proper example, in this one ([link](https://www.pschatzmann.ch/home/2021/04/29/bluetooth-a2dp-streaming-from-an-digital-i2s-microphone/) ) I get a compilation error ('Channels' not found).

One more thing: Is it possible to send raw keycodes (e.g. 'enter key') over r2dp with your libraries (or are only media buttons supported, as I suspect)?

Thanks for the great work

pschatzmann, 2022-04-21:
Just use the example from Github! I am not updating the code examples on my Blogs as they.

I do not understand your second question. What is r2dp and what are raw keycodes?

street-grease-coder, 2022-04-21:
Sorry, let me be more clear/specific.
First I tried your example [streams-i2s-a2dp.ino](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-i2s-a2dp/streams-i2s-a2dp.ino) which compiled nicely. However, if I get it right, this won't broadcast a bluetooth device name to be connected to (in my case, that would be my windows machine), but rather attempts to directly connect to a named device.

I guess my question is simply a)how do I make it broadcast/advertise so I can connect to my windows machine and b) how do I send 'more' than play/pause keys using your libraries (e.g. the 'arrow up' key, or any other keycode for that matter)

pschatzmann, 2022-04-21:
I am sorry I can't provide any Windows support. I suggest to google how to turn windows into a bluetooth sink.
I haven't worked with Windows since 20 years...
The example is intended to be used with a Bluetooth Speaker.

street-grease-coder, 2022-04-21:
ok - I think I'm just being silly for not finding the proper example in your libraries. Can you simply point me to the right example to use my external i2s microphone with my esp32 as bluetooth source?

pschatzmann, 2022-04-21:
https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-i2s-a2dp
https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-basic-api/base-i2s-a2dp

street-grease-coder, 2022-04-21:
thanks! Is there an example to advertise the bluetooth source without specifying the name of the device that is supposed to be connected to? (i.e. wait until any target device connects?)

pschatzmann, 2022-04-21:
This does not make any sense: You need to know the device you want to connect to!
That's the reason why I recommended you to test it with the phone first - there you can see the name!

street-grease-coder, 2022-04-21:
I mean, I basically want to connect a i2s microphone over bluetooth (using an esp32). maybe I misunderstand completely, but I imagined the way this would be done is:
i2s from mic > esp32 > bluetooth device is advertised > search for bluetooth device (e.g. 'my bluetooth enabled microphone') on my phone / bluetooth enabled machine > connect to source (as 'microphone') > access microphone input using Audacity on my machine/phone

pschatzmann, 2022-04-21:
Lets start with the standard case. You have a mobile phone that you connect to a bluetooth speaker: On the mobile phone you can select the bluetooth speaker.

Now to the example: The ESP32 plays the role of the mobile phone - but because you do not have a screen you need to tell in advance to which device you want to connect to.

street-grease-coder, 2022-04-21:
> Lets start with the standard case. You have a mobile phone that you connect to a bluetooth speaker: On the mobile phone you can select the bluetooth speaker.
>
> Now to the example: The ESP32 plays the [...]

using this example, my phone (i.e. Bluetooth mic) wouldn't have a screen, but my speaker (i.e. my machine with attached) would.

street-grease-coder, 2022-04-21:
i don't want to attach speakers, i want to attach a microphone *upsidedown smiley*

pschatzmann, 2022-04-21:
Then A2DP is not the right solution for your problem

street-grease-coder, 2022-04-21:
For reference for future readers:
This (the thumbnail of the video with the puking smiley face) summarizes my struggles and explains that a2dp is not suitable, recommends Bluetooth handsfree profile.
https://youtu.be/0jR-QNTfydA

There are cheap bluetooth audio transmitters (also called emitters) explained well in this video
https://youtu.be/ZQ5MWcis8rA that seem to do the work reasonably well. I have not found a better alternative

pschatzmann, 2022-04-22:
Did you have the chance to measure the lag of these modules ? 
Using an ESP32 it is too big to use it with a live a musical instrument.

street-grease-coder, 2022-04-22:
Oh I just ordered, will arrive in months... it's for speech/dictation, like a karaoke 🎤 anything less than 100ms delay should be fine, I wouldn't expect a lot worse

closed 2022-04-24 by pschatzmann