pschatzmann/ESP32-A2DP #224: Issue with Updated library

zeeshan27102000, 2022-05-08:
Hi,
I am trying to play music through Bluetooth using BluetoothA2DPSink library in Bluetooth receiver simple code. I noticed that BluetoothA2DP is updated library. Earlier it was BlootoothA2D. I have no issues with both old and updated library when I just do music playing. But when I combine the code in another code where I am doing speech recognition parallely. Old library(BlootoothA2D) works without any issue but it doesn't have new features like pause, play which I want to use it in my code. Now when I use updated library (BluetoothA2DP) which has new features, it doesn't work properly. Bluetooth disconnects very frequently. Once it disconnects, it's hard reconnect again. Same code works fine with old library. So it's the issue with the updated one.

pschatzmann, 2022-05-08:
I am confused: this library is supporting AVRC since 29 Mar 2021  - So you when you way old you really mean old!
- What Arduino ESP32 Version are you using ? 
- How can I reproduce your issue ?

zeeshan27102000, 2022-05-08:
I'm using ESP-32 WROOM 32D. I tried with both old and updated one.
I'm unable to do a2dp_sink.pause()/play() using old library. It says undefined. So how do I do it using old one?

pschatzmann, 2022-05-08:
https://github.com/pschatzmann/ESP32-A2DP/wiki/Controlling-your-Phone-with-AVRC-Commands
The lastest version is 2.0.3

zeeshan27102000, 2022-05-08:
So how can I add AVCR commands in 1.0.6?
I get no issues using that version

pschatzmann, 2022-05-08:
Sorry this is not supported by me and since the author has deactivated the issues tab he seems to be not willing to support this...

zeeshan27102000, 2022-05-08:
Any idea on how do it myself?

pschatzmann, 2022-05-08:
My recommendation would really be to use the latest versions of all software and then you need to try to figure out what you are doing wrong. My gut feeling is that 
- you are running out of memory
- do not give A2DP enough processing time

zeeshan27102000, 2022-05-08:
Sorry I really new to microcontroller programming. So I have no idea on how to restrict processing time. How do I achieve that?

zeeshan27102000, 2022-05-08:
The code with v2.0.3 and the older version mentioned above use same program space. So I think I'm not running out of memory. So I tried to add AVCR commands on my own to the library mentioned above. I get no errors during compilation when I use a2dp_sink.pause() and sketch uplloads successfully to ESP32 but it is not doing expected task i.e it does not pause. What could be the problem as there are no errors?
Here's the code

pschatzmann, 2022-05-08:
Program space is not the issue: You need to monitor the heap!
https://techtutorialsx.com/2017/12/17/esp32-arduino-getting-the-free-heap/

zeeshan27102000, 2022-05-08:
Thank you so much for giving your time. Can you please have a look at the code which i've edited. I've added link of the code.

I agree that there is extensive processing in the loop because i'm doing speech recognition but my concern is why does it work so well with older version of your library even though there is extensive processing?
I just want AVCR command features which is present in latest version and that doesn't work well with my code. It's so contradicting

pschatzmann, 2022-05-08:
Let's take one step back: What are you trying to achieve? Can you give me a little bit more detail what you sketch is doing exactly ?

zeeshan27102000, 2022-05-08:
So I'm doing speech recognition while playing music. I've set some keywords like names which will be heard using mic. The mic will continuously hear for keywords. Whenever that Keyword is heard/spotted, it should pause the music and notify the listener that his/her name is called. So that's the reason I need AVCR commands here

pschatzmann, 2022-05-08:
Yes, but what library are you using for this ?
Is there an example that you base your work on ?

zeeshan27102000, 2022-05-08:
Yes, I'm training model on edge impulse and then downloadeing a customised library of speech recognition for my project.
Here's the code which i'm working on
https://github.com/HJGrant/esp32-speech-recognition

pschatzmann, 2022-05-08:
TFlite is using a lot of Ram and A2DP as well: I would be really surprised if this works together.
If you want to use a I2S microphone, do not forget that A2DP is already using port 0 ...

zeeshan27102000, 2022-05-08:
Yes, I'm aware that microphone uses I2S port 0. So I'm using port 1 for Bluetooth music streaming. The code works together flawlessly with older version of A2DP but I'm unable to pause as it doesn't have AVCR commands.
Therefore I have to opt for latest updated version which doesn't work well. When I use latest of A2DP library, Bluetooth disconnects very frequently and speech recognition doesn't seem to happen as well.

pschatzmann, 2022-05-08:
I was combining my [TFLite Speech Example](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-audiokit/streams-audiokit-tf) and the [A2DP Example](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/basic-a2dp-audiokit/basic-a2dp-audiokit.ino) in one sketch and did not get any of your issues:
The free heap with both running was around 86308

zeeshan27102000, 2022-05-09:
Can you try checking with my example?
https://github.com/HJGrant/esp32-speech-recognition

pschatzmann, 2022-05-09:
I tried and my Bluetooth is working perfectly: Just the speech recognition does not work because it needs to much memory. Event the cleanup up the unnecessary int16_t sampleBuffer[16000]; did not help and it seems that a lot more memory is needed then we have available.
I tried with my own newly generated model, but this runs into the same limitations. 

My only explanation of the situation is, that the addtional functionaliy (which is basically only AVRC) is requiring much more memory now.

I am closing this issue because my library is working as designed and you would need to figure out if there is a way that edge-impulse is needing less heap. I find this slightly strange because they promote their library to be more  meemory efficient then TFLite - but my TFLite model is just working fine!

closed 2022-05-09 by pschatzmann