pschatzmann/ESP32-A2DP #230: Way to obtain RSSI value

vermar-js, 2022-05-20:
Hi, 

I am looking or a way to obtain the RSSI o the connection once the phone is paired with my ESP 32. Is there a way to do that? Does the A2DP sink object class support classic Arduino Bluetooth calls? 

Here is my code? 

`#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;
esp_a2d_connection_state_t last_state;
void setup() {
  // put your setup code here, to run once:
a2dp_sink.start("Pitch");
  Serial.begin(115200);
  Serial.println("started");

}

void loop() { 

  if (a2dp_sink.is_connected ()) {
      Serial.println("CONNECTED"); 
//Here call something like connecteddevice.getRSSI(); ?
       //Serial.println(advertisedDevice.getRSSI()); //this is not working 
      // do something
  }else {
      Serial.println("not connected"); 
      // do something
  }

   delay(100);
   //Serial.println("disconect");
  // a2dp_sink.disconnect();
   //esp_deep_sleep_start(); shutting down 
   //a2dp_sink.end(); 

}`

pschatzmann, 2022-05-20:
I am not sure if something like Arduino Bluetooth exists. But it will work if you use the regular ESP32 IDF functionality:
https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/api-reference/bluetooth/esp_gap_bt.html.

You might need to create a subclass: see https://github.com/pschatzmann/ESP32-A2DP/wiki/Design-Overview

vermar-js, 2022-05-20:
> I am not sure if something like Arduino Bluetooth exists. But it will work if you use the regular ESP32 IDF functionality: [...]

mmh thank you, so I cannot work within the Arduino IDE?

pschatzmann, 2022-05-20:
Yes you can...

vermar-js, 2022-05-20:
Thank you, I will try to figure it out! 
will the BluetoothSerial.h arduino library work with your library?

pschatzmann, 2022-05-20:
I have never tried, so I don't know: In any case you can have only one Bluetooth functionality active. So it will not work if A2DP is active...

vermar-js, 2022-05-20:
So it can't be paired as A2DP and mapping RSSI at the same time?  I don't need sound to go through, I am only using A2DP as a way to advertise and pair the esp32 with my phone?

pschatzmann, 2022-05-20:
I have never tried, so I don't know. Just give it a try...

vermar-js, 2022-05-20:
I understand, thank you for your advice. I will to find some help to achieve this. Have a good day :)

pschatzmann, 2022-06-24:
I have added the following methods:
- set_rssi_active()
- get_last_rssi()
- set_rssi_callback()

closed 2022-06-24 by pschatzmann