pschatzmann/ESP32-A2DP #233: DAC outputs 1.64V constantly

benjamin32561, 2022-06-02:
Tried to use the bt_music_receiver_to_internal_dac example but the output on pins 25 and 26 is constant at 1.64V even when connected and playing music
[examples/bt_music_receiver_to_internal_dac/bt_music_receiver_to_internal_dac.ino](url)

pschatzmann, 2022-06-02:
What ESP32 Board Version are you using ?
Please note the ADC is broken for version > 1.0.2 and < 2.0.3

benjamin32561, 2022-06-02:
> What ESP32 Board Version are you using ? Please note the internal DAC is broken for version > 1.0.2 and < 2.0.3

esp-wroom-32
not sure about version tho... how can ii check this?

pschatzmann, 2022-06-02:
-> Tools -> Board -> Board Manager 
Enter ESP32
The you should see 
esp32 by Espressif Systems version e.g. 2.0.3

benjamin32561, 2022-06-02:
it says:
by Espressif Systems version 2.0.3
So 2.0.3

pschatzmann, 2022-06-02:
So it should work:  I would expect to see around 1.5 volts and if you have a max signal it will osciallate between 0 and 3.0
Did you test with some headphones ?

benjamin32561, 2022-06-02:
> So it should work: I would expect to see around 1.5 volts and if you have a max signal it will osciallate between 0 and 3.0 with 1.5 volt in neutral position. Did you test with some earphones ?

I didn't try it with earphones but even when I stream music the voltage stays the same

pschatzmann, 2022-06-02:
As I was explaining before you will alway measure around 1.5 volts which is the average of the max and the min values.

benjamin32561, 2022-06-03:
OK it works, but the sound is very bad... do you know how I can solve the sound issue?