pschatzmann/ESP32-A2DP #234: Problem to send MP3 sound by ESP32 DEV kit to a Bluetooth Speaker

williamesp2015, 2022-06-05:
I have hard time to test a simple mp3 file to a bluetooth Speaker. Any help would be very appreciated.

pschatzmann, 2022-06-05:
I suggest to try to use https://github.com/pschatzmann/arduino-audio-tools:
Starting from [this example](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-sd_mp3-i2s) you could replace the I2SStream with a A2DPStream.

williamesp2015, 2022-06-05:
@pschatzmann . Thank you for suggestion. I couldn't find example to replace Bluetooth Speaker name with I2SStream. It would be very nice if you kindly add an example.

pschatzmann, 2022-06-05:
Take a look at https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-generator-a2dp

williamesp2015, 2022-06-05:
After more search I decided to do streaming MP3 file using SdFat but no sound can hear from the BT Speaker. It might be due to delicate problem in setting SdFat configuration:

#define USE_SDFAT
#include "AudioTools.h"
#include "AudioLibs/AudioA2DP.h"
#include "AudioCodecs/CodecMP3Helix.h"
const char *startFilePath="/";
const char* ext="mp3";
int speedMz = 15;
AudioSourceSdFat source(startFilePath, ext, SD_CS_PIN, speedMz);
A2DPStream out = A2DPStream::instance();  // A2DP input - A2DPStream is a singleton!
MP3DecoderHelix decoder;
AudioPlayer player(source, out, decoder);

void setup() {
  Serial.begin(115200);
Serial.print(F("Initializing SD card..."));
SPI.begin(SDCARD_SCK_PIN,SDCARD_MISO_PIN,SDCARD_MOSI_PIN,SDCARD_SS_PIN);

  AudioLogger::instance().begin(Serial, AudioLogger::Warning);

  // setup output - We send the test signal via A2DP - so we conect to the "LEXON MINO L" Bluetooth Speaker
  auto cfg = out.defaultConfig(TX_MODE);
  cfg.name = "BT SPEAKER";
  //cfg.auto_reconnect = true;  // if this is use we just quickly connect to the last device ignoring cfg.name
  out.begin(cfg);

  // setup player
  player.setVolume(0.1);
  player.begin();

}

void loop() {
  player.copy();
}

pschatzmann, 2022-06-06:
I think your speedMz is much too high.
Start with the proposed default value: which is only 2

williamesp2015, 2022-06-06:
SdFat was not implemented for the Custome SPI port and pinout.

pschatzmann, 2022-06-06:
In your case there is no need to use VSPI: Custom pin assignments work perfectly with SPI.

williamesp2015, 2022-06-07:
Yes indeed but I want it for my project using VSPI for SdFat and HSPI for another purpose.

pschatzmann, 2022-06-07:
So I would just turn it around and use VSPI for your other purpose

williamesp2015, 2022-06-08:
A bonus would be a bit modification in the Audioplaye.h to accept HSPI or VSPI with custome pinout for ESP32 :
SPIClass ADS_SPI(HSPI); // for SPI DEVICE
 SPIClass SD_SPI(VSPI);// FOR SD CARD
 #define SPI_CLOCK SD_SCK_MHZ(15) // ESP32 can handle SdFat at 15 Mhz
AudioSourceSdFat source(startFilePath, ext, SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SPI_CLOCK, &SD_SPI));
A2DPStream out = A2DPStream::instance();  // A2DP input - A2DPStream is a singleton!
MP3DecoderHelix decoder;
AudioPlayer player(source, out, decoder);
.........
void setup()
{
   Serial.begin(115200);
   Serial.print(F("Initializing SD card..."));
   bool sdinit = sd.begin(SD_CONFIG);
   delay(10);
   if (!sdinit) {
     Serial.println("SD ERROR 1");
     sd.initErrorHalt( & Serial);
     return;
   }
    else
    {
      Serial.println("SD is GOOD");  
      AudioLogger::instance().begin(Serial, AudioLogger::Warning);
      auto cfg = out.defaultConfig(TX_MODE);
      cfg.name = "BT SPEAKER";
      out.begin(cfg);
      // setup player
      player.setVolume(0.9);//0.1);
      player.begin();
}
void loop() {
  player.copy();
}

pschatzmann, 2022-06-08:
Feel free to sumbit a pull request

williamesp2015, 2022-06-08:
Unfortunately, I have no clue to modify Audioplaye.h.

pschatzmann, 2022-06-08:
I don't think any change to the AudioSourceSdFat is needed:
- You demonstrated that you know how to  assign the SPIClass object to the SdSpiConfig
- You demonstrated that you know how to assign custom pins by calling the begin method on the object of the SPIClass.

I was [testing this setup on an AudioKit](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/player-sd-audiokit/player-sd-audiokit.ino) both with HSPI and VSPI and it works perfectly.

What was your problem again ?

a442509097, 2022-06-11:
https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-player/player-sd-a2dp/player-sd-a2dp.ino
This demo works very well, but the sd compatibility is not very good, I try to change another sd card and it works fine

closed 2022-06-24 by pschatzmann

williamesp2015, 2022-06-24:
For custom SPI Pins anyone interested can do modification in the SdConfig.h and add custom SPI pinout using SPI.begin(SDCARD_SCK_PIN,SDCARD_MISO_PIN,SDCARD_MOSI_PIN,SDCARD_SS_PIN);

replace with the last line
#define SDCARD_MISO_PIN 19
#define SDCARD_MOSI_PIN 23
#define SDCARD_SCK_PIN 18
#define SDCARD_SS_PIN 5
#endif  // SdFatConfig_h
  #define USE_SDFAT
  #include "AudioTools.h"
  #include "AudioLibs/AudioA2DP.h"
  #include "AudioCodecs/CodecMP3Helix.h"
  const char startFilePath[]="/";
  const char ext[]="mp3";

  int speedMz = 15;
  const uint8_t SD_CS_PIN = SDCARD_SS_PIN;
  AudioSourceSdFat source(startFilePath, ext, SD_CS_PIN, speedMz);
  A2DPStream out = A2DPStream::instance(); // A2DP input - A2DPStream is a singleton!
  MP3DecoderHelix decoder;
  AudioPlayer player(source, out, decoder);

  void setup() {
  Serial.begin(115200);
  Serial.print(F("Initializing SD card..."));
  SPI.begin(SDCARD_SCK_PIN,SDCARD_MISO_PIN,SDCARD_MOSI_PIN,SDCARD_SS_PIN);
  delay(10);
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);

  // setup output - We send the test signal via A2DP - so we conect to the "LEXON MINO L" Bluetooth Speaker
  auto cfg = out.defaultConfig(TX_MODE);
  cfg.name = "BT SPEAKER";
  //cfg.auto_reconnect = true; // if this is use we just quickly connect to the last device ignoring cfg.name
  out.begin(cfg);

  // setup player
  player.setVolume(0.9);
  player.begin();

  }

  void loop() {
  player.copy();
  }

williamesp2015, 2022-06-24:
https://github.com/pschatzmann/ESP32-A2DP/issues/234#issuecomment-1150346682
Seems 

> 

Many thanks. It is agood start to have VSPI/HSPI but after modify AUDIOKIT_BOARD in the AudiokitSetting.h problem is with Audiokit.h 
AudioKit.h : 189 - begin faild: please verify your AUDIOKIT_BOARD setting: 6
I do not use any Audiokit and My custom design ESP32 board have START/STOP/NEXT/.. keys

williamesp2015, 2022-06-25:
This modified eample is working for ESP32
#define USE_SDFAT
#include "AudioTools.h"
#include "AudioCodecs/CodecMP3Helix.h"

const char *startFilePath="/";
const char* ext="mp3";
SPIClass SD_SPI(VSPI);
SPIClass HSPI_SPI(HSPI);//FOR OTHER PURPOSE
SdSpiConfig sdcfg(SDCARD_SS_PIN, DEDICATED_SPI, SD_SCK_MHZ(15) , &SD_SPI);
AudioSourceSdFat source(startFilePath, ext, sdcfg);
MP3DecoderHelix decoder;  // or change to MP3DecoderMAD

#include "AudioLibs/AudioA2DP.h"
A2DPStream out = A2DPStream::instance(); // A2DP input - A2DPStream is a singleton!
AudioPlayer player(source, out, decoder);

void next(bool, int, void*) {
   player.next();
}

void previous(bool, int, void*) {
   player.previous();
}

void setup() {
  Serial.begin(115200);
  //SD Init
  pinMode(SDCARD_SS_PIN, OUTPUT);
  digitalWrite(SDCARD_SS_PIN, HIGH);
  /////////////////
  //Custom SD_SPI Pinout
  // Make sure you add/changed SD card pinout in the last line of sdFatConfig.h (for ESP32)
/*  #define SDCARD_MISO_PIN 19
  #define SDCARD_MOSI_PIN 21
  #define SDCARD_SCK_PIN  18
  #define SDCARD_SS_PIN   23 */
  SD_SPI.begin(SDCARD_SCK_PIN,SDCARD_MISO_PIN,SDCARD_MOSI_PIN,SDCARD_SS_PIN);//SCK, MISO, MOSI, SS);
  delay(10);// a dely
  Serial.print("CS ");
  Serial.println(SDCARD_SS_PIN);// 23//13
  Serial.print("SO ");
  Serial.println(SDCARD_MISO_PIN);// 19//2
  Serial.print("SI ");
  Serial.println(SDCARD_MOSI_PIN);//15
  Serial.print("CLK ");
  Serial.println(SDCARD_SCK_PIN);// 19//2
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);
  auto cfg = out.defaultConfig(TX_MODE);
  cfg.name = "BT SPEAKER";
  out.begin(cfg);

  // setup player
  player.setVolume(0.9);
  player.begin();

}
void loop() {
  player.copy();
}