pschatzmann/ESP32-A2DP #236: can not return correct play_pos

biglee991228, 2022-06-17:
I wrote the follows in my loop(),but always returned 0 and Serialprint out.

uint32_t play_position = esp_avrc_ct_send_get_rn_capabilities_cmd(ESP_AVRC_RN_PLAY_POS_CHANGED);
Serial.println(play_position);
delay(1000);

I don't know why because I used new version 2.0.3

pschatzmann, 2022-06-17:
A2DP methods are returning an esp_error_t so if you get 0 it means that the command was executed successfully. 

Result processing in A2DP is always done via the event handler.  If you check my code I am doing this call already both in the source and sink and the result should be available in s_avrc_peer_rn_cap.bits.

biglee991228, 2022-06-17:
but how can I print out the "s_avrc_peer_rn_cap.bits"

pschatzmann, 2022-06-24:
I have added the get_avrc_peer_rn_cap_bits() method

closed 2022-06-24 by pschatzmann