pschatzmann/ESP32-A2DP #237: Simple Sink test - ESP IDF Example

mitchjs, 2022-06-17:
just trying the ESP IDF Example
basic one liner

void setup(){
  a2dp_sink.start("MyMusic");  
}

it works, till i start playing with the volume control on the iphone
after a bit of up/down...

***ERROR*** A stack overflow in task BtAppT has been detected.

Backtrace:0x40081bba:0x3ffd54d00x400903a9:0x3ffd54f0 0x40093526:0x3ffd5510 0x400921ca:0x3ffd5590 0x400904a8:0x3ffd55b0 0x4009045a:0x33336330  |<-CORRUPTED
0x40081bba: panic_abort at C:/Users/mitchjs/esp-v4.4/esp-idf/components/esp_system/panic.c:402

0x400903a9: esp_system_abort at C:/Users/mitchjs/esp-v4.4/esp-idf/components/esp_system/esp_system.c:128

0x40093526: vApplicationStackOverflowHook at C:/Users/mitchjs/esp-v4.4/esp-idf/components/freertos/port/xtensa/port.c:394

0x400921ca: vTaskSwitchContext at C:/Users/mitchjs/esp-v4.4/esp-idf/components/freertos/tasks.c:3504

0x400904a8: _frxt_dispatch at C:/Users/mitchjs/esp-v4.4/esp-idf/components/freertos/port/xtensa/portasm.S:436

0x4009045a: _frxt_int_exit at C:/Users/mitchjs/esp-v4.4/esp-idf/components/freertos/port/xtensa/portasm.S:231

mitchjs, 2022-06-17:
i changed line 387 in BluetoothA2DPSink.cpp to:

if (xTaskCreate(ccall_app_task_handler, "BtAppT", **3072**, NULL, task_priority, &app_task_handle) != pdPASS){

and so far its happy

pschatzmann, 2022-06-24:
I have increased the default stack size. In addition it can be defined by calling:
- set_event_stack_size()
- set_i2s_stack_size()

closed 2022-06-24 by pschatzmann