pschatzmann/ESP32-A2DP #238: packet drops when changing volume on iphone to a2dp.sink

mitchjs, 2022-06-18:
just starting to mess with this

i currently for testing purposes using A2DPNoVolume

```
class novc : public A2DPNoVolumeControl {

  public:
    void update_audio_data(Frame* data, uint16_t frameCount) { 
    }
    
    void set_volume(uint8_t volume) {
    }
};
novc vc;
```

and in setup():
`a2dp_sink.set_volume_control(&vc);`

but when i move the volume on the iphone
audio breaks up, plenty of pkt dropped messages when changing the volume

IDF V4.4.1  (if that is in question)

W (26640) BT_APPL: Pkt dropped

I (26640) BT_API: app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
I (26650) BT_AV: AVRC set absolute volume: 27%
I (26650) BT_AV: Volume is set by remote controller to 27
I (26660) BT_API: app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
I (26670) BT_AV: AVRC set absolute volume: 27%
I (26670) BT_AV: Volume is set by remote controller to 27
W (26680) BT_APPL: Pkt dropped

any thoughts?

mitchjs, 2022-06-20:
update... i tested the A2DP.SINK example wiith IDF 4.4.1, it works perfectly when it comes to volume or other messages
i think its related to they have an I2STask which is very simple that does a ringbuffer implementation
they start the task on connect, and stop it on disconnect events

maybe this approach could be done here
im not the best coder in c++, but if i get some time it looks easy to implement
change the audio data callback ::audio_data_callback() to write to the ringbuffer
and create the same task they do to process it to i2s_write

mitch

pschatzmann, 2022-06-20:
Strange, I never had any issues like this in Arduino.
Do you have the same issue - if you don't do anything - and just use the standard volume control ?
Maybe your I2S buffer is just not big enough...

mitchjs, 2022-06-20:
i never tried arduino, just took you idf example, and built it... 
it seems to happen on very rapid volume changes
moving the slider on the iphone music player up and down rapidly
then ones it starts the pkt drops, its very easy to get it to happen
even with the side button volume control

ill go back and retest thier example, with the on screen volume (since it can move up down very fast)

mitch

mitchjs, 2022-06-20:
i got it to happen on the ESP-IDF Example.... when i use the MUSIC player's or the ios Control panel on screen volume slider
since one can move it very fast, move it up/down fast... and the packets will start dropping
and once it starts, then even physical button volume changes will cause it to drop

seems only on iphone... i guess its sending so many avrcp commands

i move the slider fast as i can on my android tablet and its fine

very strange, but might prevent me from using ESP32 for A2dp sink :(

pschatzmann, 2022-06-20:
This might be related to https://github.com/espressif/esp-idf/issues/6712

mitchjs, 2022-06-20:
i did read that, the avcrp are coming during audio playback, so that isnt an issue
which queue? the one in your code i did double, no change.... im going to mention that patch in the idf forum... see where it goes

oh that reminds me, did you see my issue with the task stack size, i had to jump that from 2048 to 3072 to keep from overflowing the bt_app task

mitchjs, 2022-06-23:
@pschatzmann , so with a little change as recommended @[BetterJincheng](https://github.com/BetterJincheng)  by  see https://github.com/espressif/esp-idf/issues/9192#issuecomment-1162576002

solved the problem with the official A2DP SNIK example code

for kicks, i did the a2dp_sink.set_task_priority(configMAX_PRIORITIES - 10);
but still get audio hiccups and you can see lower level issues

W (56270) BT_APPL: Pkt dropped

```
I (56280) BT_AV: Volume is set by remote controller to 44
E (56280) BT_L2CAP: l2cab is_cong_cback_context
I (56290) BT_API: app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
E (56290) BT_L2CAP: l2cab is_cong_cback_context
I (56300) BT_AV: AVRC set absolute volume: 44%
I (56310) BT_AV: Volume is set by remote controller to 44
E (56310) BT_L2CAP: l2cab is_cong_cback_context
```

while configMAX_PRIORITIES -10 is a HUGE IMPROVEMENT over the default -3
(again only happens on the iphone touch screen volume, as a finger can make that move fast)

i believe maybe a ringbuffer, and a task to move the data into the I2s_write might be a solution
OK i added it, and BINGO no issues!
also priority of main task to : configMAX_PRIORITIES -10

id like you to look at my code... im not a very good C++ programmer
dont really follow virtual/friend... how to handle static callbacks
but i did it like your existing main task: ccall_app_task_handler

my setup() is just:
```
a2dp_sink.set_task_priority(configMAX_PRIORITIES - 10);
a2dp_sink.start("MyMusic", true);  
```

im trying to create a pull request :) gotta learn github too...

i did it, odd with the compare...said every line differnt... its not

pschatzmann, 2022-06-24:
Code has been merged

closed 2022-06-24 by pschatzmann