pschatzmann/ESP32-A2DP #241: Compilation Error in BluetoothA2DPSink.cpp

williamesp2015, 2022-06-24:
Compiled with latest PlatrormIO- ESP32

BluetoothA2DPSink.cpp:31:20: error: no match for 'operator=' (operand types are 'i2s_config_t' a
nd '<brace-enclosed initializer list>')
         i2s_config = {
                    ^

pschatzmann, 2022-06-24:
Can you try again. I think this is because a pull request has removed a cast which is necessary in some old version...

mitchjs, 2022-06-24:
yes... sorry... i didnt know why you casted, it... add that back in
@pschatzmann one thing i didnt exactly mean merge the pull request to main, i ment merge the pull request with the update to the pull request... sorry, but i guess it is what it is and done
part of learning GIT crap... 
i really felt more people should have tested the pull request...

mitchjs, 2022-06-24:
@williamesp2015 

fix its prob line 36
            .communication_format = **(i2s_comm_format_t)** (I2S_COMM_FORMAT_STAND_I2S),

pschatzmann, 2022-06-24:
No worries: With the latest change the old releases should be working again...

closed 2022-06-24 by pschatzmann